ros2/ros2 #367: Create zero initialization methods for messages

tfoote, 2017-06-20:
In porting efforts we've run into several bugs where the zero initialized behavior is relied upon, causing bugs. Having a zero initialization method as a way to make porting less fragile would be valuable. And likely can be used in other cases when the optimization of using uninitialized constructors are not as important.

This is capturing the suggestion from @clalancette in ros2/geometry2#25

mikaelarguedas, 2017-08-08:
This should be eligible for the "hitlist". If not implementing these method, at least warn users and reflect on the design documents that the messages are not zero-initialized by default. (e.g. [this](http://design.ros2.org/articles/generated_interfaces_cpp.html) suggests that default constructors do initialize the fields )

@Karsten1987 FYI

dirk-thomas, 2017-08-10:
@clalancette will look into this and come up with a design doc / proposal.

clalancette, 2017-09-06:
Closing this in favor of #393

closed 2017-09-06 by clalancette