ros2/ros2 #368: New version of cppcheck has failures

wjwwood, 2017-06-21:
While upgrading eatable I found that the new version of `cppcheck` (`1.79`) has some issues with our code base (http://ci.ros2.org/job/ci_windows_testing/2/testReport/):

- http://ci.ros2.org/job/ci_windows_testing/2/testReport/rmw_fastrtps_cpp/cppcheck/error__syntaxError__include_rmw_fastrtps_cpp_MessageTypeSupport_impl_h_28_/
- http://ci.ros2.org/job/ci_windows_testing/2/testReport/rmw_fastrtps_cpp/cppcheck/error__syntaxError__include_rmw_fastrtps_cpp_ServiceTypeSupport_impl_h_26_/
- http://ci.ros2.org/job/ci_windows_testing/2/testReport/rmw_fastrtps_cpp/cppcheck/error__syntaxError__include_rmw_fastrtps_cpp_TypeSupport_h_40_/
- http://ci.ros2.org/job/ci_windows_testing/2/testReport/rmw_fastrtps_cpp/cppcheck/error__syntaxError__include_rmw_fastrtps_cpp_TypeSupport_impl_h_35_/
- http://ci.ros2.org/job/ci_windows_testing/2/testReport/robot_state_publisher/cppcheck/error__syntaxError__include_robot_state_publisher_joint_state_listener_h_55_/
- http://ci.ros2.org/job/ci_windows_testing/2/testReport/robot_state_publisher/cppcheck/error__syntaxError__include_robot_state_publisher_robot_state_publisher_h_53_/
- http://ci.ros2.org/job/ci_windows_testing/2/testReport/urdf/cppcheck/error__syntaxError__include_urdf_model_h_51_/
- http://ci.ros2.org/job/ci_windows_testing/2/testReport/urdf/cppcheck/error__syntaxError__include_urdf_urdfdom_compatibility_h_89_/

All of these have the root cause of cppcheck assuming `.h` means C and not C++.

So my question to @ros2/team is what should we do:

- somehow indicate to cppcheck that these files are C++ (on a per file or per package basis
- or globally have it interpret `.h` files as C++?

The latter is easier, but also less precise because it will check out pure C `.h` files as C++ and may miss things.

I'm inclined to go with the latter despite that, since I'm not aware of any existing reason checking our C files as C++ would be an issue.

mikaelarguedas, 2017-06-21:
Maybe that doesnt make sense but I'd like to offer an in-between:
- renaming the files in the ros2 core that don't comply with this (looks like it's only `rmw_fastrtps_cpp` and that would make it consistent with its connext sibling).
- globally have cppcheck interpret .h files as c++ allowing us not to rename files in forked repos

wjwwood, 2017-06-21:
@mikaelarguedas are those alternatives of each other or are you suggesting we do both?

Seems to me that your first bullet is the alternative to my two options, but the second bullet is the same as my second bullet and I don't think would make sense if you did your first bullet. 😕

wjwwood, 2017-06-21:
Also this affects `robot_state_publisher` as well as `urdf_model.h` and `urdf_model_compatibility.h`. Though I guess we could fix `rmw_fastrtps` and turn off cppcheck for those other packages/headers.

dirk-thomas, 2017-06-21:
Where possible I would prefer to let `cppcheck` continue to check the file based on its extension. For the cases where that doesn't work because we don't control the sources we can either:
* disable `cppcheck` or
* add a configuration option to override file type (`cppcheck` has the option `--std=<id>` for this)

mikaelarguedas, 2017-06-21:
Yeah sorry that was unclear. My point was I think it's worth renaming the files for the code we didn't fork. And for the ones we forked / don't want to divege, tell cppcheck (on a per package basis) to interpret the .h files as c++ headers. If that's what the `--std=<id>` does :+1:

wjwwood, 2017-06-21:
Well the next question is "how do we tell cppcheck to handle `.h` files differently for one package?"

I don't know how to do this, since the test is registered automatically. @dirk-thomas what's the best way for the package to indicate it wants this alternative behavior?

dirk-thomas, 2017-06-21:
The [CMake](https://github.com/ament/ament_lint/blob/767257ff141d7957f3c8c7b92106aced95794537/ament_cmake_cppcheck/cmake/ament_cppcheck.cmake#L25) function would get a keyword argument like `STD` (similar to what other linters have).

Since the function is called by the linter hook (rather than explicitly within the package) I would suggest to use a variable like `CPPCHECK_STD` which can be set in the CMake file of a package. The CMake function should use the value of the global variable when the keyword argument is not passed explicitly. The global variable should be mentioned in the docblock of the CMake function (beside the keyword arg so that users find it "easily"). Does that sound reasonable?

allenh1, 2017-06-21:
Should we consider moving `*.h` files to `*.hpp` files in the case of a C++ app? I know this is not going to be favored, but, since we have `*.hpp` files being generated for our messages, and since anybody porting things already has to change the include extensions for the C++ messages to be `*.hpp`, maybe we should just be consistent all the way around?

wjwwood, 2017-06-21:
> Does that sound reasonable?

Yes, I'll try to implement that now.

> Should we consider moving *.h files to *.hpp files in the case of a C++ app? I know this is not going to be favored, but, since we have *.hpp files being generated for our messages, and since [...]

@mikaelarguedas suggested we do that in https://github.com/ros2/ros2/issues/368#issuecomment-309936415, but I think he meant just `rmw_fastrtps` because we've tried to avoid unnecessary changes in packages we ported for our turtlebot demo. However, I would be fine with changing the extensions, but it's also probably good for our cppcheck linter if it can take an option like this. That will make it more flexible for other people who might want to use it and don't use `.hpp` as their style. So I'll probably just do that to fix this issue. I'll probably still propose `rmw_fastrtps` change to `.hpp` though.

wjwwood, 2017-06-22:
One concern I have about the proposed approach (after starting to implement it) is that running `ament_cppcheck` locally will not reflect this setting. This is pretty annoying (in my opinion) as this is the main way I check if packages are ok or not before getting ready to commit. At the very least I find it more convenient to iterate with the command locally. If I notice the test fails, then I'll use the command line interface to see that its fixed rather than trying to narrow the tests which have complex arguments or by running all the tests again.

I could imagine a config file with a "well known" name could influence the linter. That would work for both the automated test and the local command line tool.

I probably won't do that right now, in the interest of time, but I think it would be worth making an issue and following up on later.

wjwwood, 2017-06-22:
CI with the rest of the pr's:

- Linux:
  - [image]
- Linux-aarch64:
  - [image]
- macOS:
  - [image]
- Windows:
  - [image]

I forced the windows job onto eatable and confirmed it has the new version of cppcheck.

wjwwood, 2017-06-22:
I think the failed tests on windows are unique to eatable based on comparison with the latest CI job from icecube:

http://ci.ros2.org/job/ci_windows_testing/3/testReport/

So I'm going to merge this, look into the issues with these tests specifically on eatable (I guess there's something wrong with eatable's opencv), and deal with fallout tomorrow.

closed 2017-06-22 by wjwwood

mikaelarguedas, 2017-06-22:
I added `C:\opencv\build\x64\vc14\bin` to the `PATH` that should fix it. You may need a reboot or at least a logout for it to take effect

wjwwood, 2017-06-22:
Was it missing? Thanks I was about to look into it.

I'll reboot and run some jobs to make sure.

mikaelarguedas, 2017-06-22:
:+1: 
Yeah the OpenCV_DIR variable was defined but the directory were the dll are was not on the PATH (http://docs.opencv.org/2.4/doc/tutorials/introduction/windows_install/windows_install.html#windowssetpathandenviromentvariable).

wjwwood, 2017-06-22:
Here's a CI job on eatable (no connext for speed):

- [image]

wjwwood, 2017-06-22:
Looks good, thanks @mikaelarguedas!