ros2/ros2 #369: rclcpp::shutdown() does not free all resources

mikoepf, 2017-06-22:
## Bug report
It seems that the function "_rclcpp::shutdown()_" **_does not_** destroy all global resources created when the initialization function was originally called, so that anew initioalization is not possible until the whole application has been restarted. I would appreciate any help and suggestions!
Thank you in advance!

- Operating System:
  Windows 10
- Installation type:
  from source
- Version or commit hash:
  --
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclcpp

#### Expected behavior
After releasing all the global resources needed by the middleware and the client library with "_rclcpp::shutdown()_", afresh initialization with "_rclcpp::init(argc, argv);_" should be possible.
#### Actual behavior
After releasing all the global resources with "_rclcpp::shutdown()_", afresh initialization with "_rclcpp::init(argc, argv)_;" throws following error: "_failed to initialize rmw implementation: rcl_init called while already initialized, at C:\dev\ros2\src\ros2\rcl\rcl\src\rcl\rcl.c:69_". Afterwards i have to restart the whole program in order to initialize the rmw implementation.

dirk-thomas, 2017-06-22:
I think currently the [rclcpp::shutdown()](https://github.com/ros2/rclcpp/blob/b8d72d682a9bd345e32df7395b603873b11060dd/rclcpp/src/rclcpp/utilities.cpp#L161-L183) function doesn't call the underlying `rcl_shutdown()` function. Can you please try to invoke `rcl_shutdown()` explicitly before trying to call `rclcpp::init(...)` a second time.

mikoepf, 2017-06-23:
Hello Thomas, thank you for the quick response!
I tried to invoke `rcl_shutdown()` before calling `rclcpp::init(...)`  and there was no error this time, but the node from the first run seemed to be still partially alive and so i couldn´t spin the new node with `rclcpp::spin(node);`. Also regardless of whether `rcl_shutdown()` is invoked before or after `rclcpp::shutdown()`, i still couldn´t spin the new node.
If i use `rcl_shutdown()` instead of  `rclcpp::shutdown()` at the and of the code, the node does not stop to spin.

wjwwood, 2017-07-17:
Sorry for the long delay before responding. I think we'll have to audit the init/shutdown lifecycle in rclcpp before this will work. I believe just using `rcl_init` and `rcl_shutdown` should work, but those were made after `rclcpp` was originally written...

If anyone has time to investigate it then a pull request would be appreciated.

x-integrator, 2018-07-24:
The problem still persists in July 2018, after `rclcpp::shutdown()` is not possible to start it again. With solution using of `rcl_shutdown()` before `rclcpp::init` the program will not crash as before, but also is not receiving any messages. To test it, simply run content of you test-main in the loop, first time works, second time no any messages are received.

To solve issue with "still partially alive" I used C++ class with instantiation via new and after delete it will be stopped and removed from node list, at least this is working.
My problem is that all my stuff is in a dll and then in remains in memory all the time.

mlanting, 2019-08-06:
Is this still an issue? I've tried to re-create the issue, and I think I'm seeing the same thing @x-integrator  describes: I don't get an error message, but I also don't receive messages on a node after calling shutdown and init unless I re-create the node after the second init. Is this the expected behavior, or should a node created after calling "init" a first time start working again after a shutdown/init cycle?

wjwwood, 2019-08-06:
You can do repeated `rclcpp::init()`/`rclcpp::shutdown()` calls as of these prs (some of the set):

- https://github.com/ros2/rclcpp/pull/587
- https://github.com/ros2/rclpy/pull/249

See also, tests like these:

- https://github.com/ros2/rclcpp/blob/dc3c36c7f05c06f504041a2ab2c8df1160438607/rclcpp/test/test_utilities.cpp#L57-L83
- https://github.com/ros2/rclpy/blob/0bb0a48a3a425c473e1e81a50e35251169a05ffd/rclpy/test/test_init_shutdown.py#L36-L48

I'll close this due to the above.

---

> but I also don't receive messages on a node after calling shutdown and init unless I re-create the node after the second init. Is this the expected behavior, or should a node created after calling "init" a first time start working again after a shutdown/init cycle?

That is the behavior I would expect. The documentation for Node's (at the C API level) and other entities says that it should only continue to work so long as the init they were created after has not been shutdown:

https://github.com/ros2/rcl/blob/6f20cd9480ad1f7d47e716df8e50b7806dadd8ab/rcl/include/rcl/node.h#L179-L182

However, there are some cases where we may allow things to work post shutdown in order to cleanup and notify the system of the shutdown (flush log messages to topic, etc...). So some cases may throw with "context is not valid" (the context is the object created by `init()`) or some may fail more gracefully with a warning log message or something like that. We're still working out the behaviors there, e.g.: https://github.com/ros2/rclcpp/issues/812

closed 2019-08-06 by wjwwood