ros2/ros2 #370: setup documentation for beta 2

wjwwood, 2017-06-24:
This is an aggregate issue to collect things that need to be done for the docs in beta 2.

- [ ] move the existing docs from S3 to github page (@nuclearsandwich's suggestion, but I agree)
- get clean documentation generation from packages which already have docs from beta 1:
  - [ ] rcutils (new but in the same basket imo)
  - [ ] rmw
  - [ ] rcl
  - [ ] rclcpp
- [ ] figure out how to run the new documentation script, but integrate the custom doxygen output from the packages listed above

I'll keep updating this as I do things.

wjwwood, 2017-06-24:
I fixed up the docs on all the repos that I think have substantial docs. They're in review if anyone has time.

nuclearsandwich, 2017-07-05:
We didn't end up getting any of the new documentation scripts wired up and instead built documentation manually. Docs are live at https://docs.ros2.org/beta2

closed 2017-07-05 by nuclearsandwich