ros2/ros2 #371: rmw_get_node_names should provide more up-to-date information

dirk-thomas, 2017-06-26:
Currently `rmw_get_node_names` also returns nodes which have already been terminated. This is currently the case for both rmw impl. `rmw_fastrtps_cpp` as well as `rmw_connext_cpp`.

Maybe there is a way to "ensure" that nodes which are being terminated try to let other participants know being terminated. This also depends on the long term plan to tie node names to participants or now...

dhood, 2017-09-15:
this might have been caused by the way we were terminating nodes without giving them a chance to shutdown cleanly, which has been resolved in https://github.com/ros2/ros2cli/pull/50.

Topics and nodes can still be listed after shutdown if nodes don't shutdown cleanly (e.g. sigkill) but I believe the initial ticket was referring to "general use", right @dirk-thomas ?

dirk-thomas, 2017-09-15:
I only tried it with FastRTPS recently but for `demo_nodes_cpp` / `demo_nodes_py` `talker` / `listener` I can't reproduce the problem anymore. After stopping either of the sides the node is not listed anymore. And after stopping both the topic is gone too.

dhood, 2017-09-26:
That is the behaviour that I saw as well, provided you don't sigkill, so it seems this is good to be closed now

closed 2017-09-26 by dirk-thomas