ros2/ros2 #372: Errors building overlay using packaged archives

dhood, 2017-06-28:
I am having issues building "demos" in an overlay workspace using ament+ros2 from the OS X archive.

```
11:28:09 dhood /Users/dhood/Downloads/ros2-osx-june27-2017
$ source setup.bash
$ cd ~/ros2_testing_overlay/
$ mkdir src && cd src
$ git clone https://github.com/ros2/demos
Cloning into 'demos'...
remote: Counting objects: 1926, done.
remote: Compressing objects: 100% (107/107), done.
remote: Total 1926 (delta 56), reused 76 (delta 28), pack-reused 1789
Receiving objects: 100% (1926/1926), 1.34 MiB | 1.11 MiB/s, done.
Resolving deltas: 100% (1082/1082), done.
Checking connectivity... done.
$ cd ..
$ ament build --isolated
# Topological order
 - composition
 - demo_nodes_cpp
 - demo_nodes_py
 - dummy_map_server
 - dummy_robot_bringup
 - dummy_sensors
 - image_tools
 - intra_process_demo
 - lifecycle
 - pendulum_msgs
 - pendulum_control
 - topic_monitor

Process package 'composition' with context:
--------------------------------------------------------------------------------
 source_space => /Users/dhood/ros2_testing_overlay/src/demos/composition
  build_space => /Users/dhood/ros2_testing_overlay/build_isolated/composition
install_space => /Users/dhood/ros2_testing_overlay/install_isolated/composition
   make_flags => -j8, -l8
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'composition'
Running cmake because arguments have changed.
==> '. /Users/dhood/ros2_testing_overlay/build_isolated/composition/cmake__build.sh && /usr/local/bin/cmake /Users/dhood/ros2_testing_overlay/src/demos/composition -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/Users/dhood/ros2_testing_overlay/install_isolated/composition -G Unix Makefiles' in '/Users/dhood/ros2_testing_overlay/build_isolated/composition'
-- The C compiler identification is AppleClang 7.0.2.7000181
-- The CXX compiler identification is AppleClang 7.0.2.7000181
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.0 (/Users/dhood/Downloads/ros2-osx-june27-2017/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.6.1", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/local/bin/python3
-- Found ament_index_cpp: 0.0.0 (/Users/dhood/Downloads/ros2-osx-june27-2017/share/ament_index_cpp/cmake)
-- Found class_loader: 1.0.0 (/Users/dhood/Downloads/ros2-osx-june27-2017/share/class_loader/cmake)
CMake Error at /Users/dhood/Downloads/ros2-osx-june27-2017/lib/console_bridge/cmake/console_bridge-config.cmake:15 (message):
  Library 'console_bridge' in package console_bridge is not installed
  properly
Call Stack (most recent call first):
  /Users/dhood/Downloads/ros2-osx-june27-2017/share/class_loader/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /Users/dhood/Downloads/ros2-osx-june27-2017/share/class_loader/cmake/class_loaderConfig.cmake:30 (include)
  CMakeLists.txt:16 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/dhood/ros2_testing_overlay/build_isolated/composition/CMakeFiles/CMakeOutput.log".

<== Command '. /Users/dhood/ros2_testing_overlay/build_isolated/composition/cmake__build.sh && /usr/local/bin/cmake /Users/dhood/ros2_testing_overlay/src/demos/composition -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/Users/dhood/ros2_testing_overlay/install_isolated/composition -G 'Unix Makefiles'' failed in '/Users/dhood/ros2_testing_overlay/build_isolated/composition' with exit code '1'
<== Command '. /Users/dhood/ros2_testing_overlay/build_isolated/composition/cmake__build.sh && /usr/local/bin/cmake /Users/dhood/ros2_testing_overlay/src/demos/composition -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/Users/dhood/ros2_testing_overlay/install_isolated/composition -G 'Unix Makefiles'' failed in '/Users/dhood/ros2_testing_overlay/build_isolated/composition' with exit code '1'
```

```
$ cd ~/Downloads/ros2-osx-june27-2017/
$ grep -r "jenkins" . | grep pkgconfig
./lib/pkgconfig/console_bridge.pc:prefix=/Users/osrf/jenkins/workspace/packaging_osx/ws/install
./lib/pkgconfig/console_bridge.pc:libdir=/Users/osrf/jenkins/workspace/packaging_osx/ws/install/lib
./lib/pkgconfig/console_bridge.pc:includedir=/Users/osrf/jenkins/workspace/packaging_osx/ws/install/include
./lib/pkgconfig/orocos-kdl.pc:prefix=/Users/osrf/jenkins/workspace/packaging_osx/ws/install
./lib/pkgconfig/orocos_kdl.pc:prefix=/Users/osrf/jenkins/workspace/packaging_osx/ws/install
./lib/pkgconfig/urdfdom.pc:prefix=/Users/osrf/jenkins/workspace/packaging_osx/ws/install
./lib/pkgconfig/urdfdom_headers.pc:prefix=/Users/osrf/jenkins/workspace/packaging_osx/ws/install
```

Those are ROS 1 ports, so maybe they need to be modified to export dependencies in a portable way? They're not using ament, but neither is fastrtps, so I assume this can be done in pure cmake somehow.

mikaelarguedas, 2017-06-28:
That's a known console_bridge issue that affects all platforms. see https://github.com/ros/console_bridge/pull/45

dhood, 2017-06-28:
duplicate of https://github.com/ros2/ros2/issues/310

closed 2017-06-28 by dhood