ros2/ros2 #374: ros2:command not find

Yingzhong262526, 2017-06-29:
Hi:
  Everyone!When I want to learn  [the demo](https://github.com/ros2/ros2/wiki/tf2),I run 
ros2 run tf2_ros static_transform_publisher 1 2 3 0.5 0.1 -1.0 foo bar 
before I call the enviroment using 
~ ./ros2_ws/install/setup.bash
in the ubuntu 16.04 terminal ,but there is a error:
ros2:command not find
can someone help me ?

clalancette, 2017-06-29:
What version of ROS2 are you using?  The 'ros2' command is new with the not-yet-released beta2; you'll need to track master to get it.

Yingzhong262526, 2017-06-30:
@clalancette: 
Thank you !I do as you,Now it can find the "ros2"command!

dirk-thomas, 2017-06-30:
@Yingzhong262526 If the problem is solved for you can you please close the ticket. Thanks.

closed 2017-06-30 by Yingzhong262526