ros2/ros2 #376: Crash when call rclcpp::Node::create_client("service_name")

mc-android-developer, 2017-07-07:
- Operating System:
Linux Mint = Ubuntu 16.04
- Installation type:
  from source 
- Version or commit hash:
  - beta 1;  version: 148df7e841b91da488ec3bb4abd295a2bccdb728
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - boost, restcpp

#### Steps to reproduce issue
Could not find exact conditions. It happens consistently when I do multiple calls to rclcpp::Node::create_client("service_name")

#### Expected behavior
No crash

#### Actual behavior
SIGSEGV crash happens

(gdb) bt
#0  0x00007ffff518dcb2 in std::__cxx11::list<CustomClientResponse, std::allocator<CustomClientResponse> >::empty() const () from /home/mc/ros2_ws/install/lib/librmw_fastrtps_cpp.so
#1  0x00007ffff518add8 in ClientListener::hasData() () from /home/mc/ros2_ws/install/lib/librmw_fastrtps_cpp.so
#2  0x00007ffff5183eb4 in rmw_wait () from /home/mc/ros2_ws/install/lib/librmw_fastrtps_cpp.so
#3  0x00007ffff77b1111 in rcl_wait () from /home/mc/ros2_ws/install/lib/librcl.so
#4  0x00007ffff7477512 in rclcpp::executor::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#5  0x00007ffff7478755 in rclcpp::executor::Executor::get_next_executable(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#6  0x00007ffff747fc57 in rclcpp::executors::single_threaded_executor::SingleThreadedExecutor::spin() () from /home/mc/ros2_ws/install/lib/librclcpp.so
#7  0x00007ffff747c6b4 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#8  0x00007ffff747c7d2 in rclcpp::spin(std::shared_ptr<rclcpp::node::Node>) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#9  0x000000000046d428 in main (argc=3, argv=0x7fffffffdbb8) at /home/mc/skynet-edge/core_ws/src/skynet_orchestrator/src/orchestrator.cpp:15

#### Additional information
There are no mentioning of my code in the backtrace, however I determined for sure that crash happen when I call method create_client(..) in rclcpp::Node
<pre>
auto _clientGetState = create_client<lifecycle_msgs::srv::GetState>("node1__get_state");
</pre>

This happens when the lifecycle node (named 'node1') does not exist and I want to check that it's not exist in code. So I create a get state client for this reason. I have many nodes (node1, node2, .., nodeN) and I want to check availability for each of them in the loop.

Usually create_client(..) works fine for first few nodes and then crashes. It crashes consistently but the number of nodes checked normally before crash is fluctuated. Sometimes it  crashes after 1st node, another time after 5th. (never seen the crash right away on 1st node)

mikaelarguedas, 2017-07-07:
@mc-android-developer Thanks for reporting. If possible, could you confirm that the behavior is the same with beta2 ? We now have deb packages that you can install ([instructions here](https://github.com/ros2/ros2/wiki/Installation)).

Also is it possible for you to provide a simple reproducible example of the failure ? for example by modifying the [client example](https://github.com/ros2/examples/blob/master/rclcpp/minimal_client/main.cpp).

Thanks!

dirk-thomas, 2017-07-07:
The `empty()` call in the `hasData()` function is not protected by the mutex (neither in beta 1 nor in beta 2). Please see ros2/rmw_fastrtps#132 for a proposed fix.

mc-android-developer, 2017-07-08:
Thank you for response.

I've compiled ros2-beta2 from sources. Then recompiled my code with ros2-beta2.
I've got exception instead of SIGSEGV now:

terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument

Thread 1 "skynet_orchestr" received signal SIGABRT, Aborted.
0x00007ffff53de428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff53de428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff53e002a in __GI_abort () at abort.c:89
#2  0x00007ffff5a1884d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff5a166b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff5a16701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff5a16919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5a3f7fe in std::__throw_system_error(int) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x0000000000478d69 in std::mutex::lock() ()
#8  0x000000000047a4ba in std::lock_guard<std::mutex>::lock_guard(std::mutex&) ()
#9  0x00007ffff4f6c8d8 in ClientListener::detachCondition() () from /home/mc/ros2_b2_ws/install/lib/librmw_fastrtps_cpp.so
#10 0x00007ffff4f62073 in rmw_wait () from /home/mc/ros2_b2_ws/install/lib/librmw_fastrtps_cpp.so
#11 0x00007ffff77a9baa in rcl_wait () from /home/mc/ros2_ws/install/lib/librcl.so
#12 0x00007ffff74653d8 in rclcpp::executor::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#13 0x00007ffff746661b in rclcpp::executor::Executor::get_next_executable(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#14 0x00007ffff746efb7 in rclcpp::executors::single_threaded_executor::SingleThreadedExecutor::spin() () from /home/mc/ros2_ws/install/lib/librclcpp.so
#15 0x00007ffff746a524 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#16 0x00007ffff746a642 in rclcpp::spin(std::shared_ptr<rclcpp::node::Node>) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#17 0x000000000046a8c8 in main ()
(gdb) 

The backtrace is similar though. Now I'm trying to fix it by catching the exception.

One question aside:
Previously with ros2-beta1 I used IDE for development. I was sourcing ros2 in terminal and started IDE from there. Everything including compilation and debugging worked fine in IDE.
Now when I've switched to ros2-beta2 I still can compile and debug in console. But if I run my IDE from the terminal (as I did before), the project fails on cmake configuration with the message:
<pre>
/home/mc/clion-2017.1/bin/cmake/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles"
-- Found ament_cmake: 0.0.2 (/home/mc/ros2_ws/install/share/ament_cmake/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Found rclcpp: 0.0.2 (/home/mc/ros2_ws/install/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.0.2 (/home/mc/ros2_ws/install/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rmw_fastrtps_cpp: 0.0.2 (/home/mc/ros2_ws/install/share/rmw_fastrtps_cpp/cmake)
-- Found fastrtps_cmake_module: 0.0.2 (/home/mc/ros2_ws/install/share/fastrtps_cmake_module/cmake)
CMake Error at /home/mc/clion-2017.1/bin/cmake/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find FastRTPS (missing: FastRTPS_INCLUDE_DIR FastRTPS_LIBRARIES)
Call Stack (most recent call first):
  /home/mc/clion-2017.1/bin/cmake/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /home/mc/ros2_ws/install/share/fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake:95 (find_package_handle_standard_args)
  /home/mc/ros2_ws/install/share/rmw_fastrtps_cpp/cmake/rmw_fastrtps_cpp-extras.cmake:20 (find_package)
  /home/mc/ros2_ws/install/share/rmw_fastrtps_cpp/cmake/rmw_fastrtps_cppConfig.cmake:30 (include)
  /home/mc/ros2_ws/install/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:73 (find_package)
  /home/mc/ros2_ws/install/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:30 (include)
  /home/mc/ros2_ws/install/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/mc/ros2_ws/install/share/rcl/cmake/rclConfig.cmake:30 (include)
  /home/mc/ros2_ws/install/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/mc/ros2_ws/install/share/rclcpp/cmake/rclcppConfig.cmake:30 (include)
  CMakeLists.txt:18 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/mc/skynet-edge/core_ws/src/skynet_orchestrator/cmake-build-debug/CMakeFiles/CMakeOutput.log".
</pre>

As far as it works fine in console I assume FastRTPS is in place. I tried to add FastRTPS_INCLUDE_DIR, FastRTPS_LIBRARIES to my cmake as message advises - no help. Can someone hint how to fix it to work in IDE?

dirk-thomas, 2017-07-08:
> Now I'm trying to fix it by catching the exception.

The locking of the mutex fails. That indicates that there is likely a threading problem. Just catching the exception won't be "enough". Are you using any threads in your use case? Can you maybe provide a [SSCCE](http://sscce.org/)?

> I tried to add FastRTPS_INCLUDE_DIR, FastRTPS_LIBRARIES to my cmake as message advises - no help.

The CMake module `fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake` tries to find the `fastcdr` and `fastrtps` packages. Interestingly [these two lines](https://github.com/ros2/rmw_fastrtps/blob/6695dcf223b5e29b6e32ecdfa8fb6006b30ed34e/fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake#L39-L40) pass successfully. Since it works when building from the console I suspect it must something related to the environment.

Can you please try to invoke `export CMAKE_PREFIX_PATH=$AMENT_PREFIX_PATH:$CMAKE_PREFIX_PATH` on the console before opening the IDE from there (I don't think this changed since beta 1 though).

If that doesn't lead anywhere can you please either share a list of all recursive the files in your `/home/mc/ros2_ws/install` as well as all your ROS related environment variables or describe a sequence of steps how we can try to reproduce the problem.

mc-android-developer, 2017-07-10:
After I sourced ROS2, var. $CMAKE_PREFIX_PATH was empty for me. 
Line: **export CMAKE_PREFIX_PATH=$AMENT_PREFIX_PATH:$CMAKE_PREFIX_PATH** helped
My IDE works fine now. Thank you!

SIGSEGV/SIАGABRT issue is not easy to isolate. I'm working on example now. Hopefully, I can make it today.

mc-android-developer, 2017-07-10:
I've pulled updated master branch:
```
$ git l
*   81a7835 2017-07-10 | Merge pull request #132 from ros2/missing_mutex_lock (HEAD -> master, origin/master, origin/HEAD) [Dirk Thomas]
|\  
| * 4be8c42 2017-07-10 | replace mutex lock with atomic boolean [Dirk Thomas]
| * 8ecf13f 2017-07-07 | add missing mutex lock [Dirk Thomas]
|/  
* 6695dcf 2017-06-30 | 0.0.2 (tag: release-latest, tag: release-beta2) [Dirk Thomas]
```
compiled it and recompiled my project based on the ROS2 master. The problem remained so far
```
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument

Thread 1 "skynet_orchestr" received signal SIGABRT, Aborted.
0x00007ffff53d9428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff53d9428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff53db02a in __GI_abort () at abort.c:89
#2  0x00007ffff5a1384d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff5a116b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff5a11701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff5a11919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5a3a7fe in std::__throw_system_error(int) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x0000000000478d79 in std::mutex::lock() ()
#8  0x000000000047a4ca in std::lock_guard<std::mutex>::lock_guard(std::mutex&) ()
#9  0x00007ffff4f67aa8 in ClientListener::detachCondition() () from /home/mc/ros2_master_ws/install/lib/librmw_fastrtps_cpp.so
#10 0x00007ffff4f5d0d3 in rmw_wait () from /home/mc/ros2_master_ws/install/lib/librmw_fastrtps_cpp.so
#11 0x00007ffff77a9baa in rcl_wait () from /home/mc/ros2_master_ws/install/lib/librcl.so
#12 0x00007ffff7463668 in rclcpp::executor::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_master_ws/install/lib/librclcpp.so
#13 0x00007ffff74648ab in rclcpp::executor::Executor::get_next_executable(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_master_ws/install/lib/librclcpp.so
#14 0x00007ffff746d247 in rclcpp::executors::single_threaded_executor::SingleThreadedExecutor::spin() () from /home/mc/ros2_master_ws/install/lib/librclcpp.so
#15 0x00007ffff74687b4 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /home/mc/ros2_master_ws/install/lib/librclcpp.so
#16 0x00007ffff74688d2 in rclcpp::spin(std::shared_ptr<rclcpp::node::Node>) () from /home/mc/ros2_master_ws/install/lib/librclcpp.so
#17 0x000000000046a8d8 in main ()

Inferior 1 [process 4709] will be killed.
```
Sample project hasn't completed the yet. Working on it.

dirk-thomas, 2017-07-10:
The fix only addressed the segfault in `ClientListener::hasData()`. The invalid mutex in the `ClientListener::detachCondition()` function is still a problem.

mc-android-developer, 2017-07-12:
I've finally completed simple SSCCE project illustrating the issue.
Here it is: https://drive.google.com/file/d/0B68BPOrWQeULdWRHaHRzc0Q5ZUE/view?usp=sharing

I was testing it with ROS2 master. It stabily crashes (with SIGABRT and BT posted above) for me in file testnode.cpp line 64. I put comments there.

mc-android-developer, 2017-07-14:
Does that sample project helped you to see the error?

mc-android-developer, 2017-07-18:
May I ask if there is any updates on this topic?

mikaelarguedas, 2017-09-02:
@mc-android-developer sorry for the lack of response on this.

I could compile and run your example after making a few changes:
-  remove references to restcpp no needeed by this example
- adding the missing queue include
- change service names to use `/` instead of `__`

I havent yet experienced any crash.
```
$ install_debug_isolated/testnode/bin/testnode 
[2017-09-02 11:44:32.133379] [0x00007fc040e03700] [debug]   Task handler started
Service timeout for node: node1__get_state
Node status: active
Service timeout for node: node3__get_state
Service timeout for node: node4__get_state
Service timeout for node: node5__get_state
Service timeout for node: node6__get_state
Service timeout for node: node7__get_state
Service timeout for node: node8/get_state
Service timeout for node: node9/get_state
Service timeout for node: node10/get_state
Service timeout for node: node11/get_state
Service timeout for node: node12/get_state
Service timeout for node: node13/get_state
```

I think that this issue has been fixed in https://github.com/ros2/rmw_fastrtps/pull/147.
Would you mind trying again with the latest state of master for all repos and report here with the result?

mikaelarguedas, 2017-09-12:
I'm going to close this because I can't reproduce it and believe this is fixed. feel free to comment here with any follow-up comment and we will reopen if needed

closed 2017-09-12 by mikaelarguedas