ros2/ros2 #377: Reverted wiki changes to deb instructions

wjwwood, 2017-07-10:
See: https://github.com/ros2/ros2/wiki/Linux-Install-Debians/_history

I reverted this at @tfoote's suggestion, since he said it might be confusing to users.

@christianrauch I opened this issue to let you know (there's no direct way to notify you on the wiki revert AFAIK). Maybe after discussing it with @tfoote we can restore the change.

Thanks!

wjwwood, 2017-07-10:
I'm going to close this, and reopen it if any discussion, or action items, come up.

closed 2017-07-10 by wjwwood

nuclearsandwich, 2017-07-11:
@christianrauch thanks for spotting the issue with the non-privileged io redirection. I've re-incorporated the `sh -c` portion of your change into the instruction with [this commit](https://github.com/ros2/ros2/wiki/Linux-Install-Debians/7590600916dc6a7c5844d9f0abea583141f4f401).

Is there a reason beyond specificity for wanting everyone who follows the guide to specify which architectures to retrieve package info for? In most existing ROS documentation that I've seen we let the system select the valid architectures.

christianrauch, 2017-07-11:
Flagging the repo with `[arch=amd64,arm64]` prevents error messages like `N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'http://repo.ros2.org/ubuntu/main xenial InRelease' doesn't support architecture 'i386'` for 64bit systems that also install 32bit packages.

See also: https://wiki.debian.org/Multiarch/HOWTO

nuclearsandwich, 2017-07-12:
Thanks for the clarification I did notice this warning message. I don't have strong feelings on whether or not it's worth adding to the line complexity to assuage it.

tfoote, 2017-07-12:
A basic test following the installation instructions don't cause that warning: https://gist.github.com/tfoote/46e32764a6ed86a841e53a7a8d6df603

I'd rather not lock it in as to the specific archs in case we change/extend our architecture support in the future. And if you know enough to turn on the multiarch support likely you'll know that this is not actually a problem. And can make the change to our sources if you want to suppress it.

We could resolve that warning by simply turning on i386 support in the repo, even if we leave it empty. But that only fixes it for multiarch i386 users. If they've added multiarch for armhf or armv8 cross compiling it won't suppress the warnings.