ros2/ros2 #378: Error compiling beta2 Raspberry Pi 3

abilbaotm, 2017-07-12:
Hi, I am trying to compile ROS2 beta2 directly on a Raspberry Pi 3 (no cross compilation) and I get an error during Fast-RTPS build. Thank you!

## Bug report

**Required Info:**

- Operating System:
  - Raspbian Debian 8.
- Installation type:
  - Source
- Version or commit hash:
  - release-beta2
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
# From a Raspberry Pi 3
$ mkdir -p ~/ros2_ws/src
$ cd ~/ros2_ws
$ wget https://raw.githubusercontent.com/ros2/ros2/release-beta2/ros2.repos
$ vcs-import src < ros2.repos

$ src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --cmake-args -DSECURITY=ON -- -j1

```

#### Expected behavior
Compile ROS2 with security enabled for Raspberry Pi 3 target machine.

#### Actual behavior

```
Process package 'fastrtps' with context:
--------------------------------------------------------------------------------
 source_space => /home/erle/ros2_ws/src/eProsima/Fast-RTPS
  build_space => /home/erle/ros2_ws/build/fastrtps
install_space => /home/erle/ros2_ws/install
   make_flags => -j1
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'fastrtps'
==> '. /home/erle/ros2_ws/build/fastrtps/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/erle/ros2_ws/build/fastrtps'
-- Configuring Fast RTPS
-- Version: 1.5.0
-- To change de version modify the file configure.ac
-- fastcdr library found...
-- Configuring done
-- Generating done
-- Build files have been written to: /home/erle/ros2_ws/build/fastrtps
==> '. /home/erle/ros2_ws/build/fastrtps/cmake__build.sh && /usr/bin/make -j1' in '/home/erle/ros2_ws/build/fastrtps'
Scanning dependencies of target fastrtps
[  0%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/log/Log.cpp.o
[  1%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/log/StdoutConsumer.cpp.o
[  2%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/eClock.cpp.o
[  3%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/IPFinder.cpp.o
[  4%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/md5.cpp.o
[  5%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/StringMatching.cpp.o
[  6%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/ResourceEvent.cpp.o
[  7%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/TimedEvent.cpp.o
[  8%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/TimedEventImpl.cpp.o
[  9%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/AsyncWriterThread.cpp.o
[ 10%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/AsyncInterestTree.cpp.o
[ 11%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/Endpoint.cpp.o
[ 12%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/RTPSWriter.cpp.o
[ 13%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/StatefulWriter.cpp.o
[ 14%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/ReaderProxy.cpp.o
[... 82 lines cut ...]
                                                       ^
/home/erle/ros2_ws/src/eProsima/Fast-RTPS/src/cpp/security/authentication/PKIDH.cpp: In function ‘X509* load_certificate(const std::vector<unsigned char>&)’:
/home/erle/ros2_ws/src/eProsima/Fast-RTPS/src/cpp/security/authentication/PKIDH.cpp:293:46: error: invalid conversion from ‘const void*’ to ‘void*’ [-fpermissive]
         BIO* cid = BIO_new_mem_buf(data.data(), static_cast<int>(data.size()));
                                              ^
/home/erle/ros2_ws/src/eProsima/Fast-RTPS/src/cpp/security/authentication/PKIDH.cpp: In function ‘bool check_sign_sha256(X509*, const unsigned char*, size_t, const std::vector<unsigned char>&, eprosima::fastrtps::rtps::security::SecurityException&)’:
/home/erle/ros2_ws/src/eProsima/Fast-RTPS/src/cpp/security/authentication/PKIDH.cpp:467:63: error: invalid conversion from ‘const unsigned char*’ to ‘unsigned char*’ [-fpermissive]
                 if(EVP_DigestVerifyFinal(&ctx, signature.data(), signature.size()) == 1)
                                                               ^
/home/erle/ros2_ws/src/eProsima/Fast-RTPS/src/cpp/security/authentication/PKIDH.cpp: In function ‘EVP_PKEY* generate_dh_key(int, eprosima::fastrtps::rtps::security::SecurityException&)’:
/home/erle/ros2_ws/src/eProsima/Fast-RTPS/src/cpp/security/authentication/PKIDH.cpp:603:38: error: ‘DH_get_2048_256’ was not declared in this scope
                 dh = DH_get_2048_256();
                                      ^
src/cpp/CMakeFiles/fastrtps.dir/build.make:2294: recipe for target 'src/cpp/CMakeFiles/fastrtps.dir/security/authentication/PKIDH.cpp.o' failed
make[2]: *** [src/cpp/CMakeFiles/fastrtps.dir/security/authentication/PKIDH.cpp.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'src/cpp/CMakeFiles/fastrtps.dir/all' failed
make[1]: *** [src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/erle/ros2_ws/build/fastrtps/cmake__build.sh && /usr/bin/make -j1' failed in '/home/erle/ros2_ws/build/fastrtps' with exit code '2'
<== Command '. /home/erle/ros2_ws/build/fastrtps/cmake__build.sh && /usr/bin/make -j1' failed in '/home/erle/ros2_ws/build/fastrtps' with exit code '2'
```
#### Additional information
No additional information provided.

mikaelarguedas, 2017-07-12:
Hi @abilbaotm , thanks for reporting.

It looks lilke an OpenSSL library issue. I'm not seeing the CMake configure step log in your console output, if you remove the fastrtps build folder and recompile it:
```
$ rm -r build/fastrtps
$ src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --cmake-args -DSECURITY=ON -- -j1 --only fastrtps
```

do you see a line like this ?
```
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libssl.so;/usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.0.2g") 
```

We are targetting openssl versions 1.0.2g and higher, I think that Debian Jessie ships with openssl 1.0.1t so it may be the source of the problem.

abilbaotm, 2017-07-13:
Hi @mikaelarguedas 

Seems that you are right, the default version of OpenSSL is shipped with raspbian is 1.0.1t.
```
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1t") 
```

```
 $ openssl version
OpenSSL 1.0.1t  3 May 2016
```

I will update the version of OpenSSL and I will report the results.

abilbaotm, 2017-07-17:
Hi @mikaelarguedas 

As you suggested @ https://github.com/ros2/ros2/issues/378#issuecomment-314828371 the issue is caused by the version of OpenSSL. I compiled OpenSSL 1.0.2g and installed on the RPi 3 and seems to be working. For future users here you have the commands that I used:
https://gist.github.com/abilbaotm/eaf12299c7a37d867713b5f42e820470

I think we can close this.
Regards.

closed 2017-07-17 by abilbaotm

mikaelarguedas, 2017-07-17:
Glad it worked out with a more recent OpenSSL. Thanks @abilbaotm for providing the steps into a gist.

YuehChuan, 2017-07-18:
Wow! great,  I tried gist on Raspberry pi3 but got error at the part ros1_bridge. 
It shows inner compilier erro.
It show `fail to load python  extension LZ4 `, but i already got  `ros-kinetic-roslz4` install.
The pi3 image was fresh install with turtlebot3 document
http://turtlebot3.readthedocs.io/en/latest/sbc_software.html

Mine python is 3.5.2  default and openSSL already 1.0.2g
    Operating System:
        UbuntuMATE
    Installation type:
        Source
    Version or commit hash:
        release-beta2
    DDS implementation:
        Fast-RTPS
    Client library (if applicable):
        N/A
[image]
[image]

allenh1, 2017-07-18:
That looks like you ran out of RAM/swap space. You might try formatting a USB drive, then using it for swap. If the device is `/dev/sdb`, you'd do

```
$ sudo mkswap /dev/sdb1
$ sudo swapon /dev/sdb1
```

YuehChuan, 2017-07-18:
@allenh1 I'll try it, TQ!

YuehChuan, 2017-08-13:
@allenh1 sorry for not update this thread these days.
@abilbaotm  did you successfully compile ros1_bridge on pi3 also?

I got 32GB SD card, so I make the third partition and add up swap space up to 4G.(ros1_brige need 4G)
Then native build. about 4hours
Most packages compile but ros1_bridge still got ICE issue.
ros1_bridge wiki suggest us to build it seperately with single thread.

so i follow these   https://github.com/ros2/ros1_bridge
first ignore ros1_bridge 
`src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --skip-packages ros1_bridge`
then
`src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --only ros1_bridge --force-cmake-configure --make-flags -j1
`
make with one thread, as troubleshooting part mentioned. 
https://github.com/ros2/ros2/wiki/Linux-Development-Setup
but still got the issue.

If I got luck to solve it will update here, as least other packages build, TQ!

abilbaotm, 2017-08-14:
Hi @YuehChuan 

Could you please give the console output? 
What do you mean with an `ICE issue`?

TIP: When I compiled ROS 2 beta2 on the raspberry Pi 3 I get some power issues (the Raspberry Pi 3 reboots during the compilation), I changed the power supply. Maybe reviewing the power supply that you are using for the Pi 3 can help. Also check that the Pi 3 is not overheating. 

Regards.

YuehChuan, 2017-08-15:
@abilbaotm  thank you, I will try it.

YuehChuan, 2017-08-18:
I got no luck with changing the power supply, will try to cross compile ros1_bridge

allenh1, 2017-08-18:
> I got no luck with changing the power supply, will try to cross compile ros1_bridge

I don't think that's quite necessary. What's the exact error you're getting?

YuehChuan, 2017-09-07:
@allenh1  sorry for reply this late.
It comes out the same issue although I add up swap space 2G
[image]

YuehChuan, 2017-09-16:
YAHOOOOOOOOOOOOOOO!  Got ROS2 r2b3 compile on raspberry pi3!!!
Thank you!!! @allenh1 and @abilbaotm 
@allenh1 , I tried your method plug in 8GB USB and make it as swap, this time also turn off GUI interface of ubuntuMATE.
[image]
I will post further details!

YuehChuan, 2017-09-16:
[image]

YuehChuan, 2017-09-16:
Here! 
https://hackmd.io/s/BysaB355-
@allenh1 @abilbaotm Thank you!