ros2/ros2 #379: Crash on repeated rclcpp_lifecycle::LifecycleNode::create_client(..) calls

mc-android-developer, 2017-07-19:
Operating System: Linux Mint = Ubuntu 16.04
Installation type: from source
Version or commit hash: master branch
DDS implementation: Fast-RTPS
Client library (if applicable): boost

**Steps to reproduce issue**
- I create a client for lifecycle ROS2 service by calling rclcpp_lifecycle::LifecycleNode::create_client(..)
- As a result I obtain rclcpp::client::Client instance.
- I call some methods on this instance and when it's destructed.
- After a while I create a new client for the same lifecycle ROS2 service by calling rclcpp_lifecycle::LifecycleNode::create_client(..) with the same parameters.

**SSCCE**
Sample project: https://drive.google.com/open?id=0B68BPOrWQeULdWRHaHRzc0Q5ZUE

**Expected behavior**
The second call should work fine and create rclcpp::client::Client instance like the first call.

**Actual behavior**
Crash on second call.
<pre>
test_node: tpp.c:84: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.

GNU gdb (GDB) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
No source file named /home/mc/skynet-edge/core_ws/src/skynet/src/skynet/SkynetNode.cpp.
No source file named /home/mc/skynet-edge/core_ws/src/skynet/src/skynet/SkynetNode.cpp.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff3fa0700 (LWP 16866)]
[New Thread 0x7ffff379f700 (LWP 16867)]
[New Thread 0x7ffff2f9e700 (LWP 16868)]
[New Thread 0x7ffff279d700 (LWP 16869)]
[New Thread 0x7ffff1f9c700 (LWP 16870)]
[New Thread 0x7ffff179b700 (LWP 16871)]
[New Thread 0x7fffe3d75700 (LWP 16872)]

Thread 1 "test_node" hit Breakpoint 3, activate (node=..., name="testnode", nmspace="") at /home/mc/skynet-edge/core_ws/src/skynet_node_sample/src/TestNode.cpp:48
48	    auto clientGetState = node.create_client<lifecycle_msgs::srv::GetState>(name + "/get_state");
[New Thread 0x7fffe3574700 (LWP 16881)]

Thread 1 "test_node" hit Breakpoint 3, activate (node=..., name="testnode", nmspace="") at /home/mc/skynet-edge/core_ws/src/skynet_node_sample/src/TestNode.cpp:48
48	    auto clientGetState = node.create_client<lifecycle_msgs::srv::GetState>(name + "/get_state");

Thread 8 "test_node" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe3d75700 (LWP 16872)]
0x00007ffff5f28095 in std::atomic<bool>::load(std::memory_order) const () from /home/mc/ros2_master_ws/install/lib/librmw_fastrtps_cpp.so
(gdb) bt
#0  0x00007ffff5f28095 in std::atomic<bool>::load(std::memory_order) const () from /home/mc/ros2_master_ws/install/lib/librmw_fastrtps_cpp.so
#1  0x00007ffff5f2db05 in ClientListener::hasData() () from /home/mc/ros2_master_ws/install/lib/librmw_fastrtps_cpp.so
#2  0x00007ffff5f23090 in rmw_wait () from /home/mc/ros2_master_ws/install/lib/librmw_fastrtps_cpp.so
#3  0x00007ffff77a9baa in rcl_wait () from /home/mc/ros2_ws/install/lib/librcl.so
#4  0x00007ffff7463668 in rclcpp::executor::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#5  0x00007ffff74648ab in rclcpp::executor::Executor::get_next_executable(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/mc/ros2_ws/install/lib/librclcpp.so
#6  0x00007ffff746d247 in rclcpp::executors::single_threaded_executor::SingleThreadedExecutor::spin() () from /home/mc/ros2_ws/install/lib/librclcpp.so
#7  0x000000000041b13b in <lambda()>::operator()(void) const (__closure=0xf1b728) at /home/mc/skynet-edge/core_ws/src/skynet_node_sample/src/TestNode.cpp:89
#8  0x000000000041c494 in std::_Bind_simple<main(int, char**)::<lambda()>()>::_M_invoke<>(std::_Index_tuple<>) (this=0xf1b728) at /usr/include/c++/5/functional:1531
#9  0x000000000041c3ea in std::_Bind_simple<main(int, char**)::<lambda()>()>::operator()(void) (this=0xf1b728) at /usr/include/c++/5/functional:1520
#10 0x000000000041c37a in std::thread::_Impl<std::_Bind_simple<main(int, char**)::<lambda()>()> >::_M_run(void) (this=0xf1b710) at /usr/include/c++/5/thread:115
#11 0x00007ffff6a02c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#12 0x00007ffff79b76ba in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#13 0x00007ffff64713dd in clone () from /lib/x86_64-linux-gnu/libc.so.6
</pre>

**Additional information**
Probably it's related to another issue I posted: https://github.com/ros2/ros2/issues/376
The reprosteps are different but backtraces are similar.

mikaelarguedas, 2017-09-02:
@mc-android-developer thanks for the report.
Analog to https://github.com/ros2/ros2/issues/376 I couldn't reproduce the issue using the provided example, I modified the example to query the same node every other time and can successfully spawn the client and get the status. Can you confirm if this is still an issue using the code currently on master?

Below console output:
```
Service timeout for node: node2/get_state
Node status: unconfigured
Service timeout for node: node3/get_state
Service timeout for node: node4/get_state
Node status: inactive
Node status: inactive
Service timeout for node: node5/get_state
Node status: inactive
Service timeout for node: node6/get_state
Node status: active
Service timeout for node: node7/get_state
Node status: active
Service timeout for node: node8/get_state
Node status: active
Service timeout for node: node9/get_state
Node status: active
Service timeout for node: node10/get_state
Node status: inactive
Service timeout for node: node11/get_state
Node status: inactive
Service timeout for node: node12/get_state
Node status: inactive
Service timeout for node: node13/get_state
```

mikaelarguedas, 2017-09-12:
I'm going to close this because I can't reproduce it and believe this is fixed. feel free to comment here with any follow-up comment and we will reopen if needed

closed 2017-09-12 by mikaelarguedas