ros2/ros2 #380: publishers should have a "has subscribers at present" property

BrannonKing, 2017-07-28:
It is handy to be able to avoid expensive computation for producing large messages in situations where no one is listening on the far end. I know this information is available on the node itself (through CountSubscribers("topic")), but it would be convenient to expose it on the publisher as well.

mikaelarguedas, 2017-08-02:
We are happy to take contributions for this

bailaC, 2017-09-06:
I would like to try to add this feature.

bailaC, 2017-09-06:
I would like to give it a try.

wjwwood, 2017-09-07:
@bailaC cool 😄.

This issue could really be split into two things:

- provide an API call on publisher's (and probably a corresponding one on the subscription) to get this information
- provide a better mechanism to check this

The first point is just asking the Node with something like `return node->count_subscribers(publisher->get_topic_name());`. There's some work to sort out how the publisher keeps a handle on the node and stuff like that, but it's relatively straightforward.

The second point is that when you have a publisher you can make a stronger check about the subscribers on that topic. At least right now `Node::count_subscribers()` returns the number of subscriptions to the given topic regardless of the type the subscriptions have asked for and regardless to the QoS settings they use. It's possible that `Node::count_subscribers()` returns something greater than zero but a publisher on that same topic may not be "matched" to it and therefore there is still no work to be done. It's possible for a publisher and a subscription to the same topic to not communicate due to mismatched types or due to mismatched QoS settings.

So a better function to have on a publisher would be something like `Publisher::count_matched_subscriptions()`, which would return the number of subscribers that we could possibly communicate with. DDS gives us this information, but the patch would be quite a bit more work, since you'd need to expose this through our entire stack. But it would also be a better contribution! 😉

Whichever way you decide to go, don't hesitate to ask questions, and we'll get to them asap.

bailaC, 2017-09-09:
@wjwwood Thanks a lot for your response.

I am actually setting up my ros2 development environment. I understand well the first solution you proposed.

For the later part, i am still at the learning phase of QoS and DDS features. I need some time in understanding those.

I thought of a solution, where during the subscription of a topic, if anything can be done at the publisher (of that same topic) side to update any parameter(of publisher), so that it will be updated instantly. I still need to figure out whether it is feasible or not (or even can be done).

N.B. : we can make this conversation over email (if it is fine with you), as i might need some more help and this trail would go much longer.

wjwwood, 2017-09-12:
@bailaC you may ask here or on discourse.ros.org (in the "Next Generation ROS" category), but I prefer a public method of communication, so others can benefit from your questions too.

bailaC, 2017-09-12:
@wjwwood Sure. Do you have any comment on my thought?

wjwwood, 2017-09-12:
> I thought of a solution, where during the subscription of a topic, if anything can be done at the publisher (of that same topic) side to update any parameter(of publisher), so that it will be updated instantly. I still need to figure out whether it is feasible or not (or even can be done).

I'm not sure I follow the suggestion, can you try to expand on it? In theory, yes we need to keep track of what subscriptions have been made to a topic so publishers can access this information, but that doesn't explain how it might be done in detail.

bailaC, 2017-09-12:
@wjwwood I am studying the flow. Will post once i get a clear view.

dirk-thomas, 2019-09-05:
By now there is an API to get the matched subscriber count for a publisher: https://github.com/ros2/rclcpp/blob/1fff8cbac14319b037ca1af8a9a0e19e9b2247f7/rclcpp/include/rclcpp/publisher_base.hpp#L127

Therefore closing this ticket.

closed 2019-09-05 by dirk-thomas