ros2/ros2 #381: Building 'ament_cppcheck'

Yingzhong262526, 2017-07-31:
Hi:
  Everyone,I install ros2 beta2 on the system of Win 10 as [the tutorials](https://github.com/ros2/ros2/wiki/Windows-Development-Setup),But when  I begin building it using the command:
>python src\ament\ament_tools\scripts\ament.py build
the errors come up :
'C:\Program' neigher a inner or  outside command, nor an executed program or batch file!
Traceback (most recent call last):
  File "src\ament\ament_tools\scripts\ament.py", line 158, in <module>
    sys.exit(main() or 0)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\commands\ament.py", line 89, in main
    rc = args.main(args)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\verbs\build\cli.py", line 172, in main
    return iterate_packages(opts, packages, per_package_main)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\verbs\build\cli.py", line 297, in iterate_packages
    rc = process_sequentially(jobs)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\verbs\build\cli.py", line 337, in process_sequentially
    rc = job['callback'](job['opts'])
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\verbs\build_pkg\cli.py", line 280, in main
    return run(opts, context)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\verbs\build_pkg\cli.py", line 331, in run
    handle_build_action(on_build_ret, context)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\verbs\build_pkg\cli.py", line 268, in handle_build_action
    for build_action in build_action_ret:
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\build_types\ament_python.py", line 48, in on_build
    self._update_context_with_setup_arguments(context)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\build_types\ament_python.py", line 410, in _update_context_with_setup_arguments
    AmentPythonBuildType.build_type, context)
  File "C:\dev\ros2\src\ament\ament_tools\ament_tools\setup_arguments.py", line 61, in get_setup_arguments_with_context
    ' '.join(cmd), stdout=subprocess.PIPE, shell=True, check=True)
  File "C:\Program Files\Python36\lib\subprocess.py", line 418, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command 'C:\dev\ros2\build\ament_cppcheck\ament_python__setup.bat C:\Program Files\Python36\python.exe -c "import sys;sys.path.insert(0, 'C:/dev/ros2/src/ament/ament_tools');from ament_tools.setup_arguments import get_setup_arguments;print(repr(get_setup_arguments('C:/dev/ros2/src/ament/ament_lint/ament_cppcheck/setup.py')))"' returned non-zero exit status 9009.

Someone can help me？Thanks at first!

wjwwood, 2017-07-31:
This error message:

```
'C:\Program' neigher a inner or  outside command, nor an executed program or batch file!
```

Doesn't appear anywhere in our code base that I can see. Are you sure it's not part of something you're doing?

Please provide the exact commands you ran after installing. This isn't enough information for us to help you.

gavanderhoorn, 2017-07-31:
I think it's actually an English translation of the Chinese version of the typical Windows error:

> 'X' is not recognized as an internal or external command, operable program or batch file.

Because of `C:\Program` in the error, it would appear that this is a problem with paths with spaces in them and quoting. The path to some executable being used is probably `C:\Program Files\..` and without proper quoting that gets truncated to `C:\Program`.

wjwwood, 2017-07-31:
> The path to some executable the python executable being used is C:\Program Files\Python36\python.exe and without proper quoting that gets truncated to C:\Program.

I agree, but without more detailed steps, it's hard to know why the whitespace is an issue. Perhaps it needs to be escaped when setting an environment variable or something. Or perhaps it is a general issue with `ament_cppcheck` when python is in a path with whitespace.

Yingzhong262526, 2017-08-01:
@wjwwood:
my exact commands is that 
>python src\ament\ament_tools\scripts\ament.py build
@gavanderhoorn:
Now I have a confusion：Maybe several days ago,I have installed the ros2 beta1 also on the Win10 system,but It can build normaly when I run the build command:
>python src\ament\ament_tools\scripts\ament.py build
the  python  executable also is the C:\Program Files\Python36\python.exe
Why? is the problem of the version of ros2 ?

Yingzhong262526, 2017-08-01:
The probelm actually is caused by the whitespace in the Python path  C:\Program Files\Python36\python.exe 
I replace it with C:\Progra~1\Python36\python.exe ,then It works !

Thanks above all!

closed 2017-08-01 by Yingzhong262526