ros2/ros2 #382: command:ros2 error

Yingzhong262526, 2017-08-02:
Hi:
when I have installed and built ros2 beta 2 on the sysyem of Win10 as the [tutorials](https://github.com/ros2/ros2/wiki/Windows-Development-Setup),then run the demo ：
**> call install\local_setup.bat
> ros2 run demo_nodes_py talker**
the errors occur:
Failed to load entry point 'start': DLL load failed: 找不到指定的模块。
Failed to load entry point 'status': DLL load failed: 找不到指定的模块。
Failed to load entry point 'stop': DLL load failed: 找不到指定的模块。
Failed to load entry point 'list': DLL load failed: 找不到指定的模块。
Failed to load entry point 'call': DLL load failed: 找不到指定的模块。
Failed to load entry point 'list': DLL load failed: 找不到指定的模块。
Failed to load entry point 'echo': DLL load failed: 找不到指定的模块。
Failed to load entry point 'list': DLL load failed: 找不到指定的模块。
Failed to load entry point 'pub': DLL load failed: 找不到指定的模块。
Traceback (most recent call last):
  File "C:\dev\ros2\install\lib\demo_nodes_py\talker-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.0.0', 'console_scripts', 'talker')()
  File "C:\Progra~1\Python36\lib\site-packages\pkg_resources\__init__.py", line 564, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:\Progra~1\Python36\lib\site-packages\pkg_resources\__init__.py", line 2662, in load_entry_point
    return ep.load()
  File "C:\Progra~1\Python36\lib\site-packages\pkg_resources\__init__.py", line 2316, in load
    return self.resolve()
  File "C:\Progra~1\Python36\lib\site-packages\pkg_resources\__init__.py", line 2322, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:\dev\ros2\install\Lib\site-packages\topics\talker.py", line 18, in <module>
    import rclpy
  File "C:\dev\ros2\install\Lib\site-packages\rclpy\__init__.py", line 18, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\dev\ros2\install\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = importlib.import_module('._rclpy', package='rclpy')
  File "C:\Progra~1\Python36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: 找不到指定的模块

I know that it is caused by the dll not correctly loading,but how can I solve it?

dirk-thomas, 2017-08-02:
Closing this as duplicate due to cross posting: https://discourse.ros.org/t/failed-to-load-entry-point-for-command-ros2/2363

closed 2017-08-02 by dirk-thomas

apourostad, 2017-08-27:
I have this problem too.
please help!
I spend about 1 week on it!
[image]
[image]
[image]

dirk-thomas, 2017-08-28:
@apourostad Please post your information as text rather than images since those are not searchable. Please also provide more information as requested in https://discourse.ros.org/t/failed-to-load-entry-point-for-command-ros2/2363/9

apourostad, 2017-08-28:
@dirk-thomas 
Thanks for answering.
I sent cmd image so that you can read the exact error
and the second one and third one so that you can see my environment variables 
so if I had some mistakes in it, you can tell me
my problem is simple:
I've done installation exactly from this link: https://github.com/ros2/ros2/wiki/Windows-Development-Setup
but I can't run examples and test and I've done all of things above in this post. but as you see in cmd screenshot, it fails to run ros2 and I don't know why!
Sorry for my bad English