ros2/ros2 #387: Tracking bug to fix memory leaks in test_graph.cpp

clalancette, 2017-08-10:
I found some memory leaks in ros2/rcl/rcl/test/rcl/test_graph.cpp using valgrind.  The sources of the leaks are spread across a few different repositories, so this is a tracking bug for all of the PRs related to the leaks.  When all PRs are merged, test_graph.cpp has only one small leak from thread creation, which I have not tracked down.  I'll update this bug as the PRs get filed/merged.

- [ ] Leaks in rcl (PR upcoming)
- [ ] Leaks in rcutils (PR upcoming)
- [ ] Minor string fix in rmw (PR upcoming)
- [ ] Leaks in rmw_fastrtps_cpp (PR upcoming)

clalancette, 2017-08-10:
I'm actually going to run another CI job, because there have been enough changes (between review and me finding another couple of leaks) that I think it is a good idea:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

clalancette, 2017-08-16:
All of these leaks fixed, closing.

closed 2017-08-16 by clalancette

dhood, 2017-08-22:
I want to comment here with some notes in case it's useful one day looking back. (sometimes I know I've seen an error but can't remember which job it was in and this seems like an appropriate place to comment -- don't mean for this to take away from your work @clalancette!).

On linux, the test used [to fail with](http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/718/testReport/(root)/projectroot/test_graph__rmw_fastrtps_cpp/)
```
*** Error in `/home/rosbuild/ci_scripts/ws/build/rcl/test/test_graph__rmw_fastrtps_cpp': double free or corruption (fasttop): 0x00007fe76c002300 ***
```

But it can still [fail with](http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/751/testReport/junit/(root)/projectroot/test_graph__rmw_fastrtps_cpp/)
```
*** Error in `/home/rosbuild/ci_scripts/ws/build/rcl/test/test_graph__rmw_fastrtps_cpp': free(): invalid pointer: 0x00007f4544002e60 ***
```

or [fail with](http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/752/testReport/junit/(root)/projectroot/test_graph__rmw_fastrtps_cpp/)
```
*** Error in `/home/rosbuild/ci_scripts/ws/build/rcl/test/test_graph__rmw_fastrtps_cpp': munmap_chunk(): invalid pointer: 0x00007fc1200027a0 ***
```

On OSX it [still fails with](http://ci.ros2.org/view/nightly/job/nightly_osx_repeated/793/testReport/junit/(root)/projectroot/test_graph__rmw_fastrtps_cpp/)
```
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
```

clalancette, 2017-08-22:
@dhood Ug, nasty.  Thanks for the heads up; I thought I had caught all of these, but clearly not!  I'll take another look later this week.  For now I'll leave this bug closed and if I find something new, I'll open a new bug for it.