ros2/ros2 #389: Update github labels

mikaelarguedas, 2017-08-11:
I updated the labels of all repos in ros2 and ament org using the script in that [gist](https://gist.github.com/mikaelarguedas/4e0b6722f245916f9cbb4ea13b54402c).

Note: Then only labels that have been propagated are the ones not commented out in `LABEL_DICT`.

If you have any suggestion for other labels or the color feel free to comment here

dirk-thomas, 2017-08-11:
Can you comment on how you picked the color values. Some of them look different than their default color. I would highly recommend to keep the color values of "default" tags since users are familiar with their default color.

mikaelarguedas, 2017-08-11:
hmm I took the ones that were from the "oldest" repos, I think all these colors are based on what was in the rcl repo.
Changes that I made on top:
- added an orange "hitlist" label.
- used the brown used for "wontfix" and applied it to "duplicate".
- modified the blue of enhancement to not collide with the blue used for "in review"
- modified "ready" so that it's not the same colour as all the default grayish other labels

As I said I'm open to suggestions

dirk-thomas, 2017-08-11:
I would recommend to switch back to the "default" colors for the "default" labels. I don't mind the color of "custom" labels.

mikaelarguedas, 2017-08-11:
Done!

mikaelarguedas, 2017-08-28:
Labels have been updated for all repos in the repos files. Feel free to comment here if you want to change/add/remove labels

mikaelarguedas, 2017-10-14:
@ros2/team I noticed that most of our issues are just labeled as `enhancement`: I'm considering adding 2 new labels to all repos to help differentiating the issues:
- documentation
- test

I also saw a `requires-change` label on some repos. Do we want this as well or rather use the "Request changes" feature from the github review UI?

Feedback on the colors you deem appropriate is also welcome

clalancette, 2017-10-17:
Looking at, for instance, the https://github.com/ros2/rosidl/labels repository, there are currently 16 labels.  I think there are some there are aren't that useful, and some that are missing.  My view of the ones that we really need are:

* bug
* enhancement
* documentation
* test
* help wanted
* in progress
* in review
* more-information-needed

Most of the rest either don't seem that useful to me, or seem redundant.

In terms of the `requires-changes` label, I slightly prefer adding a label with a color, although I'm also fine with doing "Request Changes" in review.

mikaelarguedas, 2017-10-17:
> some that are missing

ATM the only labels propagated to all the repos via the script are in this [dict](https://gist.github.com/mikaelarguedas/4e0b6722f245916f9cbb4ea13b54402c#file-update_labels_github-py-L10-L27). This proposal is to extend this dict.

For the labels we can find on various ros2 repos today, maybe we should add a bit of context:
- `bug, duplicate, enhancement, help wanted, invalid, question, wontfix` etc:
 are github default labels that we decided to keep as is on all repos
- `claimed/unclaimed`:
Was introduced when we wanted people to easily find what issues were available for contribution and not already being worked on. Apparently the contribution page has been [updated recently](https://github.com/ros2/ros2/wiki/Contributing/_compare/ff185c06c79f9a79c5ee7d4971a6705d4b95a1e4...0239566b53809ae621f33ad812bb89d40b057ea7) to refer to the "help wanted" label (and not the "unclaimed" one) regardless of if someone is already working on them or not. So if we're fine with that we could just remove these 2 labels from all the repos in favor of help wanted.
- the `language/*` and the `experience/*` labels were another way for people to know at a glance what "knowledge" was necessary to work on a specific issue. We usually don't go through the hassle of adding them when opening issues so they're not used much and could arguably be removed too.
- `hitlist` we recently added it because we didnt want to create several priority labels. I think it's still useful if we're looking for a ticket to grab to find which ones are the most "critical" ones but could be convinced otherwise
- `require-changes`: I'm fine with either given that both are easy to identify (colorful) on waffle. the label approach is offering a more eye catching way of seeing it on the github UI. But that means that every time we request change on a PR we'll need to remember to add the label accordingly.

My proposal was to add `test` and `documentation` to segregate all the `enhancement`s into more granular categories

mikaelarguedas, 2017-11-16:
@ros2/team Looking for feedback on:

How should we represent that a PR requires changes:
- require-changes:
1. Using the require changes features of github review: what we've done so far but is not as noticeable through github UI. Displayed on Waffle in a colorful way
1. Using a `require-changes` label: Offers a more eye catching way of seeing it on the github UI. But that means that every time we request change on a PR we'll need to remember to add the label accordingly.

How we can make the "enhancement" labeled issues more informative.
- test and documentation:
The motivation here is to provide more information at a glance of what the issue is targeting (as the enhancement label doesn't carry much information). That will be useful for us as well as external contributors to know what work / knowledge is associated with the issue.

dirk-thomas, 2017-11-16:
Since we already use a ``requires-changes` label in some ROS 2 repos (e.g. https://github.com/ros2/rmw/labels/requires-changes) as well as on several ROS 1 repos I would suggest to stick to that label.

wjwwood, 2017-11-16:
I don't have an opinion about using/changing `requires-changes`. I use a similar label on some of my ROS 1 repositories, but mostly for my own book keeping.

+1 to the test and documentation labels as-is. I don't see them as exhaustive or required, but they are probably useful where ever they are applied. I don't think we need to change our behavior to them -- I don't really see them as any different than adding a `[doc] ` or `[test] ` to the beginning of the title, they're just more visible.

mikaelarguedas, 2017-11-16:
> Since we already use a `requires-changes` label in some ROS 2 repos (e.g. https://github.com/ros2/rmw/labels/requires-changes) as well as on several ROS 1 repos I would suggest to stick to that label.

Up until recently we didn't have any ROS 2 repo with these labels and they have been silently introduced in some repos without discussion or propagation. Hence the question of "is this the approach we want to take?" If we agree that the overhead of adding the label when reviewing is not overweighing advantage of providing additional visibility I'll propagate it to all repos.

> +1 to the test and documentation labels as-is. I don't see them as exhaustive or required, but they are probably useful where ever they are applied. I don't think we need to change our behavior to [...]

:+1:

mikaelarguedas, 2018-03-08:
The gist has been updated to support milestones. I'm going to close this again but will comment here next time we reevaluate the labels we use

closed 2018-03-08 by mikaelarguedas