ros2/ros2 #390: Is the windows build of ROS2 64bit or 32bit ?

firesurfer, 2017-08-15:
As the header says. Is the windows build by default 64bit or 32bit?

mikaelarguedas, 2017-08-15:
We build everything on 64-bit machines. While some libraries may be the same, I would expect a rebuild to be required for it to work on 32bit machines

firesurfer, 2017-08-15:
Do you officially support x86 ? It's just that I need to know whether I can force x64 builds for my C# wrapper or not.

mikaelarguedas, 2017-08-15:
Given that we never build or test on x86 I would say we don't officially support it. Though, while not being officially supported it should be possible for users to build for it.

Ideally the 32 vs 64 would be decided at compile time based on the architecture of the machine you're building. I think that defaulting to/providing binaries for x64 is great. If the users can build them themselves for x86 that's a plus but not a requirement IMO

firesurfer, 2017-08-15:
Thanks for the information!

closed 2017-08-15 by firesurfer