ros2/ros2 #391: OSX Binary Install, Library not loaded: @rpath/librcl_interfaces___rosidl_typesupport_c.dylib

kengregson, 2017-08-18:
- Operating System: OSX 10.12.6
- Installation type: binary
- Version or commit hash: ros2-beta2-package-osx-fastrtps.tar.bz2
- DDS implementation: Fast-RTPS (built-in)
- Client library (if applicable): rclcpp and rclpy

Running demos as described in binary installation instructions (talker/listener, cpp/py), eg:
> ros2 run demo_nodes_cpp talker

Should result in echoing Sent/Heard console output with increasing sequence numbers

Actually results in many errors (where actual install location has been elided):
[error.txt](https://github.com/ros2/ros2/files/1235056/error.txt)

For reference $DYLD_LIBRARY_PATH=.../ros2_install/ros2-osx/lib, and
.../ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so exists

Output from -L _rclpy.cpython-36m-darwin.so is:
_rclpy.cpython-36m-darwin.so:
	@rpath/_rclpy.cpython-36m-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/python3/Frameworks/Python.framework/Versions/3.6/Python (compatibility version 3.6.0, current version 3.6.0)
	@rpath/librcl_interfaces__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	libfastrtps.1.dylib (compatibility version 1.0.0, current version 1.5.0)
	libfastcdr.1.dylib (compatibility version 1.0.0, current version 1.0.7)
	@rpath/librmw.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librmw_fastrtps_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.60.2)
	@rpath/librosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcutils.dylib (compatibility version 0.0.0, current version 0.0.0)

Get past that by directly adding @rpath lib location into _rclpy.cpython-36m-darwin.so with:
install_name_tool -add_rpath .../ros2_install/ros2-osx/lib _rclpy.cpython-36m-darwin.so

But similar (unresolved) problems with other libraries which don't appear to have been rellinked:
Failed to load entry point 'pub': dlopen(.../ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: libfastrtps.1.dylib
  Referenced from: .../ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
dyld: Library not loaded: @rpath/libexample_interfaces__rosidl_typesupport_c.dylib
  Referenced from: .../ros2_install/ros2-osx/lib/demo_nodes_cpp/talker
  Reason: image not found

dhood, 2017-08-18:
It might be that your python version is not 3.6, could you provide the output of `python3 --version`?

kengregson, 2017-08-18:
>python3 --version
Python 3.6.2

kengregson, 2017-08-18:
sorry, still open.

mikaelarguedas, 2017-08-18:
>It might be that your python version is not 3.6

based on the paths in the console output it looks like the python version is 3.6.

@kengregson can you confirm that this is a problem only with Python and that running the c++ examples work?
```
ros2 run demo_nodes_cpp talker
ros2 run demo_nodes_cpp listener
```

dhood, 2017-08-18:
you should call the executables directly though to avoid python e.g. `./ros2_install/ros2-osx/lib/demo_nodes_cpp/talker` (but from the above output it seems that also has linking issues)

kengregson, 2017-08-18:
You are correct, I get the same result/problem whether I invoke the nodes through "ros2 run" or directly for both cpp and python versions

kengregson, 2017-08-18:
@mikaelarguedas, I have same issue with both python and cpp listener/talker examples even if I run them directly (rather than via ros2 run).

mikaelarguedas, 2017-08-18:
Thanks for trying it out. I'm surprised you seem to have a setup very similar to our build machines.
Could you please provide the trace when you run just a c++ talker `./ros2_install/ros2-osx/lib/demo_nodes_cpp/talker`?
and `otool -L` on the failing lib. 

Can you try the same with [last night's archive](http://ci.ros2.org/view/packaging/job/packaging_osx/lastSuccessfulBuild/artifact/ws/ros2-package-osx-x86_64.tar.bz2) to make sure that the error is persistent?

@dhood and @wjwwood are much more familiar than me with MacOS and will likely give better direction for debugging why the RPATH is not populated properly

wjwwood, 2017-08-18:
Just looking at the discussion so far I don't have anything to add. Maybe after the info @mikaelarguedas asked for I can give some more input.

Sorry for the issues you're seeing @kengregson, hopefully we can sort them out. However, compiling from source might be a quicker work around for you:

https://github.com/ros2/ros2/wiki/OSX-Development-Setup

kengregson, 2017-08-27:
Sorry, got sidetracked trying to install all of ROS1/kinetic from source on OSX/Sierra hoping it was as easy as indigo on ElCapitan (still haven't found the end of that rainbow).

Tried the nightly and same issue.

Any chance this warning while building ROS2 from source is a clue (sounds suspiciously similar):
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   fastrtps

This warning is for project developers.  Use -Wno-dev to suppress it.

mikaelarguedas, 2017-08-30:
@kengregson thanks for the information. I don't think that the cmake warning is related to your issue. We will update our MacOS machines in the next few days and see if we can reproduce this error (we use python 3.6.1 on our machines that may be the reason). We will report here with the outcome

sivteck, 2017-09-17:
@mikaelarguedas any updates? (I am facing the facing the same issue with beta 3)

mikaelarguedas, 2017-09-17:
@sivteck can you confirm if you face this issue with both c++ and python nodes ?
Did you install Beta 3 from source or from binaries?

What is the result of the various following commands?

1) Running c++ nodes directly
`<ROS2_INSTALL_PATH>/lib/demo_nodes_cpp/talker`
`<ROS2_INSTALL_PATH>/lib/demo_nodes_cpp/listener`

1) Running c++ nodes using the `ros2` command
`ros2 run demo_nodes_cpp talker`
`ros2 run demo_nodes_cpp listener`

1) Running python nodes directly
`<ROS2_INSTALL_PATH>/lib/demo_nodes_py/talker`
`<ROS2_INSTALL_PATH>/lib/demo_nodes_py/listener`

1) Running python nodes using the `ros2` command
`ros2 run demo_nodes_py talker`
`ros2 run demo_nodes_py listener`

It may not be related but we recently found on windows that installing openssl and the relevant libraries fixed a similar issue. can you try to run `brew openssl version`? ROS2 beta3 is compiled against openssl 1.0.2l

sivteck, 2017-09-18:
I used the binaries. I was using the nix package manager, now I have changed to homebrew. Now I am able to run the C++ nodes directly but not the python nodes.
1.
```
./lib/demo_nodes_cpp/talker
Publishing: 'Hello World: 1'
Publishing: 'Hello World: 2'
```
(listener also works as expected)
2.
```
$ ros2 run demo_nodes_cpp talker
Failed to load entry point 'start': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'status': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'stop': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'list': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'call': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'list': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'echo': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'list': dlopen(/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: @rpath/librcl_interfaces__rosidl_typesupport_c.dylib
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
[... cut ...]
  Referenced from: /Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so
  Reason: image not found
The C extension '/Users/sivarambalakrishnan/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
```

also I have installed all the prerequisites,
```
$ openssl version
OpenSSL 1.0.2l  25 May 2017
```

I will try compiling from the source now.

zmk5, 2017-09-20:
I'm having the same issue as @sivteck. Calling the C++ files without `ros2 run` seem to work but not the python ones.

zmk5, 2017-09-20:
@mikaelarguedas I noticed what you wrote about openssl, and I decided to check the one I have. `brew openssl version` does not work, but just trying to install it showed that I have version 1.0.2l installed. An interesting thing I noticed, however, is that typing `openssl version` showed that version 0.9.8zh is also installed.

sivteck, 2017-09-20:
@zmk5 see
```
brew info openssl
```
for the brew installed openssl to be in your path you have to prepend it to the path
```
echo 'export PATH="/usr/local/opt/openssl/bin:$PATH"' >> ~/.zshrc
```

zmk5, 2017-09-20:
@sivteck Nice! That fixed my openssl issue, but I'm still running into the same issues as you with the binary install.

senthilmanick, 2017-09-23:
I am facing the same issue as @sivteck 
Is there a solution/workaround?

sivteck, 2017-09-24:
@senthilmanick workaround is to compile from source and use tinyxml2 4.0.1

BlockByBlock, 2017-09-25:
facing the same issue with failing to import, are we supposed to build tinyxml2 4.0.1 from source?

sivteck, 2017-09-25:
@BlockByBlock I had no luck with the binary install, so I installed tinyxml2 4.0.1 by following,
https://github.com/ros2/build_cop/issues/38#issue-237995936 and built ros 2 from source.

wjwwood, 2017-09-25:
> facing the same issue with failing to import, are we supposed to build tinyxml2 4.0.1 from source?

I would say no. Everyone should be using tinyxml2 5.0.1 and either using beta-3 binaries or building from source. This is what I did to install tinyxml2:

```
william@farl ~
% brew install tinyxml2
==> Auto-updated Homebrew!
Updated Homebrew from 46e6e9911a to 880f980f4e.
No changes to formulae.

==> Downloading https://homebrew.bintray.com/bottles/tinyxml2-5.0.1.el_capitan.bottle.tar.gz
Already downloaded: /Users/william/Library/Caches/Homebrew/tinyxml2-5.0.1.el_capitan.bottle.tar.gz
==> Pouring tinyxml2-5.0.1.el_capitan.bottle.tar.gz
🍺  /usr/local/Cellar/tinyxml2/5.0.1: 11 files, 152.9KB

william@farl ~
% brew info tinyxml2
tinyxml2: stable 5.0.1 (bottled), HEAD
Improved tinyxml (in memory efficiency and size)
http://grinninglizard.com/tinyxml2
/usr/local/Cellar/tinyxml2/5.0.1 (11 files, 152.9KB) *
  Poured from bottle on 2017-09-25 at 11:23:40
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/tinyxml2.rb
==> Dependencies
Build: cmake ✔
```

I am building from source now to check and see if the issue persists, but during testing for beta-3 last week I did not notice an issue and tinyxml2 has not changed since then.

wjwwood, 2017-09-25:
Also, this issue doesn't really have anything to do with `tinyxml2` AFAICT, so maybe consider opening a new issue if your trouble with tinyxml2 persists.

wjwwood, 2017-09-25:
I was just able to build all of ros2 from source with the tinyxml2 I posted about in the previous comment. If you're still having issues with tinyxml2 and either the beta-3 binary or building from source, please open a new issue and include as much information as you can.

dhood, 2017-10-03:
@kengregson @zmk5 @sivteck @senthilmanick @BlockByBlock and others encountering _runtime_ issues with the ROS2 OSX binaries:

If you have System Integrity Protection enabled (default for OS X >=10.11), it prevents `DYLD_LIBRARY_PATH` from being inherited by launched processes.
https://developer.apple.com/library/content/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.html details how to disable it which should fix the error you're seeing (the wiki has been updated).

kengregson, 2017-10-05:
@dhood, that got me going - thanks!

Using latest r2b3 binary install on OSX (both 10.11.6 native, and 10.12.6 in a VM), I verified:
a) with SIP enabled, able to run "./ros2-osx/lib/demo_nodes_cpp/{listener | talker}"
b) with SIP disabled, also able to run "ros2 run demo_nodes_{cpp | py} {listener | talker}"

Is this expected to remain the behavior for OSX binary installs since disabling SIP is generally not a recommended practice and is unlikely to be permitted in enterprise/work environments?

dhood, 2017-10-12:
good to hear, @kengregson; I'll close this ticket out.

RE disabling SIP, https://github.com/ros2/ros2/issues/409 has discussion of alternatives but for now disabling it is the most straightforward thing to advise users; we're open to suggestions if others have them.

closed 2017-10-12 by dhood