ros2/ros2 #393: Tracking bug for message initialization

clalancette, 2017-08-25:
This is a tracking bug to deal with field initialization of generated message code in C++, discussed in #367 (and earlier in ros2/geometry2#25).

- [ ] Change the design article about C++ constructors to mention this optional argument: ros2/design#136 (this is actually "done", but I won't merge it until the implementation is ready to go)
- [ ] Add a new repository called `rosidl_runtime_cpp` (I did this, with an empty repository with just a README.md and a LICENSE file)
- [ ] Add `rosidl_runtime_cpp` to the waffleboard
- [ ] Make the `rosidl_runtime_cpp` repository have useful infrastructure.
- [ ] Add `rosidl_runtime_cpp` to the list of repositories to check out in [ros2.repos](http://github.com/ros2/ros2/blob/master/ros2.repos)
- [ ] Change all of the repositories that generate messages to export a dependency on `rosidl_runtime_cpp`:
  - [ ] common_interfaces
  - [ ] demos
  - [ ] geometry2
  - [ ] rclcpp
  - [ ] rcl_interfaces
  - [ ] system_tests
- [ ] Change the generator in `rosidl/rosidl_generator_cpp` to include the header file from `rosidl_runtime_cpp`, and generate the constructor with the argument on how to initialize
- [ ] Write tests for the generated constructors

mikaelarguedas, 2017-08-25:
Can you please propagate the github labels to the new repo as well ? https://github.com/ros2/ros2/issues/389

mikaelarguedas, 2017-08-25:
I think that you will need to also update the repositories that generate messages in the turtlebot2_demo.repos file

clalancette, 2017-08-25:
> Can you please propagate the github labels to the new repo as well ? #389

Github labels are now setup properly.

clalancette, 2017-08-25:
> I think that you will need to also update the repositories that generate messages in the turtlebot2_demo.repos file

I did a quick scan through those repositories.  Of the additional ones in turtlebot2_demo.repos, only ros2/navigation, ros2/ros_astra_camera, and ros2/cartographer_ros have .msg or .srv files.  The navigation .msg files are all in parts of navigation that aren't currently being used in ROS2.  Ditto for ros_astra_camera.  That just leaves ros2/cartographer_ros, which I'll add to the list above.  Thanks!

dirk-thomas, 2017-08-25:
Beside `rosidl_runtime_cpp` we will also require a `rosidl_runtime_c` (assuming we want to provide the same concept in C). I am just not sure if it makes sense to create new repos for each of these. Why not add them to the existing `rosidl` repo?

clalancette, 2017-08-25:
Sure, that is fine too.  If you think that is a better way to go, then I'll close out the current one and move it into rosidl.  @mikaelarguedas Sorry for the false start, I'll be opening a new PR very similar to the old one (but in rosidl instead) soon.

dirk-thomas, 2017-08-25:
Currently some of the generator packages contain header file / Python modules which are needed at runtime from message packages. Those should be moved into the newly create runtime packages.

The goal should be that message packages only build depend on `rosidl_default_generators` and only run depend on `rosidl_default_runtime` (which doesn't need to depend on any generator). They shouldn't depend on any of the language specific generators / runtimes. So when installing Debian packages the user wouldn't need any generator package for just running stuff.

dirk-thomas, 2017-08-25:
> Currently some of the generator packages contain header file / Python modules which are needed at runtime from message packages. Those should be moved into the newly create runtime packages.

> The goal should be that message packages only build depend on `rosidl_default_generators` and only run depend on `rosidl_default_runtime` (which doesn't need to depend on any generator). They [...]

If this is too far reaching for this ticket I would suggest not to create the runtime packages for now and declare the necessary symbols in the generators. That would separate the two orthogonal enhancements without ending up in an in-between state.

mikaelarguedas, 2017-08-25:
> The goal should be that message packages only build depend on rosidl_default_generators and only run depend on rosidl_default_runtime (which doesn't need to depend on any generator). They shouldn't depend on any of the language specific generators / runtimes. 

:+1: that will also allow to target specific languages / add languages with minimal changes (ala ROS1, i.e. changing only the default rosidl packages and not have to change every existing message package)

mikaelarguedas, 2017-08-30:
@clalancette What should we do with the [original meta-ticket](https://github.com/ros2/ros2/issues/367) ? Shouldnt they be the same issue? 
Should https://github.com/ros2/example_interfaces/pull/1 be connected to this ticket as well?

clalancette, 2017-09-06:
> If this is too far reaching for this ticket I would suggest not to create the runtime packages for now and declare the necessary symbols in the generators. That would separate the two orthogonal enhancements without ending up in an in-between state.

Reducing scope here will definitely help me get this in.  I'm fine with not making the new packages for now, as that should be less changes overall.  I'll go ahead with this plan unless someone objects.

clalancette, 2017-09-06:
> @clalancette What should we do with the original meta-ticket ? Shouldnt they be the same issue?
> Should ros2/example_interfaces#1 be connected to this ticket as well?

@mikaelarguedas Yeah, I'll close out the original meta-ticket, thanks for the hint.  ros2/example_interfaces#1 is already linked to this ticket via "connects to", and via the checklist above.

mikaelarguedas, 2017-09-06:
> ros2/example_interfaces#1 is already linked to this ticket via "connects to", 

It "connects to" the wrong ticket that's why it doesnt show up connected to anything on waffle

clalancette, 2017-09-06:
> It "connects to" the wrong ticket that's why it doesnt show up connected to anything on waffle

Ah, sorry about that!  Fixed now.

mikaelarguedas, 2017-09-08:
should this be put back in backlog ?

clalancette, 2017-09-08:
Yeah, I'm not going to touch this again until after Beta 3.  I'll move it now.

mikaelarguedas, 2017-09-08:
> Yeah, I'm not going to touch this again until after Beta 3. I'll move it now.

:+1: 

Can you also delete the branches of all the referenced closed PRs if you dont need them anymore ?

clalancette, 2018-01-02:
I'm going to close this out since it is old and confusing.  I'll be opening a new bug to track what we really want to do here.

closed 2018-01-02 by clalancette