ros2/ros2 #394: Fails to build against tinyxml2 5.0.1

kengregson, 2017-08-27:
During OSX (10.12.6) Source Install, linking of fastrtps fails because of:
ld: symbol(s) not found for architecture x86_64

For a bunch of tinyxml2 symbols (eg):
tinyxml2::XMLNode::Value() const", referenced from:
      eprosima::fastrtps::xmlparser::XMLProfileParser::getXMLBool(tinyxml2::XMLElement*, bool*, unsigned char) in XMLProfileElementParser.cpp.o

With tinyxml2 5.0.1, looks suspiciously similar to:
https://github.com/eProsima/Fast-RTPS/issues/113
That was addressed for tinyxml2 5.0.0 with:
https://github.com/eProsima/Fast-RTPS/pull/114

dirk-thomas, 2017-08-28:
Please add the information requested in the [issue template](https://github.com/ros2/ros2/blob/master/.github/ISSUE_TEMPLATE.md), specifically:

* Commit hash of the FastRTPS repository or other packages for which the linking fails

allenh1, 2017-08-28:
@kengregson I got this as well -- It's useful to note that I installed `tinyxml2` via source.

@dirk-thomas For me, I was good up to `composition`:

```
Process package 'composition' with context:
--------------------------------------------------------------------------------
 source_space => /u/scratch2/allen286/ros2_ws/src/ros2/demos/composition
  build_space => /homes/allen286/scratch/ros2_ws/build/composition
install_space => /homes/allen286/scratch/ros2_ws/install
   make_flags => -j8, -l8
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'composition'
Running cmake because arguments have changed.
==> '. /homes/allen286/scratch/ros2_ws/build/composition/cmake__build.sh && /usr/bin/cmake /u/scratch2/allen286/ros2_ws/src/ros2/demos/composition -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/homes/allen286/scratch/ros2_ws/install' in '/homes/allen286/scratch/ros2_ws/build/composition'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found ament_index_cpp: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/ament_index_cpp/cmake)
-- Found class_loader: 1.0.0 (/homes/allen286/scratch/ros2_ws/install/share/class_loader/cmake)
-- Found example_interfaces: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/example_interfaces/cmake)
-- Found rclcpp: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rmw_fastrtps_cpp: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/rmw_fastrtps_cpp/cmake)
-- Found fastrtps_cmake_module: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/fastrtps_cmake_module/cmake)
-- Found FastRTPS: /homes/allen286/scratch/ros2_ws/install/include  
-- Found rosidl_default_generators: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/rosidl_default_generators/cmake)
-- Found std_msgs: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/std_msgs/cmake)
-- Using single matching rosidl_typesupport_c: rosidl_typesupport_introspection_c
-- Using single matching rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.5") 
-- Found python_cmake_module: 0.0.2 (/homes/allen286/scratch/ros2_ws/install/share/python_cmake_module/cmake)
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "3.5") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.5m
-- Using PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.5m.so
-- Found PythonExtra: .so  
-- Configuring done
-- Generating done
-- Build files have been written to: /u/scratch2/allen286/ros2_ws/build/composition
==> '. /homes/allen286/scratch/ros2_ws/build/composition/cmake__build.sh && /usr/bin/make -j8 -l8' in '/homes/allen286/scratch/ros2_ws/build/composition'
Scanning dependencies of target composition__cpp
Scanning dependencies of target server_component
Scanning dependencies of target talker_component
Scanning dependencies of target listener_component
Scanning dependencies of target dlopen_composition
Scanning dependencies of target composition__py
Scanning dependencies of target client_component
[  1%] Generating C code for ROS interfaces
[  3%] Generating C++ code for ROS interfaces
[  5%] Generating Python code for ROS interfaces
[  5%] Built target composition__cpp
[  7%] Generating C++ introspection for ROS interfaces
[... 53 lines cut ...]
/homes/allen286/scratch/ros2_ws/install/lib/libPocoFoundation.so.47: undefined reference to `pcre_compile'
/homes/allen286/scratch/ros2_ws/install/lib/libfastrtps.so: undefined reference to `tinyxml2::XMLDocument::XMLDocument(bool, tinyxml2::Whitespace)'
collect2: error: ld returned 1 exit status
CMakeFiles/dlopen_composition.dir/build.make:119: recipe for target 'dlopen_composition' failed
make[2]: *** [dlopen_composition] Error 1
CMakeFiles/Makefile2:706: recipe for target 'CMakeFiles/dlopen_composition.dir/all' failed
make[1]: *** [CMakeFiles/dlopen_composition.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 61%] Built target composition__rosidl_typesupport_c
[ 61%] Built target client_component
[ 61%] Built target talker_component
[ 61%] Built target listener_component
Scanning dependencies of target composition__rosidl_typesupport_cpp
[... cut ...]
[ 69%] Built target composition__rosidl_typesupport_cpp
[ 69%] Built target composition__py
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /homes/allen286/scratch/ros2_ws/build/composition/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/homes/allen286/scratch/ros2_ws/build/composition' with exit code '2'
<== Command '. /homes/allen286/scratch/ros2_ws/build/composition/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/homes/allen286/scratch/ros2_ws/build/composition' with exit code '2'
```

mikaelarguedas, 2017-08-28:
@allenh1 please provide the information mentioned in https://github.com/ros2/ros2/issues/394#issuecomment-325395976 as well.
Looks like you have warnings regarding poco. Do you have only 1 rmw_implementation installed ? if that's the case composition would be the only package using poco. Providing what version of tinyxml2 and well as what version of poco you have installed would be useful as well. Because it looks like you built both from source rather from debs.

allenh1, 2017-08-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04.3
- Installation type:
  - source
- Version or commit hash:
  - master
- DDS implementation:
  - rmw_fastrtps
- Client library (if applicable):
  - n/a

#### Steps to reproduce issue
I have the following installed from source:
 * asio
 * boost
 * eigen
 * openssl
 * poco
 * pcre

---

> Looks like you have warnings regarding poco. Do you have only 1 rmw_implementation installed ? if that's the case composition would be the only package using poco.

@mikaelarguedas I just ran a clean build, and the poco warnings are gone now. Just the boost ones remain (I hadn't installed pcre when I built fastrtps).

mikaelarguedas, 2017-08-28:
Can you confirm the version numbers of what you built from source please?
The console output of the fastrtps build and the updated console output of the composition build would be useful as well

kengregson, 2017-08-28:
OSX 10.12.6, TinyXML2 5.0.1 installed through homebrew
output of git rev-parse HEAD:
b96e02861239a1ca3d1cb91a4760bba4c1df3498

The work around in https://github.com/eProsima/Fast-RTPS/pull/114 appears to work for 5.0.1
(although I have to build without --build-tests to avoid a different issue, to be filed) and was reported merged in commit c3ab188 into eProsima:release/1.5.0 on Jul 4

allenh1, 2017-08-28:
> Can you confirm the version numbers of what you built from source please?

Versions
-------------
 - asio: 1.10.6
 - boost: 1.65
 - pcre: 8.41
 - openssl: 1.0.2l
 - poco: 1.7.8p3
 - tinyxml2: git revision d69e2c0d085384ce305873efa0740f33d4fadab5
 - eigen 3.3.90

I forgot I built tinyxml2 from git sources. I'll try building with 5.0.1.

dirk-thomas, 2017-08-28:
@allenh1 Please go ahead to investigate what it causing the problem here. Any help on this is appreciated. Until it is clear what is causing it and what change is necessary to fix it I will keep the issue unassigned.

allenh1, 2017-08-29:
@dirk-thomas got it.

---

Things now compile after adding this patch to eProsima/Fast-RTPS.

```patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index ee7fc73..12cacff 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -106,6 +106,12 @@ else()
     set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -pedantic -Wextra -Wno-unknown-pragmas -std=c++11")
 endif()
 
+
+###############################################################################
+# force position-independent code
+###############################################################################
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fPIC")
+
 ###############################################################################
 # GCC colors if using CCache
 ###############################################################################
```

They have an internal library for tinyxml2, so I think that's being used here. From a fresh build, this made things mostly work. `urdf` is still not working. That's the next step (also the patch doesn't need to ad `-fPIC` to the global C++ flags, it could just add them to the tinyxml build).

kengregson, 2017-08-29:
Thread hijacked? At present time there is no success path to OSX via binary install or source build.

wjwwood, 2017-08-29:
@kengregson

> At present time there is no success path to OSX via binary install or source build.

It looked like you could build on macOS with the fix in https://github.com/eProsima/Fast-RTPS/pull/114. That fix has been merged, so there's nothing more that we can do I think.

I can build the latest beta-2 as well as master on my macbook using tinyxml 5.0.1:

```
% brew info tinyxml2
tinyxml2: stable 5.0.1 (bottled), HEAD
Improved tinyxml (in memory efficiency and size)
http://grinninglizard.com/tinyxml2
/usr/local/Cellar/tinyxml2/5.0.1 (11 files, 152.9KB) *
  Poured from bottle on 2017-07-21 at 13:23:57
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/tinyxml2.rb
==> Dependencies
Build: cmake ✔
```

mikaelarguedas, 2017-08-30:
@allenh1 please create a new ticket with your results after trying the same setup as the one recommended in the installation instructions (everything dependency from binaries and the ros2 workspace from source).
Then the same thing with tinyxml2 5.0.1 (I don't think tracking master make sense, just their releasd versions)

> They have an internal library for tinyxml2, so I think that's being used here. 

If you don't pass specific cmake argument you should not used the version of tinyxml2 within the Fast-RTPS repo. The best way to confirm it is to not clone it recursively and thus not download any of the submodules

kengregson, 2017-08-31:
@wjwwood, I'm getting closer - perhaps
OSX 10.12.6, source build from master, encounter errors with rcl,
commit hash 173f1da229200d605089b0590cc3daab5ba46d9d:

CMake Error at test/cmake/rcl_add_custom_launch_test.cmake:36 (set_tests_properties):
  set_tests_properties Can not find test to add properties to:
  test_services__rmw_fastrtps_cpp

running ament without --build-tests runs to completion and talker example appears to run:
$ ros2 run demo_nodes_cpp talker
Publishing: 'Hello World: 1'
Publishing: 'Hello World: 2'
...

but, in second terminal, listener waits forever with no output:
$ ros2 run demo_nodes_cpp listener

and no topics returned from:
$ ros2 topic list
$

Likely something I'm (not) doing, have had complete success installing/testing from both binary and source on Linux. More detailed rcl error log attached in case it helps
[ros2_master-error.txt](https://github.com/ros2/ros2/files/1267811/ros2_master-error.txt)

mikaelarguedas, 2017-08-31:
Thanks @kengregson for the update
Regarding the rcl error from your file it looks like you are missing some test dependencies. This is likely because we specify only the runtime dependencies in the installation instructions from binaries. Can you give a try to installing the dependencies listed in the "install prerequisite" section of the building from [OSX building from source instructions](https://github.com/ros2/ros2/wiki/OSX-Development-Setup#install-prerequisites) and build your workspace again?

kengregson, 2017-08-31:
@mikaelarguedas, appreciate the help.  Went back through the process described in:
https://github.com/ros2/ros2/wiki/OSX-Development-Setup#install-prerequisites
and
https://github.com/ros2/ros2/wiki/Maintaining-a-Source-Checkout
to make sure I had the latest of everything and a fresh copy of master

From the attached, it appears that all the prereq dependencies are/had already been installed.
Still get the same result as you can see in the full start to finish log attached
[ros2_master_latest-prereq.txt](https://github.com/ros2/ros2/files/1268215/ros2_master_latest-prereq.txt)
[ros2_master_latest-build.txt](https://github.com/ros2/ros2/files/1268217/ros2_master_latest-build.txt)

mikaelarguedas, 2017-08-31:
hmm that's very surprising. Does nose appear in the list if you run `python3 -m pip freeze` ?

kengregson, 2017-09-01:
@mikaelarguedas yes:
$ python3 -m pip freeze | grep nose
nose==1.3.7

kengregson, 2017-09-01:
@mikaelarguedas, However nosetests isn't in /usr/local/bin as might be expected?
sudo find / -type f -name "nosetests*" -print
/usr/local/Cellar/numpy/1.13.1_1/libexec/nose/bin/nosetests
/usr/local/Cellar/numpy/1.13.1_1/libexec/nose/bin/nosetests-2.7
/usr/local/Cellar/numpy/1.13.1_1/libexec/nose/bin/nosetests-3.6
/usr/local/Cellar/numpy/1.13.1_1/libexec/nose/man/man1/nosetests.1

kengregson, 2017-09-01:
@mikaelarguedas, maybe see where you are going...

Uninstalled and re-installed nose so that nosetest is actually in /usr/local/bin and search path.  Attached install log shows that build (may) complete successfully (with --build-tests) now.  However I still get the same execution behavior for the talker/listener example (no connection and no topics advertised):

$ ros2 run demo_nodes_cpp talker
Publishing: 'Hello World: 1'
Publishing: 'Hello World: 2'
...

but, in second terminal, listener waits forever with no output:
$ ros2 run demo_nodes_cpp listener

and no topics returned from:
$ ros2 topic list
$

kengregson, 2017-09-01:
@mikaelarguedas, @wjwwood: Thanks!  Call this closed, tinyxml 5.0.1 patch in master fixes the original problem, and open a separate issue on this once I get a better handle on what the source of the (separate) problem might be?

closed 2017-09-01 by kengregson

mikaelarguedas, 2017-09-01:
thanks @kengregson for following up on this.

>  and open a separate issue on this once I get a better handle on what the source of the (separate) problem might be

Yes please, we'll look into the follow-up issue shortly, sorry for the bumpy ride