ros2/ros2 #395: MacOS failed to install ros2

DrTangCHN, 2017-09-05:
$ brew install homebrew/science/opencv --without-python
Error: No available formula with the name "homebrew/science/opencv" 
==> Searching for a previously deleted formula...
homebrew/science/opencv was deleted from homebrew/science in commit 6c38af15:
  opencv, opencv3: migrate to homebrew/core (https://github.com/Homebrew/homebrew-science/issues/6126)

To show the formula before removal run:
  git -C "$(brew --repo homebrew/science)" show 6c38af15^:opencv.rb

If you still use this formula consider creating your own tap:
  https://docs.brew.sh/How-to-Create-and-Maintain-a-Tap.html

$ ros2 run demo_nodes_cpp talker
Failed to load entry point 'start': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'status': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'stop': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'list': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'call': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'list': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'echo': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'list': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
Failed to load entry point 'pub': dlopen(/Users/tzj/ros2_install/ros2-osx/lib/python3.6/site-packages/rclpy/_rclpy.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found
dyld: Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.4.dylib
  Referenced from: /Users/tzj/ros2_install/ros2-osx/lib/libfastrtps.1.dylib
  Reason: image not found

wjwwood, 2017-09-05:
tl;dr please provide some missing info from the issue template, like what tutorial you are following.

---

For opencv, the error tells you what needs to happen, you need to do `brew install opencv --without-python` since it is no longer in `homebrew/science`.

---

Also the second error doesn't have enough information to reproduce. Did you build from source or did you install from a binary? Did you source the setup file before running the example? Again, are you following a tutorial?

All of these questions are covered in the issue template, if you'd like our help please provide the information we need to help you:

```
## Bug report

**Required Info:**

- Operating System:
  - <!-- OS and version (e.g. Windows 10, Ubuntu 16.04...) -->
- Installation type:
  - <!-- binaries or from source  -->
- Version or commit hash:
  - <!-- Output of git rev-parse HEAD, release version, or repos file  -->
- DDS implementation:
  - <!-- rmw_implementation used (e.g. Fast-RTPS, RTI Connext, etc -->
- Client library (if applicable):
  - <!-- e.g. rclcpp, rclpy, or N/A -->

#### Steps to reproduce issue
<!-- Detailed instructions on how to reliably reproduce this issue http://sscce.org/
``` code that can be copy-pasted is preferred ``` -->
```

DrTangCHN, 2017-09-05:
as current ros version mistake,rename /usr/local/opt/tinyxml2/lib/libtinyxml2.5.dylib and fixed

closed 2017-09-05 by DrTangCHN