ros2/ros2 #396: Tracking bug for message initialization

clalancette, 2017-09-06:
In porting efforts we've run into several bugs where the zero initialized behavior is relied upon, causing bugs. Having a zero initialization method as a way to make porting less fragile would be valuable. And likely can be used in other cases when the optimization of using uninitialized constructors are not as important.

This is a tracking bug to deal with field initialization of generated message code in C++ (originally discussed in ros2/geometry2#25).

- [ ] Change the design article about C++ constructors to mention this optional argument: ros2/design#136 (this is actually "done", but I won't merge it until the implementation is ready to go)
- [ ] Change the generator in `rosidl/rosidl_generator_cpp` to include the header file from `rosidl_runtime_cpp`, and generate the constructor with the argument on how to initialize
- [ ] Write tests for the generated constructors

clalancette, 2017-09-06:
CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2017-09-15:
I can't find a meta-ticket for all message initialization regardless of the language so I'm posting here. Feel free to move this comment to the relevant ticket that I couldnt find.
We originally said we didnt need to modify the Python message generation because all messages are initialized by default. But now that we offer the ability to initialize messages with only language default or with default values from msg if they exist and language zeros otherwise I think we should provide the same feature in Python

clalancette, 2017-11-01:
All of the pieces are now in, we can close this out.

mikaelarguedas, 2017-11-01:
closing this now that the connected tickets have been closed / merged.
@clalancette will open another ticket to track the changes that could/should be reverted in other repos now that message fields are initialized by default

closed 2017-11-01 by mikaelarguedas