ros2/ros2 #398: ros2 version alpha 6 install error

sagniknitr, 2017-09-11:
## Bug report

**Required Info:**

- Operating System:
  - <Ubuntu 14.04 -->
- Installation type:
  - <source  -->
- Version:
  - <alpha 6>

#### Steps to reproduce issue

ROS2 installation procedure

Build the prototype using the bootstrap script from ament_tools

Note: to build the ROS 1 bridge, read the ros1_bridge instructions.

More info on working with an ament workspace can be found in this tutorial.

cd ~/ros2_ws/
**src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install** < error occurs in this command
[image]

allenh1, 2017-09-11:
Have you tried with Beta 2? Alpha 6 was quite a while ago.

wjwwood, 2017-09-11:
As @allenh1 said, alpha6 is old, and we're not going to be supporting them. Please try the latest release.

Also, please do not post pictures of text. We cannot search, copy&paste, or comment on a picture of text.

closed 2017-09-19 by sagniknitr