ros2/ros2 #399: Optionally support opensplice in overlay after underlay built with opensplice

dhood, 2017-09-11:
Users without opensplice installed cannot use the fat archive to build overlays if the archive was built with opensplice support.

If opensplice support is built in the underlay,

1. `install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake` lists opensplice typesupport as a default generator.
1. `rmw_opensplice_cpp` will be registered in the ament index as a supported rmw implementation, so `call_for_each_rmw_implementation` will find it.

(1) causes issues because opensplice is required [in this findpackage call](https://github.com/ros2/rmw_opensplice/blob/f2273ca3784450830cd42c943970a11a8147ccd9/rmw_opensplice_cpp/rmw_opensplice_cpp-extras.cmake#L18)

(2) causes issues because [`rosidl_generator_py_get_typesupports.cmake`](https://github.com/ros2/rosidl/blob/0c7fd8bad40e66c167952aaa0b34bb30190d9780/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake#L24) calls `call_for_each_rmw_implementation` which find_packages each rmw_implementation. There might be other side effects.

Allowing [`rmw_opensplice_cpp-extras.cmake`](https://github.com/ros2/rmw_opensplice/blob/f2273ca3784450830cd42c943970a11a8147ccd9/rmw_opensplice_cpp/rmw_opensplice_cpp-extras.cmake#L18) to give a warning when it can't find opensplice, instead of fail, should address both points. If there are other points that should be considered please list them here

dhood, 2017-09-11:
also as @dirk-thomas pointed out the extensions can't be registered unless opensplice is found in https://github.com/ros2/rmw_opensplice/blob/f2273ca3784450830cd42c943970a11a8147ccd9/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp-extras.cmake.in#L6 and the c equivalent.

dhood, 2017-09-11:
Even with the extensions only registered if opensplice is found, and with the opensplice find_package calls not using `REQUIRED`, another issue is called by (2).

`rmw_opensplice_cpp` is registered in the ament index as a supported rmw implementation. `call_for_each_rmw_implementation`  can be updated to not fail when it doesn't find opensplice, but the challenge is that it's only trying to find_pacakge `rmw_opensplice_cpp`, which is actually there, just not useful. `rosidl_generator_py_get_typesupports.cmake`(https://github.com/ros2/rosidl/blob/0c7fd8bad40e66c167952aaa0b34bb30190d9780/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake#L227) therefore thinks opensplice typesupport is valid and tries to add dependencies on targets which don't actually exist.

We would need a concept of rmw implementations being "usable" as opposed to just "available" in order to fail gracefully when rmw implementations that were there when building the underlay are no longer there when building an overlay. This will take more work than initially thought and is not targeted for beta3.

We will not ship combined rmw implementation support in the fat archives (or if we do it will be with the requirement that the user installs particular vendors)

dirk-thomas, 2017-11-10:
With the four referenced PRs this use case should be working now. The following CI builds won't say anything about that though - only that the patches don't break anything:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Please review the associated PRs and consider trying this locally (building an underlay with N rmw impl. and building an overlay afterwards which has less rmw impl. available).

mikaelarguedas, 2017-11-10:
Tested the fix with an underlay with opensplice and an overlay without it and everything built fine :+1:

closed 2017-11-10 by dirk-thomas