ros2/ros2 #400: r2b3 Cannot call local_setup.bat on Windows

sloretz, 2017-09-11:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - From http://ci.ros2.org/view/packaging/job/packaging_windows/727/artifact/ws/ros2-package-windows-AMD64.zip
- Version or commit hash:
  - Job says 46905f6386789a2e0ecaabbaf1523ef23bf89647, not sure what repo that refers to
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Following https://github.com/ros2/ros2/wiki/Windows-Install-Binary reached the `Try some examples` stage.
```
C:\Users\osrf>dir C:\dev\ros2\local_setup.bat
 Volume in drive C has no label.
 Volume Serial Number is AA39-7FB6

 Directory of C:\dev\ros2

09/11/2017  02:40 PM             1,576 local_setup.bat
               1 File(s)          1,576 bytes
               0 Dir(s)   9,530,646,528 bytes free

C:\Users\osrf>call C:\dev\ros2\local_setup.bat
The system cannot find the path specified.

C:\Users\osrf>talker
'talker' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\osrf>
```

#### Expected behavior
I'd expect no output when calling `local_setup.bat` and talker to be recognized as a command.
#### Actual behavior
Despite the file `C:\dev\ros2\local_setup.bat` existing, `call`ing it has no affect

#### Additional information

dirk-thomas, 2017-09-11:
`talker` is not supposed to be on the path. You need to invoke it with `ros2 run <pkgname> talker` instead.

To debug the setup file you can set `AMENT_TRACE_SETUP_FILES=1` and the called setup file should give a verbose trace what it is doing. Maybe that narrows down what is responsible for the error message.

mikaelarguedas, 2017-09-11:
Indeed the wiki is outdated. Can you please update the wiki page to reflect that the talker and listener executables have to be invoked with `ros2 run ...`

sloretz, 2017-09-11:
Output shows the error message after `call "C:\dev\ros2\share\opensplice_cmake_module\environment\opensplice.bat"`

```
C:\Users\osrf>set AMENT_TRACE_SETUP_FILES=1
call "C:\dev\ros2\share\ament_clang_format\local_setup.bat"
call "C:\dev\ros2\share\ament_clang_format\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_clang_format\environment\path.bat"
call "C:\dev\ros2\share\ament_clang_format\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_copyright\local_setup.bat"
call "C:\dev\ros2\share\ament_copyright\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_copyright\environment\path.bat"
call "C:\dev\ros2\share\ament_copyright\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_cppcheck\local_setup.bat"
call "C:\dev\ros2\share\ament_cppcheck\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cppcheck\environment\path.bat"
call "C:\dev\ros2\share\ament_cppcheck\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_cpplint\local_setup.bat"
call "C:\dev\ros2\share\ament_cpplint\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cpplint\environment\path.bat"
call "C:\dev\ros2\share\ament_cpplint\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_flake8\local_setup.bat"
call "C:\dev\ros2\share\ament_flake8\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_flake8\environment\path.bat"
call "C:\dev\ros2\share\ament_flake8\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_index_python\local_setup.bat"
call "C:\dev\ros2\share\ament_index_python\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_index_python\environment\path.bat"
call "C:\dev\ros2\share\ament_index_python\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_lint_cmake\local_setup.bat"
call "C:\dev\ros2\share\ament_lint_cmake\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_lint_cmake\environment\path.bat"
call "C:\dev\ros2\share\ament_lint_cmake\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_package\local_setup.bat"
call "C:\dev\ros2\share\ament_package\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_package\environment\path.bat"
call "C:\dev\ros2\share\ament_package\environment\pythonpath.bat"
call "C:\dev\ros2\share\ament_cmake_core\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_core\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_core\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_export_definitions\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_export_definitions\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_export_definitions\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_export_include_directories\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_export_include_directories\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_export_include_directories\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_export_interfaces\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_export_interfaces\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_export_interfaces\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_export_libraries\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_export_libraries\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_export_libraries\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_export_link_flags\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_export_link_flags\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_export_link_flags\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_include_directories\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_include_directories\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_include_directories\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_libraries\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_libraries\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_libraries\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_export_dependencies\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_export_dependencies\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_export_dependencies\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_python\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_python\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_python\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_target_dependencies\local_setup.bat"
call "C:\dev\ros2\share\ament_cmake_target_dependencies\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\ament_cmake_target_dependencies\environment\path.bat"
call "C:\dev\ros2\share\ament_cmake_test\local_setup.bat"
[... 419 lines cut ...]
call "C:\dev\ros2\share\uncrustify\environment\path.bat"
call "C:\dev\ros2\share\urdfdom\local_setup.bat"
call "C:\dev\ros2\share\urdfdom\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\urdfdom\environment\path.bat"
call "C:\dev\ros2\share\urdfdom_headers\local_setup.bat"
call "C:\dev\ros2\share\urdfdom_headers\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\urdfdom_headers\environment\path.bat"
call "C:\dev\ros2\share\urdf\local_setup.bat"
call "C:\dev\ros2\share\urdf\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\urdf\environment\path.bat"
[... cut ...]
call "C:\dev\ros2\share\robot_state_publisher\environment\path.bat"
call "C:\dev\ros2\share\dummy_robot_bringup\local_setup.bat"
call "C:\dev\ros2\share\dummy_robot_bringup\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\dummy_robot_bringup\environment\path.bat"
call "C:\dev\ros2\share\visualization_msgs\local_setup.bat"
call "C:\dev\ros2\share\visualization_msgs\environment\ament_prefix_path.bat"
call "C:\dev\ros2\share\visualization_msgs\environment\path.bat"
call "C:\dev\ros2\share\visualization_msgs\environment\pythonpath.bat"

C:\Users\osrf>

dirk-thomas, 2017-09-12:
> Output shows the error message after `call "C:\dev\ros2\share\opensplice_cmake_module\environment\opensplice.bat"`

Have you checked the content of that file?

sloretz, 2017-09-12:
I checked the content of that file and it does exist and has content. I think it was failing on a call which used a vartiable %OPL_HOME% which was hard coded to a path on the C:\ drive. I haven't installed open splice so that path doesn't exist.

I abandoned this for the time being to use the artifact from http://ci.ros2.org/view/packaging/job/packaging_windows/728/ which does not have any packages referencing opensplice.

dirk-thomas, 2017-09-12:
That sounds like a symptom related to ros2/ros2#399.

sloretz, 2017-09-12:
It does. I doubt there's more to be learned here, so I'll close this in favor of #399

closed 2017-09-12 by sloretz