ros2/ros2 #403: r2b3 ros2 command not found using windows power shell

sloretz, 2017-09-12:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - http://ci.ros2.org/view/packaging/job/packaging_windows/728/artifact/ws/ros2-package-windows-AMD64.zip
- Version or commit hash:
  - def304b4270b6b935b7d6fbbf7671634c2532452
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
The tutorial at https://github.com/ros2/ros2/wiki/Quality-Of-Service says the instructions for windows power shell are to invoke the `.bat` file and then use the commands like normal.

```
PS C:\Users\osrf> C:\dev\ros2\local_setup.bat
PS C:\Users\osrf> ros2 run image_tools cam2image -- -b
ros2 : The term 'ros2' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included,
verify that the path is correct and try again.
At line:1 char:1
+ ros2 run image_tools cam2image -- -b
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (ros2:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

#### Expected behavior
Expect `cam2image` node to start publishing

#### Actual behavior
ros2 command not found

#### Additional information

The commands work fine using cmd.exe, but not power shell.

Not sure how environment variables work in windows, but after invoking `local_setup.bat` PATH seems to be empty in power shell

```
PS C:\Users\osrf> echo $PATH
PS C:\Users\osrf> echo %PATH%
%PATH%
PS C:\Users\osrf>
```

but not in cmd.exe

```
C:\Users\osrf>echo %PATH%
C:\dev\ros2\Scripts;C:\dev\ros2\bin;C:\Python36\Scripts\;C:\Python36\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\dev\opencv-2.4.13.2-vc14.VS2015;C:\ProgramData\chocolatey\bin;C:\ProgramData\chocolatey\lib\tinyxml2\lib;C:\OpenSSL-Win64\bin\
```

dirk-thomas, 2017-09-12:
Power shell is not supported atm. It would need to be added in the same way as `bash` for non-Windows system.

mikaelarguedas, 2017-09-12:
https://github.com/ament/ament_package/issues/58
https://github.com/ros2/ros2/issues/323

sloretz, 2018-03-08:
Powershell is no longer referenced in the tutorial. Closing in favor of the issues @mikaelarguedas  linked.

closed 2018-03-08 by sloretz