ros2/ros2 #404: r2b3 Lifecycle python client fails to start

sloretz, 2017-09-12:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - http://ci.ros2.org/view/packaging/job/packaging_windows/728/artifact/ws/ros2-package-windows-AMD64.zip
- Version or commit hash:
  - def304b4270b6b935b7d6fbbf7671634c2532452
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Source local_setup.bat and run this  command ([from this tutorial](https://github.com/ros2/ros2/wiki/Managed-Nodes))
```
ros2 run lifecycle lifecycle_service_client_py.py
```

#### Expected behavior
```
usage: lifecycle_service_client_py.py [-h]
                                      [--change-state-args {configure,cleanup,shutdown,activate,deactivate}]
                                      {change_state,get_state,get_available_states,get_available_transitions}
                                      node
```

#### Actual behavior
```
C:\Users\osrf>ros2 run lifecycle lifecycle_service_client_py.py
Traceback (most recent call last):
  File "C:\dev\ros2\Scripts\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "C:\dev\ros2\Lib\site-packages\ros2cli\cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2\Lib\site-packages\ros2run\command\run.py", line 73, in main
    return run_executable(path=path, argv=args.argv, prefix=prefix)
  File "C:\dev\ros2\Lib\site-packages\ros2run\api\__init__.py", line 55, in run_executable
    completed_process = subprocess.run(cmd)
  File "C:\Python36\lib\subprocess.py", line 403, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Python36\lib\subprocess.py", line 707, in __init__
    restore_signals, start_new_session)
  File "C:\Python36\lib\subprocess.py", line 992, in _execute_child
    startupinfo)
OSError: [WinError 193] %1 is not a valid Win32 application
```

#### Additional information

Same error if I drop the `.py`

```
C:\Users\osrf>ros2 run lifecycle lifecycle_service_client_py
Traceback (most recent call last):
  File "C:\dev\ros2\Scripts\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "C:\dev\ros2\Lib\site-packages\ros2cli\cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2\Lib\site-packages\ros2run\command\run.py", line 73, in main
    return run_executable(path=path, argv=args.argv, prefix=prefix)
  File "C:\dev\ros2\Lib\site-packages\ros2run\api\__init__.py", line 55, in run_executable
    completed_process = subprocess.run(cmd)
  File "C:\Python36\lib\subprocess.py", line 403, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Python36\lib\subprocess.py", line 707, in __init__
    restore_signals, start_new_session)
  File "C:\Python36\lib\subprocess.py", line 992, in _execute_child
    startupinfo)
OSError: [WinError 193] %1 is not a valid Win32 application
```

mikaelarguedas, 2017-09-12:
Note: it works if calling `Lib\lifecycle\lifecycle_service_client_py.py`

I think it comes from the fact that it is installed as [a program](https://github.com/ros2/demos/blob/a4d48aa57ba5752122c23bf79e0959052b96337e/lifecycle/CMakeLists.txt#L59-L62) but doesnt have an .exe and python files are not considered "Windows executables" on windows

dirk-thomas, 2017-09-12:
@sloretz Please move the ticket into the repo which contains the package (https://github.com/ros2/demos/tree/master/lifecycle).

@Karsten1987 FYI since you added this in ros2/demos#86.

sloretz, 2017-09-12:
moved to ros2/demos#173

closed 2017-09-12 by sloretz