ros2/ros2 #405: Debian Stretch Dependency Issues

qnetjoe, 2017-09-14:
## Bug report

**Required Info:**

- Operating System:
  Debian Stretch 9.1
- Installation type:
  - Binaries (installed per instructions @ [Wiki\Linux Install Debians](https://github.com/ros2/ros2/wiki/Linux-Install-Debians)
- Version or commit hash:
  - Beta3
- DDS implementation:
  - N\A
- Client library (if applicable):
  - N\A

#### Steps to reproduce issue
install per instructions @ [Wiki\Linux Install Debians](https://github.com/ros2/ros2/wiki/Linux-Install-Debians).

Dependency issue show during install

#### Expected behavior

ROS2 installs per instructions

#### Actual behavior

User gets an error due to dependencies

user@debian-stretch:~$ sudo apt install `apt list ros-r2b3-* 2> /dev/null | grep "/" | awk -F/ '{print $1}' | grep -v -e ros-r2b3-ros1-bridge -e ros-r2b3-turtlebot2- | tr "\n" " "`
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-r2b3-astra-camera : Depends: libboost-system1.58.0 but it is not installable
                         Depends: libboost-thread1.58.0 but it is not installable
 ros-r2b3-cartographer : Depends: libprotobuf9v5 but it is not installable
 ros-r2b3-cartographer-ros : Depends: libboost-iostreams1.58.0 but it is not installable
                             Depends: libboost-system1.58.0 but it is not installable
                             Depends: libpcl-common1.7 but it is not installable
                             Depends: libprotobuf9v5 but it is not installable
                             Depends: libwebp5 (>= 0.4.3) but it is not installable
 ros-r2b3-class-loader : Depends: libpocofoundation9v5 but it is not installable
 ros-r2b3-fastrtps : Depends: libssl1.0.0 (>= `1.0.2~beta3)` but it is not installable
                     Depends: libtinyxml2-2v5 (>= 2.0.2) but it is not installable
 ros-r2b3-image-tools : Depends: libopencv-highgui2.4v5 but it is not installable
 ros-r2b3-intra-process-demo : Depends: libopencv-highgui2.4v5 but it is not installable
 ros-r2b3-rmw-implementation : Depends: libpocofoundation9v5 but it is not installable
 ros-r2b3-rosidl-typesupport-c : Depends: libpocofoundation9v5 but it is not installable
 ros-r2b3-rosidl-typesupport-cpp : Depends: libpocofoundation9v5 but it is not installable
E: Unable to correct problems, you have held broken packages.
user@debian-stretch:~$ 

#### Additional information
Debian Stretch has newer versions of some of the libs. For Example:

libboost 1.62 (vs 1.58) is available by default
libpcl-common1.8 (vs 1.7) is available by default
libwebp6 (vs 5) is available by default
libssl1.0.2 (vs 1.0.2~beta3)is available by default
libtinyxml2-4/libtinyxml2.6.2v5 is available by default
libprotobuf10 (vs 9) is available by default
libopencv-highgui2.4-dev and libopencv-highgui2.4-deb0 are available by default but to not satisfy the dependency. 
libpocofoundation46 is available by default

dirk-thomas, 2017-09-14:
These Debian packages are targeting Ubuntu Xenial. Therefore they won't work on Debian as dependencies are different / have different versions. You will have to build from source on Debian.

nuclearsandwich, 2017-09-14:
Thanks for the complete report!

Our Debian packages are targeting Ubuntu Xenial and aren't supported on other distributions. The install guide fails to mention this, so I'll update it. You _might_ be able to use the source packages as a starting point for building binary deb packages for Stretch but it's probably simpler and faster to use the [Building from Source](https://github.com/ros2/ros2/wiki/Linux-Development-Setup) instructions. Sorry for the confusion!

nuclearsandwich, 2017-09-14:
[Updated](https://github.com/ros2/ros2/wiki/Linux-Install-Debians/c06a194c5a29ae3b747160ccbab90858c13ccb10) the Install from Debian packages instructions to mention the supported operating systems (just Ubuntu Xenial).

qnetjoe, 2017-09-14:
Are there any plans on supporting Stretch in the near term (pre 1.0)? Although it is not ROS2, Kinetic and Lunar both support Debian.

I am hoping to start creating nightly livecd builds and releasing them.

mikaelarguedas, 2017-09-14:
> Updated the Install from Debian packages instructions to mention the supported operating systems (just Ubuntu Xenial).

Also updated the landing install page accordingly

> Are there any plans on supporting Stretch in the near term (pre 1.0)? Although it is not ROS2, Kinetic and Lunar both support Debian.

Right know the testing infrastructure of ROS 2 is not as evolved as the ROS 1 one, so we keep the number of platforms restricted but with enough variety to be confident that building from source on other flavors (and more recent than Xenial) should go smoothly.
I don't think that we will be building Stretch binaries in the very near future, but @dirk-thomas would have a better idea of if and when this could happen

If you happen to build ROS 2 from source on Stretch and notice anything missing / mismatching in the installation instructions please let us know! We used to have a Debian section on the installation from source page but we were able to merge them when we migrated to Xenial. We are happy to update them to support Debian users.

dirk-thomas, 2017-09-14:
Since ROS 2 uses the same infrastructure as ROS 1 to build Debian packages it will be easy to enable support for Debian. As @mikaelarguedas pointed out we haven't done any testing on it and try to keep the extra effort low. We might enable one Debian platform for the next release - depending on how the release planning moves forward.

I will close this ticket for now. Please feel free to continue commenting on it.

closed 2017-09-14 by dirk-thomas

qnetjoe, 2017-09-14:
Thank you @dirk-thomas , please let me know what I can do to aid in getting support for Stretch. I am more than willing to test if needed.

dirk-thomas, 2017-09-14:
> please let me know what I can do to aid in getting support for Stretch. I am more than willing to test if needed.

Building from source, running the tests locally, and trying to use ROS 2 on Debian would be great. And creating issues / pull requests for anything you are running into.

kamalpandey1993, 2018-01-10:
I tried building from source in the `debian stretch armhf` . Everything went smoothly except while resolving dependencies `$rosdep install --from-paths src --ignore-src --rosdistro lunar -y`, it gave error for package sbcl showing:
`Package sbcl is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  sbcl-doc sbcl-source
E: Package 'sbcl' has no installation candidate
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install sbcl] failed
`
Apart from that libtinyxml2-4.0.1 issue [https://answers.ros.org/question/278733/rospack-find-throws-exception-error/](url) is found till now. I will update if i found anything anomalous .

mikaelarguedas, 2018-01-10:
@kamalpandey1993 please avoid cross-posting you already posted this question in 3 places.
This is a ros2 issue and you are trying to install ROS Lunar.
I'll reply soon on https://github.com/ros/rospack/issues/77 for a suggested fix and will cross reference it from https://github.com/hashworks/rospack/pull/1#issuecomment-324489541 and  https://answers.ros.org/question/278733/rospack-find-throws-exception-error/

mikaelarguedas, 2018-01-10:
@kamalpandey1993  see https://github.com/ros/rospack/issues/77#issuecomment-356619084

kamalpandey1993, 2018-01-18:
@mikaelarguedas sorry for the inconvenience and as you can see that i was desperate to know the solution but my intention here was not to trouble anyone but to remind them so that future users don't get stuck and this was more of an experience rather than a question that's why i shared it with the open source community.This problem was there since  October 2017. So i was a bit concerned. And last of all I am a new user to this open source community and to this software world so i apologize for this behavior. Its only possible because of this community. May the source be with you.

tfoote, 2018-01-18:
@kamalpandey1993 Please see the Etiquette section of our support guidelines: http://wiki.ros.org/Support#Etiquette