ros2/ros2 #406: Fail to build ROS2 on Mac

Christywl, 2017-09-18:
## Bug report

**Required Info:**

- Operating System:
  - Mac 10.12.6
- Installation type:
  - from source
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

According to the wiki https://github.com/ros2/ros2/wiki/OSX-Development-Setup, run the cmd `src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install` to build ROS2, but it failed.

#### Expected behavior
Could build ROS2 from source successfully
#### Actual behavior
Fail to build, an error occurs
#### Additional information
Run `src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install`: 
```
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake_ros: 0.0.3 (/Users/wanming/ros2_ws/install/share/ament_cmake_ros/cmake)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.6.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/local/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rcutils: 0.0.3 (/Users/wanming/ros2_ws/install/share/rcutils/cmake)
-- Found fastrtps_cmake_module: 0.0.3 (/Users/wanming/ros2_ws/install/share/fastrtps_cmake_module/cmake)
-- Found FastRTPS: /Users/wanming/ros2_ws/install/include  
-- Found rmw: 0.0.3 (/Users/wanming/ros2_ws/install/share/rmw/cmake)
-- Found rosidl_generator_cpp: 0.0.3 (/Users/wanming/ros2_ws/install/share/rosidl_generator_cpp/cmake)
-- Found rosidl_typesupport_introspection_c: 0.0.3 (/Users/wanming/ros2_ws/install/share/rosidl_typesupport_introspection_c/cmake)
-- Found rosidl_typesupport_introspection_cpp: 0.0.3 (/Users/wanming/ros2_ws/install/share/rosidl_typesupport_introspection_cpp/cmake)
-- Found ament_lint_auto: 0.0.3 (/Users/wanming/ros2_ws/install/share/ament_lint_auto/cmake)
-- Added test 'copyright' to check for copyright in CMake / C / C++ / Python code
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'uncrustify' to check C / C++ code style
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/wanming/ros2_ws/build/rmw_fastrtps_cpp
==> '. /Users/wanming/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' in '/Users/wanming/ros2_ws/build/rmw_fastrtps_cpp'
Scanning dependencies of target rmw_fastrtps_cpp
[  2%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/client_service_common.cpp.o
[  5%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/get_participant.cpp.o
[  8%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/get_publisher.cpp.o
[ 11%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/get_service.cpp.o
[ 14%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/get_subscriber.cpp.o
[ 17%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/identifier.cpp.o
[ 20%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/namespace_prefix.cpp.o
[ 22%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/qos.cpp.o
[ 25%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_init.cpp.o
[ 28%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_client.cpp.o
[ 31%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_compare_gids_equal.cpp.o
[ 34%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_count.cpp.o
[ 37%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_get_gid_for_publisher.cpp.o
[ 40%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_get_implementation_identifier.cpp.o
[ 42%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_guard_condition.cpp.o
[ 45%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_node.cpp.o
[ 48%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_node_names.cpp.o
[ 51%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_publish.cpp.o
[ 54%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_publisher.cpp.o
[ 57%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_request.cpp.o
[ 60%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_response.cpp.o
[ 62%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service.cpp.o
[ 65%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service_names_and_types.cpp.o
[ 68%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service_server_is_available.cpp.o
[ 71%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_subscription.cpp.o
[ 74%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_take.cpp.o
[ 77%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_topic_names_and_types.cpp.o
[ 80%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_trigger_guard_condition.cpp.o
[ 82%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_wait.cpp.o
[ 85%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_waitset.cpp.o
[ 88%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/ros_message_serialization.cpp.o
[ 91%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/type_support_common.cpp.o
[ 94%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/get_client.cpp.o
[ 97%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/demangle.cpp.o
[100%] Linking CXX shared library librmw_fastrtps_cpp.dylib
ld: library not found for -ltinyxml2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [librmw_fastrtps_cpp.dylib] Error 1
make[1]: *** [CMakeFiles/rmw_fastrtps_cpp.dir/all] Error 2
make: *** [all] Error 2

<== Command '. /Users/wanming/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/wanming/ros2_ws/build/rmw_fastrtps_cpp' with exit code '2'
<== Command '. /Users/wanming/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/wanming/ros2_ws/build/rmw_fastrtps_cpp' with exit code '2'
```
```
ZhaoMings-MBP:ros2_ws wanming$ brew info tinyxml2
tinyxml2: stable 5.0.1 (bottled), HEAD
Improved tinyxml (in memory efficiency and size)
http://grinninglizard.com/tinyxml2
/usr/local/Cellar/tinyxml2/5.0.1 (11 files, 148.8KB) *
  Poured from bottle on 2017-09-13 at 14:32:55
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/tinyxml2.rb
==> Dependencies
Build: cmake ✔
```

sivteck, 2017-09-18:
see #391 and #394 
specifically, https://github.com/ros2/ros2/issues/394#issuecomment-326600399

sivteck, 2017-09-19:
I am unable to build master from the source and getting the same error. 😕

Christywl, 2017-09-19:
Oh, bad news... Could someone take a look at this issue?

sivteck, 2017-09-20:
I am able to build and run master with TinyXML2 4.0.1 but not with TinyXML2 5.0.1, even with the https://github.com/eProsima/Fast-RTPS/pull/114

wjwwood, 2017-09-27:
I am unable to reproduce this, see:

https://github.com/ros2/ros2/issues/391#issuecomment-331981422

If someone else can reproduce it and help, that would be appreciated.

Please make sure you have SIP (https://support.apple.com/en-us/HT204899) disabled, and that tinyxml2 is installed and usable on your system (perhaps inquire which linking isn't working upstream).

dhood, 2017-10-03:
@Christywl if you have disabled SIP as wjwwood suggested and the issue persists, please try running `sudo xcode-select --install` and rebuilding from a clean workspace. (https://github.com/ros2/build_cop/issues/58 has some more context).

If that doesn't fix it please add  `--cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON` to your ament invocation so we can investigate the exact linker command.

Christywl, 2017-10-09:
Thanks for your help, it works fine for me after I disable SIP and run `sudo xcode-select --install`, so close this issue.

closed 2017-10-09 by Christywl