ros2/ros2 #407: Failed to build ros2 beta3 from source with package 'rosidl_generator_py'

gaoethan, 2017-09-18:
## Bug report

**Required Info:**

- Operating System: Ubuntu 16.04
  - 
- Installation type: From source
  - 
- Version or commit hash: ros2 beta3
  - 
- DDS implementation: Built with the default Fast-RTPS
  - 
- Client library (if applicable): N/A
  - 

#### Steps to reproduce issue

```
1. Follow the build environment preparation and get the latest ROS2 beta3 code according to https://github.com/ros2/ros2/wiki/Linux-Development-Setup.
2. Build using: src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install

```

#### Expected behavior
No build error occurred 
#### Actual behavior
+++ Building 'rosidl_generator_py'
Running cmake because arguments have changed.
==> '. /home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py/cmake__build.sh && /usr/bin/cmake /home/ethan/Myspace/rosdev/r2b3/src/ros2/rosidl/rosidl_generator_py -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/ethan/Myspace/rosdev/r2b3/install' in '/home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found ament_cmake_nose: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/ament_cmake_nose/cmake)
-- Found rosidl_cmake: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/rosidl_cmake/cmake)
-- Found rosidl_generator_c: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/rosidl_generator_c/cmake)
-- Found ament_lint_auto: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/ament_lint_auto/cmake)
-- Found rmw_implementation_cmake: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rmw_fastrtps_cpp: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/rmw_fastrtps_cpp/cmake)
-- Found fastrtps_cmake_module: 0.0.3 (/home/ethan/Myspace/rosdev/r2b3/install/share/fastrtps_cmake_module/cmake)
-- Found FastRTPS: /home/ethan/Myspace/rosdev/r2b3/install/include
-- Using single matching rosidl_typesupport_c: rosidl_typesupport_introspection_c
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.5")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "3.5")
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.5m
-- Using PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.5m.so
-- Found PythonExtra: .so
-- Using Python nosetests: /usr/bin/nosetests3 (1.3.7)
-- Added test 'copyright' to check for copyright in CMake / C / C++ / Python code
-- Added test 'flake8' to check Python code syntax and style conventions
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'pep257' to check Python code against some of the style conventions in PEP 257
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py
==> '. /home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make -j8 -l8' in '/home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py'
Scanning dependencies of target rosidl_generator_py__py
Scanning dependencies of target rosidl_generator_py__cpp
[  2%] Generating C code for ROS interfaces
[  6%] Generating Python code for ROS interfaces
[  6%] Generating C++ code for ROS interfaces
Traceback (most recent call last):
  File "/home/ethan/Myspace/rosdev/r2b3/src/ros2/rosidl/rosidl_generator_py/bin/rosidl_generator_py", line 40, in <module>
    sys.exit(main())
  File "/home/ethan/Myspace/rosdev/r2b3/src/ros2/rosidl/rosidl_generator_py/bin/rosidl_generator_py", line 36, in main
    return generate_py(args.generator_arguments_file, args.typesupport_impls.split(';'))
  File "/home/ethan/Myspace/robotics/ros2/install_isolated/rosidl_generator_py/lib/python3.5/site-packages/rosidl_generator_py/generate_py_impl.py", line 50, in generate_py
    assert os.path.exists(template_file), 'Could not find template: ' + template_file
AssertionError: Could not find template: /home/ethan/Myspace/rosdev/r2b3/src/ros2/rosidl/rosidl_generator_py/resource/_msg_support.entry_point.c.em
rosidl_generator_py__py/CMakeFiles/rosidl_generator_py__py.dir/build.make:85: recipe for target 'rosidl_generator_py/rosidl_generator_py/_rosidl_generator_py_s.ep.rosidl_typesupport_c.c' failed
make[2]: *** [rosidl_generator_py/rosidl_generator_py/_rosidl_generator_py_s.ep.rosidl_typesupport_c.c] Error 1
CMakeFiles/Makefile2:1317: recipe for target 'rosidl_generator_py__py/CMakeFiles/rosidl_generator_py__py.dir/all' failed
make[1]: *** [rosidl_generator_py__py/CMakeFiles/rosidl_generator_py__py.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  6%] Built target rosidl_generator_py__cpp
Scanning dependencies of target rosidl_generator_py__rosidl_generator_c
[ 10%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/various__functions.c.o
[ 10%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/constants__functions.c.o
[ 12%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/empty__functions.c.o
[ 14%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/nested__functions.c.o
[ 16%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/strings__functions.c.o
[ 18%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/primitives__functions.c.o
[ 20%] Linking C shared library librosidl_generator_py__rosidl_generator_c.so
[ 20%] Built target rosidl_generator_py__rosidl_generator_c
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py' with exit code '2'
<== Command '. /home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/ethan/Myspace/rosdev/r2b3/build/rosidl_generator_py' with exit code '2'

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

gaoethan, 2017-09-18:
By the way, My only difference is that I **vcs-import** the ros2-repos to a folder "_r2b3_" and the workspace "_src_" is a symbolic link to the folder _r2b3_.

mikaelarguedas, 2017-09-18:
It looks like you are using an outdated version of the code. The template _msg_support.entry_point.c.em has been renamed in beta3. Can you try to build in a fresh workspace?

mikaelarguedas, 2017-09-18:
Sorry unvoluntary close.
The template's new name in beta3 is "_msg_pkg_typesupport_entry_point.c.em"

gaoethan, 2017-09-19:
okay. thanks for your input, I logout and clean all the codebase to try 
again, and it works now, but I saw that there is a new exceptional 
message at the last, any comments ? as follows:

-- Symlinking: 
/home/ethan/Myspace/rosdev/r2b3/install/share/visualization_msgs/cmake/ament_cmake_export_libraries-extras.cmake
-- Symlinking: 
/home/ethan/Myspace/rosdev/r2b3/install/share/visualization_msgs/cmake/visualization_msgsConfig.cmake
-- Symlinking: 
/home/ethan/Myspace/rosdev/r2b3/install/share/visualization_msgs/cmake/visualization_msgsConfig-version.cmake
-- Symlinking: 
/home/ethan/Myspace/rosdev/r2b3/install/share/visualization_msgs/package.xml
Listing 
'/home/ethan/Myspace/rosdev/r2b3/install/lib/python3.5/site-packages/visualization_msgs//home/ethan/Myspace/rosdev/r2b3/build/visualization_msgs/rosidl_generator_py/visualization_msgs/__init__.py'...
Can't list 
'/home/ethan/Myspace/rosdev/r2b3/install/lib/python3.5/site-packages/visualization_msgs//home/ethan/Myspace/rosdev/r2b3/build/visualization_msgs/rosidl_generator_py/visualization_msgs/__init__.py'
Listing 
'/home/ethan/Myspace/rosdev/r2b3/install/lib/python3.5/site-packages/visualization_msgs/msg//home/ethan/Myspace/rosdev/r2b3/build/visualization_msgs/rosidl_generator_py/visualization_msgs/__init__.py'...
Can't list 
'/home/ethan/Myspace/rosdev/r2b3/install/lib/python3.5/site-packages/visualization_msgs/msg//home/ethan/Myspace/rosdev/r2b3/build/visualization_msgs/rosidl_generator_py/visualization_msgs/__init__.py'
-- [ament] Deploying: 
/home/ethan/Myspace/rosdev/r2b3/install/local_setup.bash
-- [ament] Deploying: /home/ethan/Myspace/rosdev/r2b3/install/local_setup.sh
-- [ament] Deploying: 
/home/ethan/Myspace/rosdev/r2b3/install/local_setup.zsh
-- [ament] Deploying: /home/ethan/Myspace/rosdev/r2b3/install/setup.bash
-- [ament] Deploying: /home/ethan/Myspace/rosdev/r2b3/install/setup.sh
-- [ament] Deploying: /home/ethan/Myspace/rosdev/r2b3/install/setup.zsh
-- [ament] Deploying: 
/home/ethan/Myspace/rosdev/r2b3/install/_order_packages.py

mikaelarguedas, 2017-09-26:
@gaoethan Follow-up on this: The new error messages should not impact the behavior of the system and the message package should still be found and importable without problem. Though the error message should disappear now that https://github.com/ament/ament_cmake/pull/110 has been merged

mikaelarguedas, 2017-09-26:
@gaoethan based on [your comment](https://github.com/ros2/ros2/issues/407#issuecomment-330413050) it looks like this has been fixed by building in a clean workspace so I'm going to close this. Feel free to comment here if the issue is not solved and we will reopen

closed 2017-09-26 by mikaelarguedas