ros2/ros2 #408: ROS 2 metapackages/variants support

mikaelarguedas, 2017-09-28:
This is replacing https://github.com/ros2/common_interfaces/pull/41 and https://github.com/ament/ament_cmake/issues/108

The question here is how to handle metapackages or group of packages in the ROS 2 ecosystem. 
Below is copied the original discussion from https://github.com/ros2/common_interfaces/pull/41

## Feature request

#### Feature description
It is convenient in some cases to be able to define a group of packages.
The main one being:
- Being able to install (through a debian package or rosinstall_generator-like tool) a single package pulling in a group of packages

#### Implementation considerations
Option 1:
Create a special type of package that doesn't allow any custom code and just declare dependencies on a set of packages (ala catkin_metapackage):
- pros:
  - close to what users are used to in ROS 1
  - stores the information close to where the code is (a repository with a set of packages can store the metapackage and all the code and dependencies of the repository is self-contained)
- cons:
  - needs custom logic to ensure that users don't add any code to the package that is used solely for dependency declaration
  - another type of package to document and for users to understand

Option 2:
Create a regular ament package just listing dependencies without preventing users from adding custom code and logic in it:
- pros:
  - no need for custom code on our side or to add another package type for users to learn
  - stores the information close to where the code is (a repository with a set of packages can store the metapackage and all the code and dependencies of the repository is self-contained)
- cons:
  - We may end up with a variety of "metapackages" with very different behaviors according to the maintainer's choice
  - It will be possible for a package to depend on metapackages and thus depend on much more than what the package actually need

Option 3:
Register a "variant" in a centralized place:
- pros:
  - no need to create and release a package just for dependency declaration
  - single place to browse all existing variants
  - only the infrastructure needs to know how to process these files and the users never have to interact with them directly
- cons:
  - dependency declaration lives far from the actual code
  - user may be confused to be able to install a package but not `find_package` it or depend on it

mikaelarguedas, 2017-09-28:
Original discussion:

_mikaelarguedas_
> this came up a few times and more often since we started shipping binaries.
> Creating this metapackage has 2 benefits:
> - allow people to install all common message packages with a single debian package
> - show an example of how metapackages look like in ROS2
>
> One remaining question though: we renamed all message packages from `foo_interfaces` to `foo_msgs` a while ago to simplify the mapping between ros1 and ros2 messages. Should this package be named common_msgs like its' ROS1 equivalent? If so should we also rename this repository accordingly ?

_wjwwood_
> I actually think we should avoid metapackages all together if we can. If you think about what common_msgs represents it is never something you want to depend on. It's only useful (imo) for indicating a grouping of packages or by providing a convenient apt-get target package to install. There's no need to ever actually build it, and really no need for it to have a cmake file (we restrict what it can contain anyways).
> 
> I'd rather see us address those use cases without metapackages which can never be used directly in code and make the concept of a "package" unnecessarily more complicated to explain.
> 
> For documentation purposes, i.e. grouping of packages, like the package clouds on the ROS 1 wiki, I'd rather see a tagging system or something like that.
> 
> For apt-get able packages, perhaps we should just bring back the concept of variants and have the build farm generate those directly.
> 
> For people pulling from source, the tools like rosinstall generator just need to be able to understand these variants and pull the packages that make them up.
> 
> Another alternative is to just make common_msgs a normal package, but again I think that just leads to useless packages that I wouldn't recommend people use because it encourages them to depend on more than they need.

_gbiggs_
> I agree with @wjwwood. Such a metapackage would encourage people to depend on more than they actually use, which makes maintenance harder.

_wjwwood_
> A little bit of history on metapackages, just for context:
> 
> We made them to have a representation for stacks from the rosbuild days. Eventually we converted the variants (as they used to be called), e.g. robot, desktop, viz, simulation, etc..., to metapackages as well.
> 
> But from my perspective it was because we wanted to consolidate the concept of variants and metapckages (old stacks) but we could not remove the existing metapackages like common_msgs. So perhaps this is a chance to go the other way, have just "packages" and have another high level concept which the build farm and rosinstall generator understand (maybe a yaml file or something in rosdistro) which is used to group packages in the wiki and used to give convenient targets for apt-get and rosinstall generator.

_gbiggs_
> It may help to clarify the two (I think) relevant use cases:
> 
> 1) Depending on a package that provides something you need. In this case, it is best to depend on as small a unit as possible, so depending on something like an "all commonly used messages" package would be bad.
> 
> 2) Installing a bunch of related packages so you can use their collective functionality, such as a navigation stack implementation. In this case, ease of use suggests having a single entity that you can apt-get to bring in the lot.
> 
> If the same entity is usable for both, then it will be used for both. Is it worth the effort to ensure a metapackage can only be used for the second? I think it is.

_mikaelarguedas_
> This resulted from an offline conversation that I thought (wrongly) was to specific to be integrated to a high-level migration discussion. What prompted me to start this was the appearance of questions or hacks to [install set of packages](https://github.com/osrf/docker_images/pull/68/files#diff-d3c0a2ef6b4a2afbae88e83169cbbad3R45) as well a people porting ROS 1 repositories by simply deleting the metapackages without providing an alternative.
> 
> I agree that we don't expect/want people to depend on metapackages. And that the main use case for `metapackages` is the ability to apt-get a single entity and bring in a group of things you want to install.
> I can see only very few corner cases, if any, where depending on a metapackage would make sense, and I don't think such use cases do justify the overhead of creating ROS 2 metapackages and the associated CMake boilerplate.
> 
> Variants or any alternative not relying on creating actual packages and allowing the infrastructure and tools to "group" a set of packages together will satisfy the requirement. I am not familiar with how variants were defined during the rosbuild days and am happy to hear about how it was done or how you guys think it could be done better in ROS 2.

_wjwwood_
> Some other info about how variants used to work:
> 
> - http://www.ros.org/reps/rep-0108.html
> - http://www.ros.org/reps/rep-0131.html
> 
> Basically they were defined in yaml and there was web service called `gen_rosisntall` that could convert those variants (as well as individual packages) to rosinstall files, pretty similar to what `rosinstall_generator` can do now.

_wjwwood_
> I couldn't find the original repository for `gen_rosinstall`, but here's a fork:
> 
> https://github.com/mbeards/gen_rosinstall
> 
> And some other mentions:
> 
> - https://answers.ros.org/question/9672/gen_rosinstallpy-generating-invalid-install-files/
[... cut ...] oo.
> 
> I still think the simplicity of using variants defined in the rosdistro outweighs the value of being able to do --end-with common_msgs. That use case could also be addressed if we implemented the groups, then you could build all of your local message packages with --end-with-group interface_packages or something similar.
> 
> I guess metapackages always seemed like a hack to me which was required to support migration from stacks, but if the group would rather keep them, that's fine too.

allenh1, 2017-09-28:
I'm personally a fan of option 3.

> dependency declaration lives far from the actual code

This isn't necessarily a bad thing. And this doesn't explicitly disable the likes of option 2, so this can be avoided if it is a concern.

> user may be confused to be able to install a package but not find_package it or depend on it

This is a good point, but perhaps the `find_package` command can be adapted to say something like this.

```
package not found (foo is a metapackage).
```

dirk-thomas, 2017-09-28:
To summarize what the different options mean:

* Option 1 is what we have in ROS 1. The effort to implement the `metapackages` in ROS 2 is minimal. The missing pieces can easily be added to `ament_cmake`. Other tools already handle `metapackages` - in case they need updates for ROS 2 they just have to do "the same thing" as for ROS 1.

* Option 2 is a simplication in terms of removing the special status of a metapackage. Basically a "normal" package takes over the role. While our recommendations what *not* to do in a metapackage (which is enforced in ROS 1) wouldn't change it wouldn't be enforced anymore. So if users want to depend on such packages form "normal" packages that would be allowed - at the end it is their choice to not follow recommendations.

* Options 3 introduces a new concept of entities on a per distro level. This would require additional implementation in order to be fully supported, just a few which came to my mind (there might be more):

  * `rosinstall_generator` needs to be updated to support these so that users can easily checkout a variant including its dependencies.
  * `rosdep` needs to support these since the variant names can be used as dependency keys.
  * The buildfarm needs to generate Debian packages for these variants which don't have any code / repository associated with it.
  * There is still a disconnect that a variant isn't represented by a package in the source space which might be very confusing. Concepts like "build all packages up to `variant-x`" wouldn't work in a build.

Since I don't think the effort for option 3 is doable for us atm I don't think it is a valid option for now. If we choose option 2 for now we can always implement option 1 or 3 in the future with an easy transitioning path.

Personally I prefer option 2 over option 1 since it simplifies the system (if people want to do not recommended stuff I don't mind - it should be their choice :boom:). If we pick option 2 we just need to make sure to document that metapackages intentionally don't exist in ROS 2.

wjwwood, 2017-09-28:
In general, I'm also ok with option 2, with the option in the future to pursue option 1 or 3 later.

----

> rosdep needs to support these since the variant names can be used as dependency keys.

I don't think this is true. Why would we let packages depend on variants? I don't believe `rosdep` needs to understand variants unless `rosdep` is somehow used by the build farm to generate the "dummy" deb packages that represent the variant.

dirk-thomas, 2017-09-28:
> > rosdep needs to support these since the variant names can be used as dependency keys.

> I don't think this is true. Why would we let packages depend on variants? I don't believe rosdep needs to understand variants unless rosdep is somehow used by the build farm to generate the "dummy" deb packages that represent the variant.

Good point. Normal packages shouldn't be able to depend on variants. :+1:

tfoote, 2017-09-29:
I'm generally ok with option 2. Most other packaging systems don't have two classes of packages.

The one place that the metapackages are quite nice is in the documentation where we can auto index peers in metapackages. However I think that this sort of behavior might be better captured via using tags or other metadata/annotation in the package.xml files.

dirk-thomas, 2017-09-29:
> The one place that the metapackages are quite nice is in the documentation where we can auto index peers in metapackages. However I think that this sort of behavior might be better captured via using tags or other metadata/annotation in the package.xml files.

We can keep the `metapackage` tag in the export section of the manifest for this.

dirk-thomas, 2019-09-05:
Closing since we are not planning to add a custom "metapackage" package type.

dirk-thomas, 2019-09-05:
The ROS variants defined in the REPs are still present in the form of "normal" ROS packages: https://github.com/ros2/variants

closed 2019-09-05 by dirk-thomas