ros2/ros2 #409: Requiring users to disable SIP on OS X

dhood, 2017-09-30:
We have a requirement of System Integrity Protection being disabled on OSX which @wjwwood explained is necessary so that we can extend the DYLD_LIBRARY_PATH

This had been left out of the installation instructions which would explain why users are having issues building from source on OS X lately, namely since tinyxml2 >=5.0.0 which required some workarounds in fastrtps that [export the target by name](https://github.com/eProsima/Fast-RTPS/pull/114/files). Without SIP disabled, the target can't be resolved at link time for packages such as `rmw_fastrtps_cpp` which have references to `-ltinyxml2`.

Straightforward fix is to add disabling SIP to the instructions, but since @mikaelarguedas has a workaround that would allow `rmw_fastrtps_cpp` to be linked, @wjwwood expressed curiosity which other packages would require it to be disabled, and if its little effort we might not need it anymore.

To that end I'm detailing the packages that encountered issue building with SIP enabled:
- `rmw_fastrtps_cpp` (`-ltinyxml2`)
- `demo_nodes_cpp_native` (presumably from https://github.com/ros2/rmw_fastrtps/pull/148/files#diff-b3150e41248fe311273fde9f561b205cR23)

From https://github.com/ros2/rmw_fastrtps/issues/149 the latter doesn't seem trivial to resolve, so I'll close this out and update the install instructions. This ticket is mostly for future reference.

<details>
  <summary>rmw_fastrtps_cpp error</summary>
  [  2%] Linking CXX shared library librmw_fastrtps_cpp.dylib

/usr/local/Cellar/cmake/3.9.2/bin/cmake -E cmake_link_script CMakeFiles/rmw_fastrtps_cpp.dir/link.txt --verbose=1

/usr/local/opt/ccache/libexec/c++  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -dynamiclib -Wl,-headerpad_max_install_names  -o librmw_fastrtps_cpp.dylib -install_name @rpath/librmw_fastrtps_cpp.dylib CMakeFiles/rmw_fastrtps_cpp.dir/src/client_service_common.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/demangle.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_client.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_participant.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_publisher.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_service.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_subscriber.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/identifier.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/namespace_prefix.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/qos.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_init.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_client.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_compare_gids_equal.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_count.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_get_gid_for_publisher.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_get_implementation_identifier.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_guard_condition.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_node.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_node_names.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_publish.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_publisher.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_request.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_response.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service_names_and_types.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service_server_is_available.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_subscription.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_take.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_topic_names_and_types.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_trigger_guard_condition.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_wait.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_waitset.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/ros_message_serialization.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/type_support_common.cpp.o -Wl,-rpath,/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_introspection_c/lib -Wl,-rpath,/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_introspection_cpp/lib -Wl,-rpath,/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rcutils/lib -Wl,-rpath,/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rmw/lib -Wl,-rpath,/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_generator_c/lib /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/fastcdr/lib/libfastcdr.1.0.7.dylib /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/fastrtps/lib/libfastrtps.1.5.0.dylib /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_introspection_c/lib/librosidl_typesupport_introspection_c.dylib /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_introspection_cpp/lib/librosidl_typesupport_introspection_cpp.dylib /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rcutils/lib/librcutils.dylib /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rmw/lib/librmw.dylib /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_generator_c/lib/librosidl_generator_c.dylib /usr/local/opt/openssl/lib/libssl.dylib /usr/local/opt/openssl/lib/libcrypto.dylib -ltinyxml2 

ld: library not found for -ltinyxml2
</details>

<details>
  <summary>demo_nodes_cpp_native error</summary>
```
[ 50%] Building CXX object CMakeFiles/talker.dir/src/talker.cpp.o

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/builtin_interfaces/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_generator_cpp/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_interface/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_c/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_cpp/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_generator_c/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_introspection_c/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rosidl_typesupport_introspection_cpp/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rcl_interfaces/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rcutils/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rmw/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rcl/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rclcpp/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/std_msgs/include -I/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rmw_fastrtps_cpp/include  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk   -Wall -Wextra -Wpedantic -std=gnu++14 -o CMakeFiles/talker.dir/src/talker.cpp.o -c /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/src/ros2/demos/demo_nodes_cpp_native/src/talker.cpp

In file included from /Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/src/ros2/demos/demo_nodes_cpp_native/src/talker.cpp:23:

/Users/osrf/jenkins/workspace/nightly_osx_repeated/ws/install/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/get_participant.hpp:18:10: fatal error: 'fastrtps/participant/Participant.h'

      file not found

#include "fastrtps/participant/Participant.h"
```
</details>

dhood, 2017-10-03:
Actually, we were too quick to conclude that SIP was the issue. Disabling SIP didn't fix it; running `sudo xcode-select --install` was what fixed it (see https://github.com/ros2/build_cop/issues/58). The machine has been able to build jobs with SIP enabled, so we might be fine to remove it from the wiki. I don't have a complete understanding of the circumstances in which you'd need it to be disabled, maybe it would impact users building overlays? What do you think @wjwwood? (p.s. the reason my mac didn't have SIP disabled is because it's 10.10 and SIP is >=10.11)

wjwwood, 2017-10-03:
That's strange, I haven't seen that before.

I also did not see any issue on my fresh mac book install (I did it this weekend).

I guess we could remove the recommendation in that case.

dhood, 2017-10-03:
I think that particular case of "things 95% working" came from dosa being _upgraded_. Perhaps somehow xcode got upgraded but not the command-line tools so it was in a weird state.

OK I'll remove the SIP disabling step in the instructions.

closed 2017-10-03 by dhood

dhood, 2017-10-03:
OK, still learning more things about the role that SIP plays here (hopefully this will be the last update...)

SIP _does_ need to be disabled, but apparently not for building/linking, but at runtime, e.g. running tests.

Users without SIP disabled will run into the following situation:
- They can manually run executables fine, provided that they've sourced the appropriate setup files.
- They cannot run tests through ament, because even though the setup files of test dependencies have been sourced by ament, `DYLD_LIBRARY_PATH` doesn't get passed through to launch files.

Some users state they can run executables directly but not through `ros2 run`, which is probably the same cause underneath.

So the conclusion from all of this from my perspective is:
- Build and link issues likely caused by something off in the paths of clang; was fixed on Dosa by `sudo xcode-select --install`, but [manually updating the paths](https://github.com/Automattic/node-canvas/issues/927#issuecomment-310181808) can also work.
- Runtime library loading issues likely caused by SIP not allowing `DYLD_LIBRARY_PATH` to be passed through to other processes; disabling SIP fixes it.

I'll report this back to users that ran into either of these issues lately.

dhood, 2017-10-03:
OK, still learning more things about the role that SIP plays here (hopefully this will be the last update... Dosa has done a standard CI job and come out green.)

SIP _does_ need to be disabled, but apparently not for building/linking, but at runtime, e.g. running tests.

Users without SIP disabled will run into the following situation:
- They can manually run executables fine, provided that they've sourced the appropriate setup files.
- They cannot run tests through ament, because even though the setup files of test dependencies have been sourced by ament, `DYLD_LIBRARY_PATH` doesn't get passed through to launch files.

Some users state they can run executables directly but not through `ros2 run`, which is probably the same cause underneath.

So the conclusion from all of this from my perspective is:
- Build and link issues likely caused by something off in the paths of clang; was fixed on Dosa by `sudo xcode-select --install`, but [manually updating the paths](https://github.com/Automattic/node-canvas/issues/927#issuecomment-310181808) can also work.
- Runtime library loading issues likely caused by SIP not allowing `DYLD_LIBRARY_PATH` to be passed through to other processes; disabling SIP fixes it.

I'll report this back to users that ran into either of these issues lately, and make sure disabling SIP is also listed as a runtime requirement for users using binaries.

dhood, 2017-10-12:
For some still unexplained reason, disabling SIP on Dosa (after being upgraded to High Sierra) makes its builds run much slower. Because of that I have been investigating what it would take to get it to run jobs with SIP enabled.

These are the things that were stopping it from running tests with SIP enabled. Remember that SIP only strips the `DYLD_LIBRARY_PATH` for processes launched by a "protected" process e.g. things in `/usr/bin` (but not things in `/usr/local/bin`):

1. `subprocess` calls using a shell [such as this one in `ament_tools`](https://github.com/ament/ament_tools/blob/f8585a4ef4c4d20727d5ba9f13a662ed5e819d51/ament_tools/verbs/build_pkg/cli.py#L253) will always [invoke `/bin/sh`](https://github.com/python/cpython/blob/fb4c28c032e26b3cdbe67eae3769d45207ac3507/Lib/subprocess.py#L1223). I change subprocess.py to use the `SHELL` environment variable if set. @wjwwood suggested we might be able to avoid using the `shell=True` option instead but I haven't looked into that.
1. `/bin/sh` is selected by `/usr/bin/env sh` in [our generated setup scripts](https://github.com/ament/ament_tools/blob/f8585a4ef4c4d20727d5ba9f13a662ed5e819d51/ament_tools/build_type.py#L233). I added a `/usr/local/bin/sh` copy of `/usr/bin/sh` (not suggesting this as best practice of course) so it gets used instead.
1. `/usr/bin/make` is invoked when running tests. This can be fixed by installing make from brew (`brew install make --with-default-names` so `/usr/local/bin/make` [is used instead](https://github.com/ament/ament_tools/blob/2ed6995fbf52357d708f061e32b1a570eecfc40c/ament_tools/build_types/cmake_common.py#L23).

These workarounds are not ideal for users that want to use the default `make`, `sh` executables; @wjwwood mentioned we might be able to avoid the reliance on the `DYLD_LIBRARY_PATH` if we link against libraries specified with paths instead of names.

@wjwwood will report back with whether or not disabling SIP makes builds slow on their new High Sierra installation (not upgrade), which will give us some more info on the urgency of supporting SIP-enabled builds.

wjwwood, 2017-10-16:
I did not experience a slow down by just disabling SIP:

With SIP:

```
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install  1867.17s user 262.26s system 124% cpu 28:28.57 total
```

Without SIP:

```
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install  1922.47s user 352.97s system 142% cpu 26:36.21 total
```

I do not have ccache installed either.

dhood, 2017-11-27:
Dosa is still slower with SIP disabled even after a reinstall of High Sierra, so this topic has come up again.

New update on my understanding of SIP:
Before a test executable runs from ament, it has to go through a number of process invocations (listed above with greater detail), namely `python` (`ament.py`), `sh` called from `subprocess.py`, `sh` when the setup file is sourced, and `make` from the ctest invocation.

Originally I thought that if *any* step in the chain was a protected process, the test executable would be considered a protected process and have the `DYLD_LIBRARY_PATH` stripped. This is not the case: it is not an issue if _one_ of the links in the chain is a protected process, provided that nothing has been done to the `DYLD_LIBRARY_PATH` by that point in the chain. For example, `subprocess.py` calling to `/bin/sh` in step 1 will strip the env var, but step 2+ will still be able to repopulate it if there are no other protected processes in the chain.

Until https://github.com/ros2/ci/pull/107 we didn't care if the `DYLD_LIBRARY_PATH` got stripped at step 1, as long as sourcing the local_setup.sh can take effect, and we can [do what we need to for tests that use memory_tools](https://github.com/ros2/rcl/blob/d433ee193c749bf31dd46666b71c63df95a84cf0/rcl/test/memory_tools/CMakeLists.txt#L12) (I'm a bit uncertain on the details of that step). At the time of that PR I thought that we needed all links in the chain to be unprotected processes, which is why the `subprocess` workaround mentioned above was in place, allowing that PR to work even on Dosa with SIP disabled. If we find a different approach to https://github.com/ros2/ci/pull/107, then to enable SIP I think we only need workarounds 2 and 3 listed above, and they are not extreme workarounds. If however we want to support users enabling SIP without those workarounds (e.g. so they can use the default protected `make` and `sh` processes) we will probably need some changes to the code base to avoid using `DYLD_LIBRARY_PATH`.

nuclearsandwich, 2017-11-28:
After re-imaging Dosa and running a few test builds like [ci_osx#2923](http://ci.ros2.org/job/ci_osx/2923/timestamps/?elapsed=HH:mm:ss&appendLog), and [ci_osx#2925](http://ci.ros2.org/job/ci_osx/2925/timestamps/?elapsed=HH:mm:ss&appendLog) we confirmed that this was still an issue.

Running a build with SIP disabled finishes the build step at an acceptable 30:41 ([ci_osx#2928](http://ci.ros2.org/job/ci_osx/2928/timestamps/?elapsed=HH:mm:ss&appendLog)). When looking into the issue this morning I found out that the command to disable SIP accepts some parameters so that individual components of system integrity protection can be enabled and disabled separately. My hope was that the SIP component responsible for the performance hit when disabled and the component responsible for 
masking the DYLD_ variables were distinct and if not I could at least verify which SIP component was causing the issue for further investigation.

I found no official documentation regarding the components of SIP or the complete `rcutils` interface but I did turn up [this question on StackOverflow](https://stackoverflow.com/questions/35113709/cannot-disable-dtrace-restrictions-nvram-protection-and-debugging-restrictions#35192945) which enumerates a few options.

This morning I partially renabled SIP using the following commands in recovery mode.

```
# csrutil disable
# csrutil enable --without debug
```

When the machine restarted here was the output of `csrutil status`

```
System Integrity Protection status: enabled (Custom Configuration).

Configuration:
	Apple Internal: disabled
	Kext Signing: enabled
	Filesystem Protections: enabled
	Debugging Restrictions: disabled
	DTrace Restrictions: enabled
	NVRAM Protections: enabled
	BaseSystem Verification: enabled

This is an unsupported configuration, likely to break in the future and leave your machine in an unknown state.
```

I tested whether DYLD_ variables were still masked with the command 

```
DYLD_LIBRARY_PATH=foo /bin/sh -c 'echo $DYLD_LIBRARY_PATH'
foo
```

I then ran a timed build of ROS 2 with `time python3 src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install`

 __real:__ 41m24.495s
 __user:__ 63m50.512s                                                                                            
__sys:__ 13m35.241s

I'm running another build now to verify this wasn't a ccache fluke.

Concurrent with my efforts @clalancette also tested builds with and without SIP enabled on a mac mini recently updated to High Sierra and replicated the slower build with SIP disabled.

His investigation brought him to Console.app where he found huge numbers of `MacOS error: -67062` which come from taskgated, a daemon related to code signature verification and access control.

I haven't tried it yet, but with system integrity protection completely disabled it may be possible to disable the taskgated process entirely which may resolve our build performance issue but results in a pretty unnatural MacOS system for end users to conform to.

nuclearsandwich, 2017-11-28:
@wjwwood despite not seeing any performance impact, do you see any reference to these errors in your system's Console.app?

wjwwood, 2017-11-28:
I don't see that exact error in my `Console.app` after doing a full build. I also looked for other `MacOS error:` like errors.

scpeters, 2018-03-05:
This may be unrelated, but I just recently upgraded my laptop to high sierra from mountain lion and ran into an issue with running `make test` in the build folder of a catkin workspace and being unable to resolve libraries. I think this is related to DYLD and SIP, but I found that running `ctest` seems to work instead.

nuclearsandwich, 2018-03-05:
@scpeters it seems likely related. Especially if `which make` on your system turns up a binary that's distributed with MacOS. Protected binaries strip certain environment variables which means that if `/usr/bin/make` is invoked by the build process it will strip these variables. One workaround for make is to install GNU make from Homebrew and then symlink it as `make` higher in path precedence than the system make.

You'll probably also run into issues with /bin/sh. In general we can't recommend replacing /bin/sh with an unprotected binary but dash, or bash in sh compatibility mode might be able to fill those shoes.

erwincoumans, 2018-08-22:
Disabling SIP to run ROS2 is unacceptable for many users. Are there still efforts to avoid relying on DYLD_LIBRARY_PATH, and using /usr/local.. etc?

mikaelarguedas, 2018-08-22:
We understand that it is undesired for some users, you can see [this open issue](https://github.com/ros2/ros2/issues/457) tracking the possible improvements to allow users to keep SIP enabled.

We are not actively working on this issue but will happily review contributions

wjwwood, 2018-08-22:
Relying on `/usr/local/lib` being on the library path is also not acceptable, since you cannot have more than one working copy at a time. It should be possible to address all the issues by using RPATH instead. Currently I run with SIP enabled, the only thing that doesn't work is some of the tests which call `/usr/bin/bash` (I think that's right, but it's from memory, either bash or make) do not work.

zmk5, 2019-07-10:
https://www.reddit.com/r/MacOS/comments/caiue5/macos_catalina_readonly_file_system_with_sip/

According to new people on the Catalina Beta, disabling SIP will render the file-system as read only.

wjwwood, 2019-07-25:
@zmk5 I could be wrong, but my interpretation of that thread is that only the separate partition for `/System` is read-only, and it's read-only whether or not SIP is disabled. They're just complaining that disabling SIP doesn't also make that read-write too.

We don't need to modify `/System`, we only need to have homebrew working (which it should) and to have builtin binaries (like `/bin/bash` or `/bin/sh` etc...) to not wipe `DYLD_LIBRARY_PATH`.

wjwwood, 2019-07-26:
Also, my colleague @sloretz just pointed out to me that you may disable part of SIP (the debugging restrictions) and that might be a compromise for us:

https://github.com/rbenv/rbenv/issues/962#issuecomment-275858404

Again, with SIP fully enabled, most things will "just work", the shared library path thing only bites us if we're running tests (due to `make`) or if you're using py-env because it uses `sh` to control where the python executable comes from.

nuclearsandwich, 2019-07-26:
> Also, my colleague @sloretz just pointed out to me that you may disable part of SIP (the debugging restrictions) and that might be a compromise for us:

This compromise may be perfect for our CI hosts assuming the extreme performance hit we encountered disabling SIP on High Sierra has been fixed in a more recent version of macOS. When we were going through the whole saga trying to get the CI host Dosa back online partially disabling SIP allowed tests to run but didn't resolve the perf issues.