ros2/ros2 #410: Implementing ros 2 into an already existing vs2015 project (windows 10)

keimura, 2017-10-03:
I am currently trying to implement ros2 with an openvr project I have in windows 10. But I am finding a lack of documentation on how to bring in an already existing vs2015 project into the ros environment. Is there an example that I can reference or does anyone have any advice on how to about doing this? 

I took a look at the example code for windows but am confused as to why they structured the directories as they did and if I need to follow their structure. I'm currently under the impression that all I need to make are cmakelists.txt and package.xml files that  satisfy both the original project and ros.

dirk-thomas, 2017-10-04:
> I am currently trying to implement ros2 with an openvr project I have in windows 10. But I am finding a lack of documentation on how to bring in an already existing vs2015 project into the ros [...]

Each ROS package is a separate CMake project. As such for each package CMake generates a separate Visual Studio project file. So ROS doesn't do anything special - you might be able to find examples for plain CMake for your use case.

> I took a look at the example code for windows but am confused as to why they structured the directories as they did and if I need to follow their structure.

The file system layout of installed packages is the same as on Linux which follows the [FHS](https://wiki.linuxfoundation.org/lsb/fhs). Since Windows doesn't have its own standard and it makes it easier ROS packages use the same layout across all platforms.

> I'm currently under the impression that all I need to make are cmakelists.txt and package.xml files that satisfy both the original project and ros.

Yes, that should be sufficient in order to depend on existing ROS packages and to place your own code into this new package.

mikaelarguedas, 2017-10-11:
closing due to no response

closed 2017-10-11 by mikaelarguedas