ros2/ros2 #411: should we recommend or suggest using master?

wjwwood, 2017-10-12:
I just reverted this change:

https://github.com/ros2/ros2/wiki/Linux-Development-Setup/_compare/e6fe9c4cf8ec3c68b54809fee77411631027c722...9269263d81b076be7fc92570b894d020e15335f7

@patrickelectric We appreciate you trying to improve the docs, but we were suggesting master on purpose atm, but we could change that. We'll discuss it here and if we decide to I'll under my revert and restore your change in some fashion.

If we choose to keep the suggestion or change the recommendation, then we should also update all the wiki pages to say something similar (i.e. the macOS and Windows instructions too).

We've commonly had this problem that we don't want to support the beta's (or at least a common response is "have you tried master?"), so should we switch to always suggesting master for our from source builds or should we just add a note mentioning that master might be preferable in some cases?

patrickelectric, 2017-10-12:
@wjwwood release-latest is impossible to compile in Arch Linux. I am running master here without problems.

wjwwood, 2017-10-12:
@patrickelectric I understand that, but that doesn't change what we want to recommend or what we want to support.

dirk-thomas, 2017-10-12:
We should continue to recommend the latest release tag. That is the state we have tested. We can't guarantee that the current master works well (while aiming for that to be the case). A note how to use master could be added though for those which want / need to try.

patrickelectric, 2017-10-13:
Ok... So, I need to contribute with the latest release ? The correction is already in master,  I'll just do a cherry-pick and create a PR ?
Because it appears that the latest tag don't receive bug corrections but master do and this doesn't make sense to me. The master receive new features and both need to receive bug corrections.

dirk-thomas, 2017-10-13:
The beta releases are only "snapshots" and only have a tag not a branch because we don't ship any "updated" betas. So any contribution goes to the default branch (usually `master`) and will therefore be available in the next release.

dirk-thomas, 2017-10-13:
Once we get to non-beta releases they will get proper branches which can receive bugfixes.

wjwwood, 2017-10-13:
@patrickelectric until we have a supported release (i.e. a release we will take patches for and push bugfix releases), then I think the best we can do is recommend the latest tag, and then give a note (as you did in your original wiki change) that users may want to try the master branch if they run into issues. We still need to do this on each of the wiki pages, rather than just on the Linux one.

wjwwood, 2017-10-13:
I made a change to all the source installation wiki pages similar to this one:

https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/5143276ba053dbb1b45554709820b0ed89e6f28d...f2ebf9836b7f09c2e31fd04f625701334addafe4

Thanks for bringing the issue up originally @patrickelectric.

closed 2017-10-13 by wjwwood

patrickelectric, 2017-10-14:
@wjwwood your patch doesn't make sense because you have repositories without master tag/branch...
Just say: use ros2.repos of the master branch.

mikaelarguedas, 2017-10-16:
@patrickelectric it looks to me that the patch referenced and what you suggest are strictly equivalent.
If I follow the instructions in the patch submitted by wjwwood and "replacing `release-latest` in the url above with `master`" I end up with 
`wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos` that is the url of the ros2.repos file of the master branch.
If this is unclear, could you comment here with a suggestion of how we could rephrase this to avoid user confusion ?

patrickelectric, 2017-10-16:
@mikaelarguedas But https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos don't have `release-latest` now..

mikaelarguedas, 2017-10-16:
`release-latest` is a tag that always maps to our latest release, in this case `release-beta3`

patrickelectric, 2017-10-16:
The problem continue.. Like I said before.. Some packages with `release-latest` don't have master branch.
Replacing all `release-beta3` with `master` results in:
```bash
 ✘ patrick@dell  /tmp/ros  vcs-import src < ros2.repos
...............E...............E..........E.E.
=== src/ament/ament_cmake (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ament/ament_index (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ament/ament_lint (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ament/ament_package (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ament/ament_tools (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ament/googletest (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ament/osrf_pycommon (git) ===

HEAD is now at 3ef6d6a... Merge pull request #41 from osrf/flake8_plugins
=== src/ament/uncrustify (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/eProsima/Fast-CDR (git) ===

HEAD is now at 0860e7c... Merge pull request #11 from dirk-thomas/patch-1
=== src/eProsima/Fast-RTPS (git) ===
Cloning into '.'...
Note: checking out 'a3031901795bff023f4bece4623f6778fd09d3ec'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at a3031901... Refs #2213. Fixed error with security and besteffort
=== src/ros/class_loader (git) ===

HEAD is now at 7d7bf4e... Fully resolved placeholder to avoid confusion (#57)
=== src/ros/console_bridge (git) ===

HEAD is now at 8c3c625... Don't link against pthread on Android (#47)
=== src/ros2/ament_cmake_ros (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/common_interfaces (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/demos (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/example_interfaces (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/examples (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/geometry2 (git) ===
Could not checkout ref 'master': error: pathspec 'master' did not match any file(s) known to git.
=== src/ros2/launch (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/orocos_kinematics_dynamics (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/poco_vendor (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rcl (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rcl_interfaces (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rclcpp (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rclpy (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rcutils (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/realtime_support (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rmw (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rmw_connext (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rmw_fastrtps (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rmw_implementation (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rmw_opensplice (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/robot_model (git) ===
Could not checkout ref 'master': error: pathspec 'master' did not match any file(s) known to git.
=== src/ros2/robot_state_publisher (git) ===
Could not checkout ref 'master': error: pathspec 'master' did not match any file(s) known to git.
=== src/ros2/ros1_bridge (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/ros2cli (git) ===

Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rosidl (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rosidl_dds (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/rosidl_typesupport (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/sros2 (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/system_tests (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/tinyxml_vendor (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/tlsf (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/urdfdom (git) ===
Cloning into '.'...
Switched to a new branch 'master'
Branch master set up to track remote branch master from origin.
=== src/ros2/urdfdom_headers (git) ===
Cloning into '.'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
=== src/ros2/vision_opencv (git) ===
Could not checkout ref 'master': error: pathspec 'master' did not match any file(s) known to git.
```

tfoote, 2017-10-16:
@patrickelectric The substitution is in the url of the file not all the urls in the contents of the repos file. 

`https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos` becomes `https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos`

patrickelectric, 2017-10-17:
Oh, my fault, sorry for this misunderstanding.
Thanks @tfoote.

mikaelarguedas, 2017-10-17:
You should not try to change anything in the repos file itself, but adapt the URL where the file is fetched from. So the URL `https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos` becomes `https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos` if you replace `release-latest` with `master` in the URL.

The resulting commands would be:
```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
# wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
vcs import src < ros2.repos
```

mikaelarguedas, 2017-10-17:
Thanks for testing @patrickelectric, as said in earlier comment, if you have an idea of how to rephrase this to make it less confusing we're open to suggestions

patrickelectric, 2017-10-17:
Thanks @mikaelarguedas, @wjwwood did a good job, was my fault.