ros2/ros2 #413: Build from Source Fails

gnorthfork, 2017-10-13:
Not sure if this is the correct place to post this.  If not, my apologies.

OS: Windows 10
Framework:  VS2017,  with latest updated applied.
Install procedure:  https://github.com/ros2/ros2/wiki/Windows-Development-Setup

Followed the steps as instructed, no issues, until time to build ROS2 source.   Unresolved external symbols  when trying to link /build/builtin_interfaces/Release/builtin_interfaces_s__rosidl_typesupport_c.lib.

Anyone else having this issue?  All comments appreciated.

[build.log](https://github.com/ros2/ros2/files/1384200/build.log)

## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

dirk-thomas, 2017-10-13:
ROS 2 currently only supports Visual Studio 2015. Can you reproduce the same problem with that version?

gnorthfork, 2017-10-14:
Thanks for the suggestion.  Just downloaded VS2015 Professional.  

1. Removed the /dev/ros2 and reloaded from GIT.  
2. Open command prompt as administrator, and executed the VS2015 shell command.
3. From dos shell, verified all environment variables referenced vs2015 and not vs2017
4. Ran python src\ament\ament_tools\scripts\ament.py build

Unfortunately, same error at same place in the build process.

Any other ideas?

clalancette, 2017-10-18:
Would you mind attaching another log with your latest output?  We just want to check a few things with the VS2015 setup.  Thanks.

gnorthfork, 2017-10-19:
Sorry, I gave up and uninstalled ROS2

clalancette, 2017-10-19:
All right, thanks for the response.  Since we don't have a reproducer for now, I'm going to close this out.  If you get a chance to try this out again in the future, feel free to re-open this bug and we'll try to track it down.

closed 2017-10-19 by clalancette