ros2/ros2 #414: Incorrect information in setup.py files, use pytest in the future

dirk-thomas, 2017-10-20:
ROS 2 Python packages commonly contain the following in their `setup.py` file:

```
test_suite='test',
```

For an example see [ament_package](https://github.com/ament/ament_package/blob/44d3e8905c6bbc0d5dbf265c82d3ff540de4e429/setup.py#L26)

But most of them don't contain a `unittest` suite with the name `test` (which is that this attribute actually represents). Since `ament_tools` doesn't invoke `setup.py test` the wrong attribute isn't used but if you call that command the result is basically wrong. Anyway I think it should be removed from all Python packages.

---

Instead I would like to propose to add the following instead:

```
tests_require=['pytest'],
```

This declares a test dependency on `pytest`. In the future I would like to invoke `pytest` rather than `nose` for testing Python packages for the following reasons:

* `pytest` is much more actively developed than `nose` ([last commit](https://github.com/nose-devs/nose) from March, 2016)
* `pytest` has many plugins which we can utilize (e.g. `repeat`, `rerunfailures`)
* easy and fast to contribute (e.g. [this](https://github.com/pytest-dev/pytest/pull/2824) PR)

---

If we find a consensus on this approach I would rather commit the two changes (remove `test_suite`, add `tests_require`) without creating PRs for each repo.

mikaelarguedas, 2017-10-20:
> Since ament_tools doesn't invoke setup.py test the wrong attribute isn't used but if you call that command the result is basically wrong. Anyway I think it should be removed from all Python packages.

I'm fine with that

> Instead I would like to propose to add the following instead:

While related, this seems to be a different discussion. Adding this would allow us to run the tests using `setup.py test`, but via ament all our tests are registered and ran as nosetests. I'm pretty excited about pytest in geenral but didnt have time to do a proper comparison. Though they have several features that are not available or easy to exercise in nose and it seems to be a project getting more and more popular and used in the Python community.

Can the first change be applied without breaking anything and delay the second change to after we make a decision about how we want to test python code in the future + (if we decide to switch) the appropriate change in our cmake infrastructure (and our macros to register tests) to run jobs with pytest and play with the various options and plugins?

dirk-thomas, 2017-10-20:
> Can the first change be applied without breaking anything and delay the second change to after we make a decision

That is certainly possible. Since the presence of `test_suite` is not a problem at the moment I would not touch all these files twice but wait until we are ready for the second part.

dirk-thomas, 2017-11-10:
Once ament/ament_tools#169 is merged I propose to remove the `test_suite` entry from all Python packages as well as add the `pytest` `tests_require`. If I would get an approval on this here I would commit this change to all repos directly without the overhead of creating numerous PRs :wink:

I can e.g. comment with all the commits here for the record after its done.

mikaelarguedas, 2017-11-10:
Sounds good to me

> I can e.g. comment with all the commits here for the record after its done.

:+1:

dirk-thomas, 2017-11-11:
* [ament_index](https://github.com/ament/ament_index/commit/e613250d157343d4919b092a22a1e16a1f20b27f)
* [ament_lint](https://github.com/ament/ament_lint/commit/000a288b878afcd091edea97cfade83efff2c688)
* [ament_package](https://github.com/ament/ament_package/commit/28a250660e343f12e0a0718c1c049fa800ca3e35)
* [demos](https://github.com/ros2/demos/commit/a45d2554a05a1f6836ff37472cd898ed742503d1)
* [examples](https://github.com/ros2/examples/commit/12475d39f01ea5976cb97d77b9731f82c08e4b0a)
* [launch](https://github.com/ros2/launch/commit/0caf2e752a4c5d74581a98a364b3b49979e334cf)
* [ros2cli](https://github.com/ros2/ros2cli/commit/931c9bec98d3bc28f41f594e869619ebf263decc)

closed 2017-11-11 by dirk-thomas

mikaelarguedas, 2017-11-14:
It looks like some packages have been omitted. Is it on purpose ?
- osrf_pycommon
- ament_tools
- sros2
- teleop_twist_keyboard
Opening PRs for each of them in case they're valuable feel free to comment/close on the individual ones

dirk-thomas, 2017-11-14:
> It looks like some packages have been omitted. Is it on purpose ?
> - osrf_pycommon
> - ament_tools

Since this will likely be replaced with the new build tool I didn't spend any time on updating those. Also both repositories use Travis to test with `nose` so I expected that those would need to be updated too to use `pytest`.

> - sros2

I guess this repo was just not in my local workspace so I didn't catch it. :+1:

> - teleop_twist_keyboard

This repo is not part of the ros2.repos file so I didn't consider it.