ros2/ros2 #416: ROS2 beta 3 not compiling on OSX

astralien3000, 2017-10-24:
## Bug report

- Operating System: OSX 10.11.6
- Installation type: from source
- Version or commit hash: release-beta3
- DDS implementation: Fast-RTPS ? (the default one)

#### Steps to reproduce issue

I followed [this tutorial](https://github.com/ros2/ros2/wiki/OSX-Development-Setup)

```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
vcs import src < ros2.repos
```

```
cd ~/ros2_ws/
src/ament/ament_tools/scripts/ament.py build --symlink-install
```

#### Error code

```
/usr/local/bin/idlpp: illegal option -- N
Usage: /usr/local/bin/idlpp [-c preprocessor-path] [-b ORB-template-path]
       [-n <include-suffix>] [-I path] [-D macro[=definition]] [-S | -C] 
       [-l (c | c++ | cpp | isocpp | isoc++ | cs | java)] [-j [old]:<new>] [-d directory] [-i] 
       [-P dll_macro_name[,<h-file>]] [-o (dds-types | custom-psm | no-equality)] <filename>
Traceback (most recent call last):
  File "/Users/loicdauphin/ros2_ndn_ws/install/lib/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp", line 60, in <module>
    sys.exit(main())
  File "/Users/loicdauphin/ros2_ndn_ws/install/lib/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp", line 55, in main
    args.idl_pp,
  File "/Users/loicdauphin/ros2_ndn_ws/install/lib/python3.6/site-packages/rosidl_typesupport_opensplice_cpp/__init__.py", line 74, in generate_dds_opensplice_cpp
    subprocess.check_call(cmd)
  File "/usr/local/Cellar/python3/3.6.2/Frameworks/Python.framework/Versions/3.6/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/idlpp', '-I', '/Users/loicdauphin/ros2_ndn_ws/build/example_interfaces/rosidl_generator_dds_idl', '-I', '/usr/local/etc/opensplice/idlpp', '-S', '-l', 'cpp', '-N', '-d', '/Users/loicdauphin/ros2_ndn_ws/build/example_interfaces/rosidl_typesupport_opensplice_cpp/example_interfaces/srv/dds_opensplice', '/Users/loicdauphin/ros2_ndn_ws/build/example_interfaces/rosidl_generator_dds_idl/example_interfaces/srv/dds_opensplice/Sample_AddTwoInts_Request_.idl']' returned non-zero exit status 255.
make[2]: *** [rosidl_typesupport_opensplice_cpp/example_interfaces/srv/add_two_ints__rosidl_typesupport_opensplice_cpp.hpp] Error 1
make[2]: *** Deleting file `rosidl_typesupport_opensplice_cpp/example_interfaces/srv/add_two_ints__rosidl_typesupport_opensplice_cpp.hpp'
make[1]: *** [CMakeFiles/example_interfaces__rosidl_typesupport_opensplice_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 42%] Linking CXX shared library libexample_interfaces__rosidl_typesupport_introspection_cpp.dylib
[ 42%] Built target example_interfaces__rosidl_typesupport_introspection_cpp
make: *** [all] Error 2

<== Command '. /Users/loicdauphin/ros2_ndn_ws/build/example_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/loicdauphin/ros2_ndn_ws/build/example_interfaces' with exit code '2'
```

#### Additional information

I succeeded to compile with release-beta1 (I haven't tested beta2).

dirk-thomas, 2017-10-24:
I guess you still have the OpenSplice version installed which was used during beta 1. Please try to remove the Debian package of it (`libopensplice64` if I recall correctly) and then try a fresh build.

If you want to try Open Splice beside the default FastRTPS you can also install the latest version (`libopensplice67`).

astralien3000, 2017-10-24:
Thank you !
I removed the opensplice package.
Now it works.

closed 2017-10-24 by astralien3000