ros2/ros2 #418: Error compiling Beta3 on Raspberry Pi 3

Seiwert, 2017-10-30:
## Bug report

**Required Info:**

- Operating System:
  -  Raspbian GNU/Linux 8.0 (jessie) 
- Installation type:
  - Source 
- Version or commit hash:
  -  release-beta3 
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  -  N/A

#### Steps to reproduce issue

```
# On a Raspberry Pi 3 Model B
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
vcs-import src < ros2.repos

src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --skip-packages ros1_bridge --make-flags -j1
```

#### Expected behavior
ROS 2 to compile.

#### Actual behavior
```
Process package 'ament_cmake_core' with context:
--------------------------------------------------------------------------------
 source_space => /root/ros2_ws/src/ament/ament_cmake/ament_cmake_core
  build_space => /root/ros2_ws/build/ament_cmake_core
install_space => /root/ros2_ws/install
   make_flags => -j1
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'ament_cmake_core'
Running cmake because arguments have changed.
==> '. /root/ros2_ws/build/ament_cmake_core/cmake__build.sh && /usr/bin/cmake /root/ros2_ws/src/ament/ament_cmake/ament_cmake_core -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/root/ros2_ws/install' in '/root/ros2_ws/build/ament_cmake_core'
Segmentation fault

<== Command '. /root/ros2_ws/build/ament_cmake_core/cmake__build.sh && /usr/bin/cmake /root/ros2_ws/src/ament/ament_cmake/ament_cmake_core -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/root/ros2_ws/install' failed in '/root/ros2_ws/build/ament_cmake_core' with exit code '139'
<== Command '. /root/ros2_ws/build/ament_cmake_core/cmake__build.sh && /usr/bin/cmake /root/ros2_ws/src/ament/ament_cmake/ament_cmake_core -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/root/ros2_ws/install' failed in '/root/ros2_ws/build/ament_cmake_core' with exit code '139'
```
#### Additional information

I followed this tutorial: https://hackmd.io/s/BysaB355- for additional guidance. Given that, I currently have 60G of SWAP space set up (if this info is helpful).

Seiwert, 2017-10-30:
I think this may be an issue with my cmake installation. 
Running
```
cmake --version
```
returned a segmentation fault so something is definitely wrong there. I'm reinstalling cmake and investigating that.

allenh1, 2017-10-30:
> Running cmake --version returned a segmentation fault so something is definitely wrong there. I'm reinstalling cmake and investigating that.

Ya, that might do it ;) Did you install `cmake` from source? That's rather odd behavior for a debian package.

Seiwert, 2017-10-30:
Not originally. I used **sudo apt-get install cmake** to install it the first time. I built it from source this time and everything seems to be running smoothly now. ROS2 is still building but once it's done I'll close the issue. Don't wanna close it too soon and jinx myself :P

mikaelarguedas, 2017-10-30:
One thing to keep in mind if you build natively in the future is that Jessie, (and I assume raspbian Jessie) ships with a version of cmake [older(3.0.2)](https://packages.debian.org/jessie/cmake) than the ros2 minimal required by ros2 (3.5). Not saying it should segfault ^^ just that the compilation would have likely failed later on because of the cmake version.

Seiwert, 2017-10-31:
Thanks for the help everyone. I did get it to compile (after dealing with the OpenSSL issue described [here](https://github.com/ros2/ros2/issues/378)) but I'm still having some problems.

Whenever I run anything ros2 (e.g. just plain **ros2** or **ros2 run demo_nodes_cpp talker**) I get this error:
```
Failed to load entry point 'start': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'status': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'stop': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'list': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'call': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'list': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'echo': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'list': stat: path should be string, bytes, os.PathLike or integer, not NoneType
Failed to load entry point 'pub': stat: path should be string, bytes, os.PathLike or integer, not NoneType
```

Has anyone encountered this before?

dhood, 2017-10-31:
The stat error might be coming from rclpy (some `ros2` tools use rospy). We have seen this happen lately [(see here)](https://github.com/ros2/rclpy/pull/134#issuecomment-339821972) because the c extension used for rclpy couldn't be imported.

Please try rebuilding rclpy and if that doesn't fix it:

1. Can you try to run `./install/demo_nodes_py/lib/demo_nodes_py/talker` and confirm that that gives the same error?
1. Running a c++ example directly (`./install_isolated/demo_nodes_cpp/lib/demo_nodes_cpp/talker`) probably doesn't give you any issues, right?
1. Could you output the contents of 
`install/rclpy/lib/python3.6/site-packages/rclpy/` so we can see if the C extension is there at least

Seiwert, 2017-11-01:
Hi @dhood thanks for the response.

Based on your instructions, I don't think that my install directory is structured the way that it should be. Here is my install directory:
```
total 44
drwxr-xr-x   2 erle erle 4096 Oct 31 16:59 bin
drwxr-xr-x   3 erle erle 4096 Oct 30 20:58 examples
drwxr-xr-x  30 erle erle 4096 Oct 30 23:29 include
drwxr-xr-x  16 erle erle 4096 Oct 31 16:57 lib
lrwxrwxrwx   1 erle erle   95 Oct 30 14:41 local_setup.bash -> /home/erle/ros2_ws/src/ament/ament_package/ament_package/template/prefix_level/local_setup.bash
lrwxrwxrwx   1 erle erle   48 Oct 31 18:06 local_setup.sh -> /home/erle/ros2_ws/build/nav_msgs/local_setup.sh
lrwxrwxrwx   1 erle erle   94 Oct 30 14:41 local_setup.zsh -> /home/erle/ros2_ws/src/ament/ament_package/ament_package/template/prefix_level/local_setup.zsh
lrwxrwxrwx   1 erle erle   97 Oct 30 14:41 _order_packages.py -> /home/erle/ros2_ws/src/ament/ament_package/ament_package/template/prefix_level/_order_packages.py
lrwxrwxrwx   1 erle erle   89 Oct 30 14:41 setup.bash -> /home/erle/ros2_ws/src/ament/ament_package/ament_package/template/prefix_level/setup.bash
lrwxrwxrwx   1 erle erle   42 Oct 31 18:06 setup.sh -> /home/erle/ros2_ws/build/nav_msgs/setup.sh
lrwxrwxrwx   1 erle erle   88 Oct 30 14:41 setup.zsh -> /home/erle/ros2_ws/src/ament/ament_package/ament_package/template/prefix_level/setup.zsh
drwxr-xr-x 101 erle erle 4096 Oct 30 23:29 share
drwxr-xr-x   4 erle erle 4096 Oct 30 20:58 src
```
As you can see, I'm missing the ``` demo_nodes_py ``` and ``` rclpy ``` directories. Given that, I attempted to rebuild rclpy with this commande:
``` 
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --only rclpy --force-cmake-configure --make-flags -j1
```
And got this error:
```
Process package 'rclpy' with context:
--------------------------------------------------------------------------------
 source_space => /home/erle/ros2_ws/src/ros2/rclpy/rclpy
  build_space => /home/erle/ros2_ws/build/rclpy
install_space => /home/erle/ros2_ws/install
   make_flags => -j1
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'rclpy'
Running cmake because arguments have changed.
==> '. /home/erle/ros2_ws/build/rclpy/cmake__build.sh && /usr/local/bin/cmake /home/erle/ros2_ws/src/ros2/rclpy/rclpy -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/erle/ros2_ws/install' in '/home/erle/ros2_ws/build/rclpy'
-- The C compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ament_cmake: 0.0.3 (/home/erle/ros2_ws/install/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "Findrcl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "rcl", but
  CMake did not find one.

  Could not find a package configuration file provided by "rcl" with any of
  the following names:

    rclConfig.cmake
    rcl-config.cmake

  Add the installation prefix of "rcl" to CMAKE_PREFIX_PATH or set "rcl_DIR"
  to a directory containing one of the above files.  If "rcl" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/erle/ros2_ws/build/rclpy/CMakeFiles/CMakeOutput.log".

<== Command '. /home/erle/ros2_ws/build/rclpy/cmake__build.sh && /usr/local/bin/cmake /home/erle/ros2_ws/src/ros2/rclpy/rclpy -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/erle/ros2_ws/install' failed in '/home/erle/ros2_ws/build/rclpy' with exit code '1'
<== Command '. /home/erle/ros2_ws/build/rclpy/cmake__build.sh && /usr/local/bin/cmake /home/erle/ros2_ws/src/ros2/rclpy/rclpy -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/erle/ros2_ws/install' failed in '/home/erle/ros2_ws/build/rclpy' with exit code '1'
```
So that tells me that rcl didn't install so then I ran this:
```
[... 65 lines cut ...]

-- Generating done
-- Build files have been written to: /home/erle/ros2_ws/build/rcl
==> '. /home/erle/ros2_ws/build/rcl/cmake__build.sh && /usr/bin/make -j1' in '/home/erle/ros2_ws/build/rcl'
[ 27%] Built target rcl
[ 31%] Built target gtest
[ 34%] Built target rcl_memory_tools
[ 37%] Built target gtest_main
[ 39%] Linking CXX executable test_publisher__rmw_fastrtps_cpp
../librcl.so: undefined reference to `__atomic_exchange_8'
../librcl.so: undefined reference to `__atomic_load_8'
../librcl.so: undefined reference to `__atomic_store_8'
/home/erle/ros2_ws/install/lib/libfastrtps.so.1.5.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
test/CMakeFiles/test_publisher__rmw_fastrtps_cpp.dir/build.make:134: recipe for target 'test/test_publisher__rmw_fastrtps_cpp' failed
make[2]: *** [test/test_publisher__rmw_fastrtps_cpp] Error 1
CMakeFiles/Makefile2:1087: recipe for target 'test/CMakeFiles/test_publisher__rmw_fastrtps_cpp.dir/all' failed
[... cut ...]
<== Command '. /home/erle/ros2_ws/build/rcl/cmake__build.sh && /usr/bin/make -j1' failed in '/home/erle/ros2_ws/build/rcl' with exit code '2'
<== Command '. /home/erle/ros2_ws/build/rcl/cmake__build.sh && /usr/bin/make -j1' failed in '/home/erle/ros2_ws/build/rcl' with exit code '2'
```
So now I'm starting to think that my build didn't correctly build at all. Should I just attempt to rebuild from scratch now that I've got my dependencies sorted out (e.g. cmake and OpenSSL) or would you suggest something else?

dhood, 2017-11-01:
> As you can see, I'm missing the demo_nodes_py and rclpy directories.

Ah, that was my fault, I was asking you questions as if you had built using `--isolated`, in which case packages install to a named subdirectory. Yours would be in e.g. `install/lib/demo_nodes_py/talker`.

Still, it does indeed look like your build has had some issues; you've got the right idea trying to rebuild from scratch in a clean workspace, let us know how that goes.

Seiwert, 2017-11-01:
Doesn't look like it's in my ``` install/lib/ ``` directory either.

```
total 17148
drwxr-xr-x 3 erle erle     4096 Oct 30 21:15 cmake
drwxr-xr-x 3 erle erle     4096 Oct 30 20:28 console_bridge
drwxr-xr-x 3 erle erle     4096 Oct 30 20:29 fastcdr
drwxr-xr-x 3 erle erle     4096 Oct 30 20:58 fastrtps
lrwxrwxrwx 1 erle erle       80 Oct 30 22:48 libactionlib_msgs__rosidl_generator_c.so -> /home/erle/ros2_ws/build/actionlib_msgs/libactionlib_msgs__rosidl_generator_c.so
lrwxrwxrwx 1 erle erle       84 Oct 30 22:48 libactionlib_msgs__rosidl_typesupport_cpp.so -> /home/erle/ros2_ws/build/actionlib_msgs/libactionlib_msgs__rosidl_typesupport_cpp.so
lrwxrwxrwx 1 erle erle       82 Oct 30 22:48 libactionlib_msgs__rosidl_typesupport_c.so -> /home/erle/ros2_ws/build/actionlib_msgs/libactionlib_msgs__rosidl_typesupport_c.so
lrwxrwxrwx 1 erle erle       98 Oct 30 22:48 libactionlib_msgs__rosidl_typesupport_introspection_cpp.so -> /home/erle/ros2_ws/build/actionlib_msgs/libactionlib_msgs__rosidl_typesupport_introspection_cpp.so
lrwxrwxrwx 1 erle erle       96 Oct 30 22:48 libactionlib_msgs__rosidl_typesupport_introspection_c.so -> /home/erle/ros2_ws/build/actionlib_msgs/libactionlib_msgs__rosidl_typesupport_introspection_c.so
-rw-r--r-- 1 erle erle    90888 Oct 30 21:18 libament_index_cpp.so
lrwxrwxrwx 1 erle erle       88 Oct 30 21:44 libbuiltin_interfaces__rosidl_generator_c.so -> /home/erle/ros2_ws/build/builtin_interfaces/libbuiltin_interfaces__rosidl_generator_c.so
lrwxrwxrwx 1 erle erle       92 Oct 30 21:44 libbuiltin_interfaces__rosidl_typesupport_cpp.so -> /home/erle/ros2_ws/build/builtin_interfaces/libbuiltin_interfaces__rosidl_typesupport_cpp.so
lrwxrwxrwx 1 erle erle       90 Oct 30 21:44 libbuiltin_interfaces__rosidl_typesupport_c.so -> /home/erle/ros2_ws/build/builtin_interfaces/libbuiltin_interfaces__rosidl_typesupport_c.so
lrwxrwxrwx 1 erle erle      106 Oct 30 21:44 libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.so -> /home/erle/ros2_ws/build/builtin_interfaces/libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.so
lrwxrwxrwx 1 erle erle      104 Oct 30 21:44 libbuiltin_interfaces__rosidl_typesupport_introspection_c.so -> /home/erle/ros2_ws/build/builtin_interfaces/libbuiltin_interfaces__rosidl_typesupport_introspection_c.so
lrwxrwxrwx 1 erle erle       56 Oct 30 21:15 libclass_loader.so -> /home/erle/ros2_ws/build/class_loader/libclass_loader.so
lrwxrwxrwx 1 erle erle       24 Oct 30 20:28 libconsole_bridge.so -> libconsole_bridge.so.0.3
-rw-r--r-- 1 erle erle    20116 Oct 30 20:27 libconsole_bridge.so.0.3
lrwxrwxrwx 1 erle erle       82 Oct 30 23:19 libdiagnostic_msgs__rosidl_generator_c.so -> /home/erle/ros2_ws/build/diagnostic_msgs/libdiagnostic_msgs__rosidl_generator_c.so
lrwxrwxrwx 1 erle erle       86 Oct 30 23:19 libdiagnostic_msgs__rosidl_typesupport_cpp.so -> /home/erle/ros2_ws/build/diagnostic_msgs/libdiagnostic_msgs__rosidl_typesupport_cpp.so
lrwxrwxrwx 1 erle erle       84 Oct 30 23:19 libdiagnostic_msgs__rosidl_typesupport_c.so -> /home/erle/ros2_ws/build/diagnostic_msgs/libdiagnostic_msgs__rosidl_typesupport_c.so
lrwxrwxrwx 1 erle erle      100 Oct 30 23:19 libdiagnostic_msgs__rosidl_typesupport_introspection_cpp.so -> /home/erle/ros2_ws/build/diagnostic_msgs/libdiagnostic_msgs__rosidl_typesupport_introspection_cpp.so
lrwxrwxrwx 1 erle erle       98 Oct 30 23:19 libdiagnostic_msgs__rosidl_typesupport_introspection_c.so -> /home/erle/ros2_ws/build/diagnostic_msgs/libdiagnostic_msgs__rosidl_typesupport_introspection_c.so
lrwxrwxrwx 1 erle erle       88 Oct 30 21:47 libexample_interfaces__rosidl_generator_c.so -> /home/erle/ros2_ws/build/example_interfaces/libexample_interfaces__rosidl_generator_c.so
lrwxrwxrwx 1 erle erle       92 Oct 30 21:47 libexample_interfaces__rosidl_typesupport_cpp.so -> /home/erle/ros2_ws/build/example_interfaces/libexample_interfaces__rosidl_typesupport_cpp.so
lrwxrwxrwx 1 erle erle       90 Oct 30 21:47 libexample_interfaces__rosidl_typesupport_c.so -> /home/erle/ros2_ws/build/example_interfaces/libexample_interfaces__rosidl_typesupport_c.so
lrwxrwxrwx 1 erle erle      106 Oct 30 21:47 libexample_interfaces__rosidl_typesupport_introspection_cpp.so -> /home/erle/ros2_ws/build/example_interfaces/libexample_interfaces__rosidl_typesupport_introspection_cpp.so
lrwxrwxrwx 1 erle erle      104 Oct 30 21:47 libexample_interfaces__rosidl_typesupport_introspection_c.so -> /home/erle/ros2_ws/build/example_interfaces/libexample_interfaces__rosidl_typesupport_introspection_c.so
lrwxrwxrwx 1 erle erle       15 Oct 30 20:29 libfastcdr.so -> libfastcdr.so.1
lrwxrwxrwx 1 erle erle       19 Oct 30 20:29 libfastcdr.so.1 -> libfastcdr.so.1.0.7
-rw-r--r-- 1 erle erle    74152 Oct 30 20:29 libfastcdr.so.1.0.7
lrwxrwxrwx 1 erle erle       16 Oct 30 20:58 libfastrtps.so -> libfastrtps.so.1
lrwxrwxrwx 1 erle erle       20 Oct 30 20:58 libfastrtps.so.1 -> libfastrtps.so.1.5.0
-rw-r--r-- 1 erle erle  2510960 Nov  1 16:25 libfastrtps.so.1.5.0
lrwxrwxrwx 1 erle erle       78 Oct 30 23:12 libgeometry_msgs__rosidl_generator_c.so -> /home/erle/ros2_ws/build/geometry_msgs/libgeometry_msgs__rosidl_generator_c.so
lrwxrwxrwx 1 erle erle       82 Oct 30 23:12 libgeometry_msgs__rosidl_typesupport_cpp.so -> /home/erle/ros2_ws/build/geometry_msgs/libgeometry_msgs__rosidl_typesupport_cpp.so
lrwxrwxrwx 1 erle erle       80 Oct 30 23:12 libgeometry_msgs__rosidl_typesupport_c.so -> /home/erle/ros2_ws/build/geometry_msgs/libgeometry_msgs__rosidl_typesupport_c.so
[... cut ...]
drwxr-xr-x 2 erle erle     4096 Oct 30 21:25 rosidl_generator_dds_idl
drwxr-xr-x 2 erle erle     4096 Oct 30 21:42 rosidl_generator_py
drwxr-xr-x 2 erle erle     4096 Oct 30 21:31 rosidl_typesupport_c
drwxr-xr-x 2 erle erle     4096 Oct 30 21:32 rosidl_typesupport_cpp
drwxr-xr-x 2 erle erle     4096 Oct 30 21:32 rosidl_typesupport_introspection_c
drwxr-xr-x 2 erle erle     4096 Oct 30 21:33 rosidl_typesupport_introspection_cpp
```

I don't know what happened but nothing good apparently. I'll rebuild it and let you all know. Thanks.

Seiwert, 2017-11-02:
So I made a new work-space and built from scratch but got this error again:

```
[ 31%] Linking CXX static library libgtest.a
[ 31%] Built target gtest
Scanning dependencies of target rcl_memory_tools
[ 32%] Building CXX object test/memory_tools/CMakeFiles/rcl_memory_tools.dir/memory_tools.cpp.o
[ 34%] Linking CXX shared library librcl_memory_tools.so
[ 34%] Built target rcl_memory_tools
Scanning dependencies of target gtest_main
[ 36%] Building CXX object gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[ 37%] Linking CXX static library libgtest_main.a
[ 37%] Built target gtest_main
Scanning dependencies of target test_publisher__rmw_fastrtps_cpp
[ 39%] Building CXX object test/CMakeFiles/test_publisher__rmw_fastrtps_cpp.dir/rcl/test_publisher.cpp.o
[ 41%] Linking CXX executable test_publisher__rmw_fastrtps_cpp
../librcl.so: undefined reference to `__atomic_exchange_8'
../librcl.so: undefined reference to `__atomic_load_8'
../librcl.so: undefined reference to `__atomic_store_8'
/home/erle/ros2_ws/install/lib/libfastrtps.so.1.5.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
test/CMakeFiles/test_publisher__rmw_fastrtps_cpp.dir/build.make:134: recipe for target 'test/test_publisher__rmw_fastrtps_cpp' failed
make[2]: *** [test/test_publisher__rmw_fastrtps_cpp] Error 1
CMakeFiles/Makefile2:1087: recipe for target 'test/CMakeFiles/test_publisher__rmw_fastrtps_cpp.dir/all' failed
make[1]: *** [test/CMakeFiles/test_publisher__rmw_fastrtps_cpp.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/erle/ros2_ws/build/rcl/cmake__build.sh && /usr/bin/make -j1' failed in '/home/erle/ros2_ws/build/rcl' with exit code '2'
<== Command '. /home/erle/ros2_ws/build/rcl/cmake__build.sh && /usr/bin/make -j1' failed in '/home/erle/ros2_ws/build/rcl' with exit code '2'
```
It looks like an issue with referencing the **libatomic** library. However, I thought that shipped with gcc versions 4.7+ ?
Here is my current version:
```
gcc --version
gcc (Raspbian 7.2.0-11) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc
```

mikaelarguedas, 2017-11-02:
Yeah that's surprising indeed after 4.7 every gcc should ship libatomic.

TBH you're blasting new trails here because Jessie is pretty old and we don't test for it or support it. While not being officially supported, Debian Stretch is much more recent and has system dependencies versions closer to what we usually test with. So (if it's possible for you) I would recommend to try Raspbian 9 (based on Stretch) that should have recent enough versions of Python, cmake, gcc, openssl ...

allenh1, 2017-11-02:
>  So (if it's possible for you) I would recommend to try Raspbian 9 (based on Stretch) that should have recent enough versions of Python, cmake, gcc, openssl

+1

Seiwert, 2017-11-03:
@mikaelarguedas @allenh1 @dhood  Thanks for the help!

At this time I'm not able to upgrade to Raspbian 9 but I should be able to in the near future. For the time being, I'll close this issue. If I run into any additional issues after installing Raspbian 9, I'll re-open this thread.

Thanks again everyone.

closed 2017-11-03 by Seiwert

mikaelarguedas, 2017-11-03:
Sounds good, keep us updated!

Seiwert, 2017-11-09:
Hey everyone, I got ROS2 on the raspberry pi! 

@mikaelarguedas , you were correct. Installing Raspbian 9 fixed everything. ROS 2 built without any errors at all. Thanks again everyone for all of your help!

LanderU, 2017-12-13:
Hi!

I was able to compile ROS2 in `jessie`. I'll left here the steps.

### System description

```
erle@erle-brain:~ $ uname -a
Linux erle-brain 4.4.50-v7+ #1 SMP PREEMPT Tue Jun 27 12:03:45 CEST 2017 armv7l GNU/Linux
```
```
erle@erle-brain:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```
```
erle@erle-brain:~ $ gcc --version
gcc (Raspbian 6.3.0-18+rpi1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
erle@erle-brain:~ $ g++ --version
g++ (Raspbian 6.3.0-18+rpi1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```

erle@erle-brain:~ $ cmake --version
cmake version 3.9.6

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

### Changes on the FS

My `sources.list` looks like this:

```
deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi
#Erle Robotics sources 
deb http://packages.erlerobotics.com/ erle main
```

I set **stretch** instead of **jessie**.

I have incremented the swap space to 1GB, source: https://raspberrypi.stackexchange.com/questions/70/how-to-set-up-swap-space

```
erle@erle-brain:~ $ free -h
             total       used       free     shared    buffers     cached
Mem:          862M       234M       627M        32M        25M       147M
-/+ buffers/cache:        61M       800M
Swap:         1.0G         0B       1.0G
```

Also, I remove the **cmake** old version and I compile it from sources (the old version always returned segmentation fault): https://cmake.org/install/

I install all the dependencies/setup the workspace as described in the wiki.

### Compile it

I used the following command to compile it:

```
src/ament/ament_tools/scripts/ament.py build --install-space /opt/ros2 -j 2
```

### Run it

Fast-RTPS has some problems when you try to run ROS2 with two interfaces with ip address: https://github.com/eProsima/Fast-RTPS/issues/77

But, if I down one of the interfaces everything is working well, [video](https://www.youtube.com/watch?v=lKZpPTQeYbI).

machinekoder, 2019-08-20:
I encountered the same problem on the latest Raspbian with Buster. Workaround: just comment out the test executable which triggers the error message.