ros2/ros2 #421: Windows run ros2 demo FAILED The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system

oPiZiL, 2017-11-02:
## Bug report

**Required Info:**

- Operating System:
  -  Windows 10
- Installation type:
  -  binaries
- Version or commit hash:
  -  ros2-beta3-package-windows-fastrtps-AMD64.zip
- DDS implementation:
  -  
- Client library (if applicable):
  - 

#### Steps to reproduce issue

1. All the required prerequisites

````
C:\WINDOWS\system32>choco install -y -s C:\Users\louwei\Desktop\ros2 asio eigen tinyxml-usestl tinyxml2
Chocolatey v0.10.8
Installing the following packages:
asio;eigen;tinyxml-usestl;tinyxml2
By installing you accept licenses for the packages.
asio v1.10.6 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.
eigen v3.3.3 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.
tinyxml-usestl v2.6.2 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.
tinyxml2 v4.0.1 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.

Chocolatey installed 0/4 packages.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Warnings:
 - tinyxml2 - tinyxml2 v4.0.1 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.
 - asio - asio v1.10.6 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.
 - tinyxml-usestl - tinyxml-usestl v2.6.2 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.
 - eigen - eigen v3.3.3 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.

C:\WINDOWS\system32>python -m pip install -U pyyaml setuptools
Requirement already up-to-date: pyyaml in c:\python36\lib\site-packages
Requirement already up-to-date: setuptools in c:\python36\lib\site-packages
````

2. I followed the install tips

> call C:\dev\ros2\local_setup.bat
> ros2 run demo_nodes_cpp talker

BUT errors coccurred!
```
C:\WINDOWS\system32>ros2 run demo_nodes_cpp talker
Failed to load entry point 'start': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'status': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'stop': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'list': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'call': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'list': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'echo': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'list': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'pub': DLL load failed: 操作系统无法运行 %1。
The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
```

**HERE is the PATH set** 
[image]

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

mikaelarguedas, 2017-11-16:
sorry @oPiZiL for the late reply.

From experience this happens when some DLL are missing on the system. Can you try to run the talker executable directly? Windows may pop-up a windows telling you which DLL is missing.

```
> call D:\dev\ros2\setup.bat
> D:\dev\ros2\Lib\demo_nodes_cpp\talker.exe
```

oPiZiL, 2017-11-22:
[image]
[image]
[image]
[image]

mikaelarguedas, 2017-11-22:
Thanks for the screenshots.
Are you sure you downloaded `ros2-beta3-package-windows-fastrtps-AMD64.zip` ? If yes that means something went wrong in our package generation as these dlls should not be used by Fast-RTPS.

More context:
We provide 2 builds of ROS 2 beta3, one built for FastRTPS (our default DDS implementation) and one built for Opensplice.
We provide the built dll for FastRTPS but not for Opensplice. The missing dlls in your screenshot are dll from Opensplice. You should be able to download it from [their github repository](https://github.com/ADLINK-IST/opensplice/releases).

We recommend using the version built with Fast-RTPS as this is our default target implementation (https://github.com/ros2/ros2/releases/download/release-beta3/ros2-beta3-package-windows-fastrtps-AMD64.zip).

Hope this helps

mikaelarguedas, 2018-02-12:
@oPiZiL I'm going to close this due to no response. Please feel free to comment here if you still face the issue with the [latest release](https://github.com/ros2/ros2/releases/tag/release-ardent-20180209) and we can reopen it.

closed 2018-02-12 by mikaelarguedas

mcevoyandy, 2018-07-08:
I'm currently facing what seems to be a similar issue trying to install using `ros2-package-windows-AMD64.zip`

```
Failed to load entry point 'launch': DLL load failed: %1 is not a valid Win32 application.
The C extension 'c:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'echo': DLL load failed: %1 is not a valid Win32 application.
The C extension 'c:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions

>>> [rcutils|error_handling.c:155] rcutils_set_error_state()
This error state is being overwritten:

  'failed to load shared library of rmw implementation. Exception: Cannot load library: c:\dev\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:135'

with this new error message:

  'failed to load shared library of rmw implementation. Exception: Cannot load library: c:\dev\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:135'
```

Along with this, what version of Python should be installed? Choco will install Python 3.7 by default, but it seems ros2 expects 3.6? Then it's not clear if the 32 or 64 bit version is needed...

tfoote, 2018-07-09:
Please read our support policy ettiquette section: http://wiki.ros.org/Support

In particular please don't multiply post your questions [1](https://github.com/ros2/ros2/issues/421#issuecomment-403289505) [2](https://github.com/ros2/ros2/issues/525#issuecomment-403319623) [3](https://discourse.ros.org/t/r2b3-windows-failed-to-create-process-installation-test/2896/2)

Similarly please don't post onto old closed threads and hijack them.