ros2/ros2 #422: Tracking bug for reverting changes made to ported packages for field initialization

clalancette, 2017-11-02:
During the development of ROS2 beta2 and beta3, we fixed a bunch of flaky tests by manually initializing fields in the test code.  Now that the zero initialization work has gone in (https://github.com/ros2/rosidl/pull/236), we can revert those changes.  In some cases, these repositories are forks from upstream ROS1 versions, so these reverts will get the forks closer to upstream.  While we are at it, we should probably consider merging fixes from the ROS2 forked versions to their upstreams, and merging fixes from their upstreams to the ROS2 forked versions.

* [ ] rcl_interfaces
  * [ ] https://github.com/ros2/rcl_interfaces/pull/18 (already reverted in ros2/rcl_interfaces#24)
* [ ] geometry2
  * [ ] ros2/geometry2#19
  * [ ] ros2/geometry2#18
  * [ ] https://github.com/ros2/geometry2/commit/31791b01abceb2ed01a459962ef75774d88e0113
  * [ ] ros2/geometry2#28
  * [ ] https://github.com/ros2/geometry2/commit/d957ba6eda228ae30ef8d5d9a25b79306dd3e4d2
* [ ] vision_opencv
  * [ ] All of the code in the constructor: https://github.com/ros2/vision_opencv/blob/ros2/image_geometry/src/pinhole_camera_model.cpp#L33
  * [ ] The code setting fields to zero in `cam_info_`, starting here: https://github.com/ros2/vision_opencv/blob/ros2/image_geometry/test/utest.cpp#L30
* [ ] depthimage_to_laserscan
  * [ ] The code that sets fields of `depth_msg_` and `info_msg_` to 0, starting here: https://github.com/ros2/depthimage_to_laserscan/blob/ros2/test/DepthImageToLaserScanTest.cpp#L70
* [ ] ros_astra_camera
  * [ ] https://github.com/ros2/ros_astra_camera/commit/3f16915725cef179e672331533960aaa6c1cd8bb
* [ ] teleop_twist_joy
  * [ ] The code initializing the cmd_vel_msg to zero, starting here: https://github.com/ros2/teleop_twist_joy/blob/ros2/src/teleop_twist_joy.cpp#L160
* [ ] turtlebot2_demo
  * [ ] Various pieces of message initialization in the kobuki_node, starting around here: https://github.com/ros2/turtlebot2_demo/blob/master/turtlebot2_drivers/src/kobuki_node.cpp#L153
  * [ ] Initialization of cmd_vel fields to 0 in the follower, starting here: https://github.com/ros2/turtlebot2_demo/blob/master/turtlebot2_follower/src/follower.cpp#L221

mikaelarguedas, 2018-03-08:
@clalancette is it something that's still on your radar?

clalancette, 2018-03-08:
While this would be a good thing to do to get us back closer to the respective upstreams, it is not a priority for me right now.

dirk-thomas, 2018-03-08:
Please prioritize this item in order to close it out and reduce the difference of the ROS 2 forks/branches.

mikaelarguedas, 2018-09-10:
@clalancette FYI I reverted the changes on the first few repos and updated the issue description accordingly.

Can you please use permalinks in the future? (several links were dead making it harder to track down the changes)
Thanks!

clalancette, 2020-08-13:
The first half of this was done long ago.  The remaining items either are old (and need an update to bring up to new ROS 2 anyway), deprecated, or already done.  Because of all of that, I'm going to go ahead and close this.

closed 2020-08-13 by clalancette