ros2/ros2 #424: reuse code between rmw_implementation and rosidl_typesupport

BrannonKing, 2017-11-08:
Both rmw_implementation and rosidl_typesupport implement the same code for dynamic library loading. (In fact, rosidl_typesupport implements that same code twice, once for C and once for C++.) This includes the Poco calls, reading environment variables, and splitting strings. This seems to violate the first principle of programming: keep it DRY.

dirk-thomas, 2017-11-08:
Please consider to provide pull request to improve this.

BrannonKing, 2017-11-08:
I need someone to explain the appropriate project hierarchy for this. What is the (desired and/or current) dependency tree for the rosidl and rmw projects?

dirk-thomas, 2017-11-08:
Each rmw implementation depends on some rosidl type support packages. And one rosidl type support package can depend on another type support package. Sometime the C++ type support package depends on the C type support package and uses some C API / structs. Sometime the C type support package uses implementation from the C++ type support (in order to not reimplement the logic).

kunaltyagi, 2017-11-30:
I've started a rather kludgy integration [here](https://github.com/kunaltyagi/rosidl_typesupport_common). I've gotten it to compile individually for C and C++ (though there is still some work to be done).

Is this the correct method, or did you have something else in mind?

:crossed_fingers: Ignore the mess that is C preprocessor hacks :crossed_fingers:

BrannonKing, 2017-11-30:
@kunaltyagi : explain a little more what you did in this project? You pulled some code from where exactly and got it compiling as stand-alone?

kunaltyagi, 2017-12-01:
I copied code from rosidl_typesuport_c. Most of the modifications are related to namespace and variable prefix difference between C and the respective C++ files from the rosidl_typesuport_cpp

The compilation involves going over the source files 2 times, once to generate C and a second time to generate C++ library. I've managed to get it compile in both modes in a loop (see [this](https://github.com/kunaltyagi/rosidl_typesupport_common/blob/master/CMakeLists.txt#L37)). There was a ament issue earlier, due to which I could not loop, but that's resolved (that was what I referred to when I said *compile individually*).

dirk-thomas, 2017-12-03:
> I've started a rather kludgy integration [here](https://github.com/kunaltyagi/rosidl_typesupport_common).

@kunaltyagi The referenced repository is not a fork and therefore it is not straight forward to see what changes you have actually done. Can you please change this in a way that we can see a diff of the proposed changes.

kunaltyagi, 2017-12-04:
Sorry about that. I've created a fork, updated the code there as well as added 2 root commits in the non-fork repo to compare against

dirk-thomas, 2017-12-04:
> I've created a fork, updated the code there as well as added 2 root commits in the non-fork repo to compare against

Can you please post a link to the diff.

kunaltyagi, 2017-12-04:
A manual diff is [here](https://pastebin.com/RHQpCsEX), and the fork is [here](https://github.com/kunaltyagi/rosidl_typesupport/tree/integration). 

The changes are only a few. I created a new folder, so that messed things up.

dirk-thomas, 2017-12-04:
I am sorry but in the current shape the changes are almost impossible to review / comment on. Please use the following process to make this easier:

* fork the repo
* create a custom branch in the forked repo
* (if necessary) move/copy files to new/different location (in a separate commit)
* use additional commits to modify the code
* create a PR

kunaltyagi, 2017-12-04:
Done. Thanks for the help

dirk-thomas, 2019-09-05:
Closing due to inactivity. If anyone is interested in implementing this we would be happy to review pull requests for it.

closed 2019-09-05 by dirk-thomas