ros2/ros2 #425: Logging meta-ticket

dhood, 2017-11-14:
Steps towards the use of logging API and minimal logger severity configuration [limitations in square brackets].

Minimal functionality
- [x] Logger objects in rclpy https://github.com/ros2/rclpy/issues/103
- [ ] rclpy nodes have loggers associated with them [naming conventions hard-coded]
- [ ] C++ macros wrapping rcutils [specifying logger name instead of logger object]
- [ ] Update demos to use logging [callbacks need access to node logger, C++ passing names not logger object]
- [ ] ROS 2 core using debug logging [no subloggers of nodes' loggers]
---
Demo/documentation
- [ ] Demo of service that sets logger level at runtime for logger name
- [ ] Wiki page on logger configuration
---
C++ logger object
- [ ] Node logger name added to nodes in rcl, wrapped in python
- [ ] C++ macros taking logger object, factory functions
- [ ] ROS 2 core using subloggers from node's logger
---
Enhancements
- [ ] Configuration of non-node loggers' context
- [ ] Improve ease-of-use of accessing node logger from callbacks
- [ ] Wrapping of "throttled" filter in C++ macros

DLu, 2018-06-27:
The migration guide really should be updated to include some basic usage. It still says to use printf, and it took me an hour to sort through this to figure out the new syntax, and that was once I found this wiki page. 

[image]

wjwwood, 2018-06-27:
> The migration guide really should be updated to include some basic usage.

Sorry the migration guide is out-of-date, but the good news is that it's a wiki so you can edit it:

https://github.com/ros2/ros2/wiki/Migration-Guide/_edit

> and it took me an hour to sort through this to figure out the new syntax, and that was once I found this wiki page.

I understand that the migration guide was misleading, but if you just do a search (I did incognito) for "ros2 logging", you get something like this:

[image]

The first hit is this wiki page that @dhood made that explains how to configure logging and links to examples in both C++ and Python:

https://github.com/ros2/ros2/wiki/Logging

The second hit is a wiki page describing the logging demo she made:

https://github.com/ros2/ros2/wiki/Logging-and-logger-configuration

wjwwood, 2018-06-27:
Also, all of our examples and demos use the new logging syntax if you want to look at them for additional examples of using logging, e.g.:

https://github.com/ros2/examples/blob/f6027edfb3a28f012438d1f2d587d0a980e9341c/rclcpp/minimal_subscriber/not_composable.cpp#L27

DLu, 2018-06-27:
Thanks for those links, sincerely. Unfortunately, they did not show up in my Google searches for `ros2 ros_info` or `ros2 rclcpp_info`. 

`david.google_fu--`

wjwwood, 2018-06-27:
No worries, but we're busy in the middle of a release, so we would also (sincerely) appreciate help updating any documentation.