ros2/ros2 #427: [cross-compilation] Unable to build with test due to pthreads

mikaelarguedas, 2017-11-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Xenial
- Installation type:
  - Source install, cross compilation
- Version or commit hash:
  - master
#### Steps to reproduce issue

Follow the arm cross compilation [readme](https://github.com/ros2/ros2/blob/master/arm_crosscompilation/README.md).
When you reach the [build invocation](https://github.com/ros2/ros2/blob/master/arm_crosscompilation/README.md#trigger-a-build), pass to ament the flag `--build-tests`
```
src/ament/ament_tools/scripts/ament.py build --force-cmake-configure --cmake-args -DCMAKE_TOOLCHAIN_FILE=`pwd`/aarch64_toolchainfile.cmake -DTHIRDPARTY=ON -- --build-tests
```

#### Expected behavior
It builds all the packages.

#### Actual behavior
Build fails with CMake error:
```
-- Check if compiler accepts -pthread
CMake Error: TRY_RUN() invoked in cross-compiling mode, please set the following cache variables appropriately:
   THREADS_PTHREAD_ARG (advanced)
For details see /tmp/arm_cc/ros2_ws/build/ament_index_cpp/TryRunResults.cmake
```
Content of the TRY_RUN file:
```
# This file was generated by CMake because it detected TRY_RUN() commands
# in crosscompiling mode. It will be overwritten by the next CMake run.
# Copy it to a safe location, set the variables to appropriate values
# and use it then to preset the CMake cache (using -C).

# THREADS_PTHREAD_ARG
#    indicates whether the executable would have been able to run on its
#    target platform. If so, set THREADS_PTHREAD_ARG to
#    the exit code (in many cases 0 for success), otherwise enter "FAILED_TO_RUN".
# The THREADS_HAVE_PTHREAD_ARG variable holds the build result for this TRY_RUN().
# 
# Source file   : /usr/share/cmake-3.5/Modules/CheckForPthreads.c
# Executable    : /tmp/arm_cc/ros2_ws/build/ament_index_cpp/CMakeFiles/cmTC_2e1c1-THREADS_PTHREAD_ARG
# Run arguments : 
#    Called from: [6]	/usr/share/cmake-3.5/Modules/FindThreads.cmake
#                 [5]	/usr/share/cmake-3.5/Modules/FindThreads.cmake
#                 [4]	/tmp/arm_cc/ros2_ws/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake
#                 [3]	/tmp/arm_cc/ros2_ws/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake
#                 [2]	/tmp/arm_cc/ros2_ws/install/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake
#                 [1]	/tmp/arm_cc/ros2_ws/src/ament/ament_index/ament_index_cpp/CMakeLists.txt

set( THREADS_PTHREAD_ARG 
     "PLEASE_FILL_OUT-FAILED_TO_RUN"
     CACHE STRING "Result from TRY_RUN" FORCE)

```

@ros2/arm-team FYI

lmayencourt, 2017-11-17:
#428 and #428 seem necessary to allow the cross-compilation with `--build-tests` but they are not enough, `rosidl_generator_py` package fails with: 
```
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3.5")
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindPythonLibs.cmake:265 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
```
As we are not using python right now, the most simple solution is to `AMENT_IGNORE` it

mikaelarguedas, 2017-11-24:
closing this are related PRs have been merged and it is now possible to compile with `--build-tests` enable

closed 2017-11-24 by mikaelarguedas