ros2/ros2 #430: Build with tests fails on Mac due to missing nosetests

mw-csalzber, 2017-11-17:
## Bug report

**Required Info:**

- Operating System:
  - macOS 10.12.6
- Installation type:
  - From source had the issue
  - Installing from binaries seemed to work fine
- Version or commit hash:
  - release-beta3
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
vcs import src < ros2.repos
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install
```

#### Expected behavior
ROS 2 to compile

#### Actual behavior
CMake errors:

```
Process package 'rcl' with context:

 source_space => /Users/csalzber/ros2_ws/src/ros2/rcl/rcl
  build_space => /Users/csalzber/ros2_ws/build/rcl
install_space => /Users/csalzber/ros2_ws/install
   make_flags => -j8, -l8
  build_tests => True

...

-- Creating tests for 'rmw_fastrtps_cpp'
-- Found gtest sources under '/Users/csalzber/ros2_ws/install/src/gtest_vendor': C++ tests using 'Google Test' will be built
CMake Warning at /Users/csalzber/ros2_ws/install/share/ament_cmake_nose/cmake/ament_cmake_nose-extras.cmake:57 (message):
  'nosetests' not found, Python nose tests can not be run (e.g.  on
  Ubuntu/Debian install the package 'python3-nose')
Call Stack (most recent call first):
  /Users/csalzber/ros2_ws/install/share/ament_cmake_nose/cmake/ament_add_nose_test.cmake:46 (_ament_cmake_nose_find_nosetests)
  test/cmake/rcl_add_custom_launch_test.cmake:35 (ament_add_nose_test)
  test/CMakeLists.txt:168 (rcl_add_custom_launch_test)
  test/CMakeLists.txt:22 (test_target_function)
  /Users/csalzber/ros2_ws/build/rcl/call_for_each_rmw_implementation.cmake:1 (test_target)
  /Users/csalzber/ros2_ws/install/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:63 (include)
  test/CMakeLists.txt:188 (call_for_each_rmw_implementation)

CMake Error at test/cmake/rcl_add_custom_launch_test.cmake:36 (set_tests_properties):
  set_tests_properties Can not find test to add properties to:
  test_services__rmw_fastrtps_cpp
Call Stack (most recent call first):
  test/CMakeLists.txt:168 (rcl_add_custom_launch_test)
  test/CMakeLists.txt:22 (test_target_function)
  /Users/csalzber/ros2_ws/build/rcl/call_for_each_rmw_implementation.cmake:1 (test_target)
  /Users/csalzber/ros2_ws/install/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:63 (include)
  test/CMakeLists.txt:188 (call_for_each_rmw_implementation)

-- Found gtest sources under '/Users/csalzber/ros2_ws/install/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Added test 'copyright' to check for copyright in CMake / C / C++ / Python code
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'uncrustify' to check C / C++ code style
-- Configuring incomplete, errors occurred!
See also "/Users/csalzber/ros2_ws/build/rcl/CMakeFiles/CMakeOutput.log".

<== Command '. /Users/csalzber/ros2_ws/build/rcl/cmake__build.sh && /usr/local/bin/cmake /Users/csalzber/ros2_ws/src/ros2/rcl/rcl -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/Users/csalzber/ros2_ws/install -G 'Unix Makefiles'' failed in '/Users/csalzber/ros2_ws/build/rcl' with exit code '1'
<== Command '. /Users/csalzber/ros2_ws/build/rcl/cmake__build.sh && /usr/local/bin/cmake /Users/csalzber/ros2_ws/src/ros2/rcl/rcl -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/Users/csalzber/ros2_ws/install -G 'Unix Makefiles'' failed in '/Users/csalzber/ros2_ws/build/rcl' with exit code '1'
```

Same "nosetests not found" error occurs when trying to process package 'rcutils', 'rosidl_parser', 'rosidl_generator_py'.

#### Additional information

After seeing a similar-ish issue [here](https://github.com/ros2/ros2/issues/378), I made sure that the "openssl" command would target the version installed through brew, rather than the older version that is pre-installed. I didn't want to get rid of the old version just in case of reasons mentioned [here](https://www.dinochiesa.net/?p=1517), so I simply updated the path for that terminal session before doing the build:

```
openssl version
OpenSSL 0.9.8zh 14 Jan 2016
export PATH=/usr/local/opt/openssl/bin:$PATH
openssl version
OpenSSL 1.0.2m  2 Nov 2017
```

When I searched back through the earlier messages of the build, based on mikaelarguedas' comment [here](https://github.com/ros2/ros2/issues/378), I did see:
```
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_INCLUDE_DIR) 
```
So I defined the appropriate environment variable:
```
which openssl
/usr/local/opt/openssl/bin/openssl
export OPENSSL_ROOT_DIR=/usr/local/opt/openssl/
```
After doing this, I confirmed that the message changed to:
```
-- Found OpenSSL: /usr/local/opt/openssl/lib/libcrypto.dylib (found version "1.0.2m") 
```

Only other "not found" error seems to be when building package 'uncrustify':
```
-- Looking for strchr - not found
```

Although I know that disabling SIP shouldn't matter until runtime from [this thread](https://github.com/ros2/tinyxml_vendor/issues/6), I still had done that already. I also made sure that the xcode-select tools are installed, based on [this thread](https://github.com/ros2/build_cop/issues/58), and they are up-to-date by running:

```
xcode-select --install
softwareupdate --list
softwareupdate --install "Command Line Tools (macOS Sierra version 10.12) for Xcode-9.1"
xcode-select --version
xcode-select version 2347.
```
(based on the name of the Xcode tools that needed updating from softwareupdate --list)

Also, minor request for update to the instructions: May need to ask users to use brew to install Wget and CMake if they don't have it already.

```
brew install wget
brew install cmake
```

Incidentally, some other versions:
```
cmake --version
cmake version 3.9.6

python3 --version
Python 3.6.3

gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.0.0 (clang-900.0.38)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

Do you have any suggestions for getting the build from source working?

Thanks!

dhood, 2017-11-17:
It looks like you might have missed a section of dependency installation in the ["building from source"](https://github.com/ros2/ros2/wiki/OSX-Development-Setup) instructions. Steps to install `wget` and `cmake` are already listed there. The one that you're missing is `python3 -m pip install nose` which is listed with other test dependencies.

Could you (re)run each of the steps in https://github.com/ros2/ros2/wiki/OSX-Development-Setup#install-prerequisites and let us know if that fixes things for you?

dirk-thomas, 2017-11-17:
> 
CMake Error at test/cmake/rcl_add_custom_launch_test.cmake:36 (set_tests_properties):
  set_tests_properties Can not find test to add properties to:
  test_services__rmw_fastrtps_cpp

This still needs to be fixed. Before using the test it needs to be checked that it exists (in case `nose` isn't available).

mw-csalzber, 2017-11-17:
@dhood Hmm, that's odd. I definitely ran that line before, else I wouldn't have had Python 3. Not sure why cmake and wget were missing, but sorry for not reading carefully. Running the line again netted me cppcheck, but the rest are already installed.

I had actually checked that the `nose` package was installed before posting, but forgot to mention it. My output of everything now is:

```
brew install python3 wget cmake cppcheck tinyxml eigen pcre asio tinyxml2 opencv
Warning: python3 3.6.3 is already installed
Warning: wget 1.19.2 is already installed
Warning: cmake 3.9.6 is already installed
Warning: cppcheck 1.81 is already installed
Warning: tinyxml 2.6.2 is already installed
Warning: eigen 3.3.4 is already installed
Warning: pcre 8.41 is already installed
Warning: asio 1.10.8_1 is already installed
Warning: tinyxml2 5.0.1 is already installed
Warning: opencv 3.3.1_1 is already installed

python3 -m pip install argcomplete coverage empy flake8 flake8-blind-except flake8-builtins flake8-class-newline flake8-comprehensions flake8-deprecated flake8-docstrings flake8-import-order flake8-quotes mock nose pep8 pydocstyle pyflakes pyparsing pytest pytest-cov pytest-runner pyyaml setuptools vcstool
Requirement already satisfied: argcomplete in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: coverage in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: empy in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8 in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-blind-except in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-builtins in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-class-newline in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-comprehensions in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-deprecated in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-docstrings in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-import-order in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: flake8-quotes in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: mock in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: nose in /usr/local/Cellar/numpy/1.13.3/libexec/nose/lib/python3.6/site-packages
Requirement already satisfied: pep8 in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: pydocstyle in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: pyflakes in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: pyparsing in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: pytest in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: pytest-cov in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: pytest-runner in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: pyyaml in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: setuptools in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: vcstool in /usr/local/lib/python3.6/site-packages
Requirement already satisfied: mccabe<0.7.0,>=0.6.0 in /usr/local/lib/python3.6/site-packages (from flake8)
Requirement already satisfied: pycodestyle<2.4.0,>=2.0.0 in /usr/local/lib/python3.6/site-packages (from flake8)
Requirement already satisfied: flake8-polyfill in /usr/local/lib/python3.6/site-packages (from flake8-docstrings)
Requirement already satisfied: asttokens in /usr/local/lib/python3.6/site-packages (from flake8-import-order)
Requirement already satisfied: six>=1.9 in /usr/local/lib/python3.6/site-packages (from mock)
Requirement already satisfied: pbr>=0.11 in /usr/local/lib/python3.6/site-packages (from mock)
Requirement already satisfied: snowballstemmer in /usr/local/lib/python3.6/site-packages (from pydocstyle)
Requirement already satisfied: py>=1.4.33 in /usr/local/lib/python3.6/site-packages (from pytest)
```

Still had the same issue after going through the prerequisites section again

@dirk-thomas Thanks for looking into that makefile. Will the fix for that fix the issue though, or just make it more obvious that `nose` isn't available and should be installed? Any advice on what I can do about this as of now?

Thanks!

dirk-thomas, 2017-12-03:
> Will the fix for that fix the issue though, or just make it more obvious that nose isn't available and should be installed?

@mw-csalzber Please see the referenced PRs. They will ensure that CMake doesn't fail when `nose` / `pytest` is not available. As a consequence it won't run those tests but CMake will still print a warning as it did before.

> Any advice on what I can do about this as of now?

Just install `nose` (and `pytest` as of recently).

closed 2017-12-03 by dirk-thomas