ros2/ros2 #432: Parameters refactor

mikaelarguedas, 2017-11-21:
# Feature request

The goal of this ticket is to discuss the parameter refactor in rcl as well as the design and implementation considerations

## Feature description
Provide C parameters by implementing parameters in rcl.
Relevant resources:
- Parameter design document
- C parameters initial PR : https://github.com/ros2/rcl/pull/54

### Design

- Non-parameter design decisions impacting parameter implementation:
  - Node name uniqueness + Services addressing
    - How to address services belonging to a specific node?
    - Below I will consider that nodes can be accessed uniquely by their name and that services used for parameters are known names not used to name nodes
- Parameter design decisions
  - Parameter addressing: how to address a parameter under a given node without ambiguity?
    - the current assumption is to use <NODE_FQN>/<PARAMETER_NAME_NAMESPACED_USING_DOT_SEPARATOR>
  - Uniqueness of parameters under a given node and parameter restrictions
     - Assuming that parameter names are unique under a given node
    - we will need to decide how the system should behave on parameter change:
      - adding a new parameter matching an existing name
      - deleting a parameter defined by the node itself (notion of required parameters)
      - how to change the type of a parameter
      - What changes are allowed within a node's code vs via ROS interfaces

### Interfaces

- Homogenize interface definition (https://github.com/ros2/rcl_interfaces/pull/23)
- Add array for all parameter types defined in https://github.com/ros2/design/pull/152 once merged (https://github.com/ros2/rcl_interfaces/compare/paremeter_type_array)
- Parameter Events:
  - `parameter_event` topic location (applies to `parameter_event_descriptors` as well)
    - Right now the [Design document](http://design.ros2.org/articles/ros_parameters.html) (section Parameter Events) as well as [rcl_interfaces documentation](https://github.com/ros2/rcl_interfaces/tree/master/rcl_interfaces#topics) specify that the parameter event topic should be located in the node namespace.
    - Some discussion during implementation changed it to being a unique topic in the global namespace that all topics pubish to (https://github.com/ros2/rclcpp/issues/243)
    - should the `parameter_events` topic be under the node namespace? or a single topic /parameter_event shared by all?
      - Single topic:
        + convenient for introspection as there is no need to establish N connections to get the parameter events from N nodes
        - security risk: a rogue node (or malicious actor) can get a lot of information about the state of the system just by sniffing this topic
        - If using a single shared topic: How should the parameter events provide information about which node is sending the event?
  - Should parameter events have a timestamp field?

- Parameter description: Update `ParameterDescriptor.msg` to actually provide a parameter description (define metadata to be included in a ParameterDescriptor)
  - Suggested metadata:
    - Description String
    - Valid set of parameter values (min, max, range, enums, something more complex?) This will allow 2 things:
      - generate parameter validation functions automatically ala dynamic_reconfigure and `.cfg` files
      - allow other nodes to query this information to know a request is invalid before sending it over the wire
        - How should external nodes access this information?

### Implementation
Possible approaches:
- rcl defines parameter interfaces (service topics) and provide data as C structures. The client libraries will need to convert this data as well as implementing the logic of parameter validation and stoe theparameters on success or roll-back on failure. 
- rcl processes parameter "requests" and store parameters, rclcpp register callbacks for parameter validation and parameter change event. Also need to provide numerous c utilities (array conversion, array resizing etc). rclcpp will provide conversion functions to provide c++ representations of the parameters (for arrays).

I'm considering prototyping option 2, pending feedback on this,  as this would provide the least code duplication in case of multiple client libraries.

sloretz, 2018-01-05:
If every node publishes to a global topic `/parameter_events` and there are two robots on the same `ROS_DOMAIN_ID` (all nodes are in a robot specific namespace `/robot1/...` and`/robot2/...`), does the field `name` in `rcl_interfaces/Parameter.msg` contain enough information to tell which robot the event happened on?

sloretz, 2018-01-05:
It looks like option 2 has rcl automatically storing parameters in RAM. Would the node have a chance to prevent that? A node on an embedded system may want to conserve the space.

Would the utilities mentioned in option 2 be able to reuse the code that converts to/from a `Parameter.msg` to a client library specific type? What if rcl stored them as the client library type?

mikaelarguedas, 2018-01-07:
> If every node publishes to a global topic /parameter_events and there are two robots on the same ROS_DOMAIN_ID (all nodes are in a robot specific namespace /robot1/... and /robot2/...), does the [...]

https://github.com/ros2/rcl_interfaces/pull/27 adds a RclHeader meassage to provide the fully qualified name of the node reporting the parameter change. The name in the Parameter.msg is the parameter name but will not carry any information on the node or the namespace the node lives in.

wjwwood, 2018-01-18:
> Node name uniqueness

I think we've already been utilizing this assumption. It would be really great to finally write that down and then think about what we need to do in order to enforce that or to at least let users know when it's being violated.

> Services addressing

Once we have name uniqueness, we could move forward with a node oriented service addressing scheme. Technically there's not thing requiring us to do this, since we can just put the node name into the service names (and since the service names are fixed for parameters we don't have to worry about collisions).

Long term I think it would be best to move to a node based service addressing, but for now it's not required.

> How to address services belonging to a specific node?

As I said we can just put the node namespace + name into the services, essentially `~/service_name`.

> - Parameter addressing: how to address a parameter under a given node without ambiguity?
>   - the current assumption is to use <NODE_FQN>/<PARAMETER_NAME_NAMESPACED_USING_DOT_SEPARATOR>

That's fine, but just as good alternatives (imo) include:

- `<node_fqn_using_dot_separator>/<parameter_name>`
- `<node_fqn_using_dot_separator_and_reversed_order>/<parameter_name>`
- `<node_fqn>?param=<parameter_name>`

> Assuming that parameter names are unique under a given node

What does this mean? That you don't try to set a parameter twice or with different types?

Or do you mean the node name + parameter name is a unique combination?

> adding a new parameter matching an existing name

I suppose it might be useful to have add/modify be the same instruction (set maybe?)

> deleting a parameter defined by the node itself (notion of required parameters)

It can always be rejected by the node?

> how to change the type of a parameter

Either let set change the type as well as the value, or fail and require the caller to delete and then add it again with a new type.

> What changes are allowed within a node's code vs via ROS interfaces

Within a node, I'd say all changes need to be allowed right? I'm not even sure we have "on change and reject" behavior without the ROS interface. Is there a need for rejecting changes on a node locally?

> Right now the Design document (section Parameter Events) as well as rcl_interfaces documentation specify that the parameter event topic should be located in the node namespace.
Some discussion during implementation changed it to being a unique topic in the global namespace that all topics pubish to (ros2/rclcpp#243)
should the parameter_events topic be under the node namespace? or a single topic /parameter_event shared by all?

I was sort of speaking off-the-cuff in that issue. If the design doc says use separate namespaced topics, and since you pointed out some real drawbacks to a single topic, I guess we should keep it as a namespace topic. However, it can still be remapped and we discussed off-line how a system might use separate topics, a single topic, or a mixture depending on the use case. Since they might still remap all of them to a single topic, i.e. `~/parameter_events:=/parameter_events`, I still stand by my support of putting the node name in the event message.

> Should parameter events have a timestamp field?

I think so, which I justify in the other issue where we're discussing this.

> Suggested metadata:

+1 to the suggested bullets

> How should external nodes access this information?

A service call? All nodes could have a service like `~/get_parameter_descriptions` (I thought this was already in the design doc, but I didn't go looking for it just now). It could return them all or it could be narrowed down by the request asking for the description of a single parameter.

> I'm considering prototyping option 2, pending feedback on this, as this would provide the least code duplication in case of multiple client libraries.

The only thing I say about that is to consider what needs to work when the ROS interface is not present. Storing the parameters in `rcl` will probably be expensive (since many accesses will need to be converted to C++ objects), so storing them in C++ and only refactoring the ROS interface to `rcl` might be a better trade-off. But without digging into it myself, I can't say for sure.

wjwwood, 2018-01-22:
@sloretz unfortunately I don't think that's sufficient. Consider this case (the primary case brought up in ROS 1):

- two parameters:
  - ns: `foo` node: `bar` param: `baz/some_param`
  - ns: `foo/bar` node: `baz` param: `some_param`

Both would result in `rosparam:///foo/bar/baz/some_param`. That's why you either need a separator between the node and parameter names or you need to have a different delimiter for one or the other.

dejanpan, 2018-05-07:
@mikaelarguedas @wjwwood are you guys working on this feature as we speak?

ruffsl, 2018-08-18:
> * should the parameter_events topic be under the node namespace? or a single topic /parameter_event shared by all?
>   * Single topic:
>       * ...
>       * security risk: a rogue node (or malicious actor) can get a lot of information about the state of the system just by sniffing this topic

Indeed, the current use of a global `/parameter_events` topic strikes me as being quite nebulous from a security perspective, given the all nodes would  require both publish and subscription permission, making it difficult to insulate or protect particular subsystems of the ROS2 graph while still maintaining system wide functionality.

I still do like the idea of a common data bus participants can monitor for global event changes, and would be ok with the premise as long as there remained a mechanism for which to selectively control who could send and receive a given particular event messages.

At the wire level, DDS data tags might be serve as suitable criteria to both differentiate parameter ownership and regulate access control. E.g. data tag name could be the FQN node namespace, and the key could be the parameter name and namespace, thus participants could only send and/or receive events for parameters only to and/or from participants that have been granted equivalent data tag permissions for.

Another example: ns: `/foo` node: `bar` param: `/baz/some_param` . As a suggestion, something like `rn` (ros node) and `rp` (ros param) could be used to avoid clobberin the entire use of data tags from other used in the ROS2 system:

``` xml
<data_tags>
   <tag>
     <name>rn/foo/bar</name>
     <value>rp/baz/some_param</value>
   </tag>
 </data_tags>
```

> 9.4.1.3.2.3.1.5 Allowed Data tags condition section [DDS-SECURITY 1.1](https://www.omg.org/spec/DDS-SECURITY/1.1/PDF)

One thing about data tags at present is that they do not support POSIX `fnmatch`ing, so when using the default plugins presently, one would need to explicitly allocate permissions for all anticipated parameters a node may request. though, perhaps this might be supported in future iterations of the security standard.

mikaelarguedas, 2018-08-18:
@tfoote as we were talking about the drawbacks of a global `/parameter_events` topic earlier today

tfoote, 2018-08-18:
Yes, with our transition to having parameters local to the nodes, the events should also be specific to the nodes. This is how we wrote the design document: http://design.ros2.org/articles/ros_parameters.html 

There shouldn't be a single firehose of events for all nodes on the system. If you do want that for a diagnostic display etc something like a parameter_event_aggregator pattern would make sense and it could republish etc. But the whole API was designed to be node specific.

wjwwood, 2018-08-18:
> There shouldn't be a single firehose of events for all nodes on the system.

I'm not sure I agree. And to be clear, I probably advocated for separate topics per node at some point, but my opinion has changed I think.

From my perspective, I see three things which are specific to parameters and make me lean towards one topic with the node name in the message:

- In my opinion, watching all events will be a far more common use case than watching events from one node or just a few.
- Making a subscription to N topics where N is the number of nodes is likely to be far more expensive than subscribing to a single topic and filtering through all events to find the ones you want. 
- I don't think parameter events will be extremely frequent and therefore I'm even less concerned about the amount traffic compared to the number of pub/sub entities needed to connect everything.

At the time of writing of the design document, I think we (includes me) were thinking that the event topic would be used to generate callbacks for "on change" using a subscription. And while that is interesting from a "dog-fooding" perspective, I think it's also unnecessary overhead within the node which owns the parameters.

One downside to having a single topic is that the message needs to contain identification of the node to which it belongs. This is redundant information which has to be sent every time, but on the other hand in the cases where you'd run the proposed `parameter_event_aggregator` you'd need a new message type to include that information over a single topic. Perhaps the "caller id" from the subscription could be used to get this information without sending it in each message instance, but then redirections and things like rosbag would loose that information. Also, you can always create the opposite of a parameter event aggregator, like a `parameter_event_demux` and think that would result in less overhead in general (but especially if it were parameterized to only republish for certain nodes).

Long story short, I think the use case where a node/tool will be monitoring all nodes is at least as common as the use case where a node will be monitoring only a few of its peers.

----

One final note is that if we expose the "Keys" concept from DDS in our messages we can use that to make a single topic even more efficient for the "watch a few nodes" case (content side filtering of nodes you're interested in but with one subscription), but also make it addressable in the security settings (I believe keys are either already supported or RTI said it was a planned addition).

wjwwood, 2018-08-18:
Another option would be to have all nodes publish to both a node namespace'd topic and a global aggregate topic, but only lazily (if there's some listening). That would increase the number of publishers a lot, but it would allow for the best of both worlds until we have "Keys".

mikaelarguedas, 2018-08-20:
I believe we tabled the final decision on this as we were waiting to see if key as first class security citizen will be in DDS-Security 1.2 (they are currently not in DDS-Security 1.1).

In the meantime we could namespace the `parameter_events` on the node and provide an option to also publish on a global `/parameter_events` topic, or add the information about the sending node in the message itself to allow keeping the information even in the case of republishing (approach suggested in https://github.com/ros2/rcl_interfaces/pull/27)

wjwwood, 2018-08-20:
> In the meantime we could namespace the parameter_events on the node and provide an option to also publish on a global /parameter_events topic, or add the information about the sending node in the message itself to allow keeping the information even in the case of republishing (approach suggested in ros2/rcl_interfaces#27)

👍 for publishing to namespaced and optionally publishing to global. 👍 for including the node name in the message despite the overhead.

clalancette, 2020-09-03:
This one is really stale.  There still may need to be work done to make this happen, but it will likely have to be rethought.  I'm going to close this, @wjwwood to determine whether there is additional stuff here we should add to our long-term roadmap.

closed 2020-09-03 by clalancette