ros2/ros2 #433: Failed to build after updating to newer cmake

gaoethan, 2017-11-23:
## Bug report

**Required Info:**

- Operating System:
  - macOS
- Installation type:
  - source
- Version or commit hash:
  - master
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Firstly, try to build the master code base with current cmake version (e.g 3.9.6) and everything goes well with a final successful building result, then upgrade to a new cmake version (e. g cmake version 3.10.0), then rebuild the master codebase which has already been built successfully, but it fails to build. maybe it's caused by certain hardcode to the build tool like cmake and the error is as follows:

```
--------------------------------------------------------------------------------
+++ Building 'ament_package'
+++ Installing 'ament_package'
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/share/ament_package/package.xml
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/share/ament_package/environment/ament_prefix_path.sh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/share/ament_package/environment/path.sh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/share/ament_package/environment/pythonpath.sh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/share/ament_package/local_setup.bash
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/share/ament_package/local_setup.sh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/share/ament_package/local_setup.zsh
==> '. /Users/roser/workspace/rosdev/master/build/ament_package/ament_python__install.sh && /usr/local/opt/python3/bin/python3.6 setup.py develop --prefix /Users/roser/workspace/rosdev/master/install --no-deps' in '/Users/roser/workspace/rosdev/master/build/ament_package'
running develop
running egg_info
writing ament_package.egg-info/PKG-INFO
writing dependency_links to ament_package.egg-info/dependency_links.txt
writing top-level names to ament_package.egg-info/top_level.txt
reading manifest file 'ament_package.egg-info/SOURCES.txt'
writing manifest file 'ament_package.egg-info/SOURCES.txt'
running build_ext
Creating /Users/roser/workspace/rosdev/master/install/lib/python3.6/site-packages/ament-package.egg-link (link to .)
ament-package 0.0.0 is already the active version in easy-install.pth

Installed /Users/roser/workspace/rosdev/master/build/ament_package
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/local_setup.bash
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/local_setup.sh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/local_setup.zsh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/setup.bash
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/setup.sh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/setup.zsh
-- [ament] Deploying: /Users/roser/workspace/rosdev/master/install/_order_packages.py

Process package 'ament_cmake_core' with context:
--------------------------------------------------------------------------------
 source_space => /Users/roser/workspace/rosdev/master/src/ament/ament_cmake/ament_cmake_core
  build_space => /Users/roser/workspace/rosdev/master/build/ament_cmake_core
install_space => /Users/roser/workspace/rosdev/master/install
   make_flags => -j4, -l4
  build_tests => True
--------------------------------------------------------------------------------
**+++ Building 'ament_cmake_core'
==> '. /Users/roser/workspace/rosdev/master/build/ament_cmake_core/cmake__build.sh && /usr/bin/make -j4 -l4' in '/Users/roser/workspace/rosdev/master/build/ament_cmake_core'
make: /usr/local/Cellar/cmake/3.9.6/bin/cmake: No such file or directory
make: *** [cmake_check_build_system] Error 1
<== Command '. /Users/roser/workspace/rosdev/master/build/ament_cmake_core/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/roser/workspace/rosdev/master/build/ament_cmake_core' with exit code '2'**

<== Command '. /Users/roser/workspace/rosdev/master/build/ament_cmake_core/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/roser/workspace/rosdev/master/build/ament_cmake_core' with exit code '2'
```

#### Expected behavior
No build error while rebuilding after updating to newer build tool version
#### Actual behavior
Failed to build due to failure of finding the old build tool
#### Additional information
it has nothing to do with the release of cmake, because this issue also occurs while I updated to 3.9.6 from 3.8.

----

gaoethan, 2017-11-23:
I suppose that this issue results from certain hard-link in ament script to the cmake, so it doesn't use the latest one. marked it here firstly and further investigate it next. thanks.

dirk-thomas, 2017-12-03:
This looks to me like a problem in the Makefile which is generated by CMake. CMake embeds the absolute path of `cmake` which I assume isn't available anymore when you install a new version of it. In this case I suspect you will force a re-run of CMake using the option `ament build --force-cmake-configure`.

Since I expect that `ament_tools` can't do anything about this I will close this ticket for now. Please feel free to comment and the ticket can be reopened if necessary.

closed 2017-12-03 by dirk-thomas