ros2/ros2 #437: Multi-RMW OS X binary with runtime support for only one RMW impl

dhood, 2017-12-06:
FWIU this is a use case that we expect to be working after https://github.com/ros2/ros2/issues/399: I have only fastrtps installed, and I want to use the opensplice+fastrtps combo archive, but am unable to unless I install opensplice.

```
$ ./lib/demo_nodes_cpp/talker
dyld: Library not loaded: @rpath/libcmxml.dylib
  Referenced from: /Users/deanna/Downloads/ros2-ardent-opsl-fastrtps/lib/libexample_interfaces__rosidl_typesupport_opensplice_c.dylib
  Reason: image not found
Trace/BPT trap: 5
```

I had previously had an older version of opensplice installed but have removed it.

@dirk-thomas is this supposed to be supported?

dirk-thomas, 2017-12-06:
> is this supposed to be supported?

#399 was focused on enabling to build an overlay on top of the multi rmw archive even if you don't have all the rmw impl. installed.

I guess this case wasn't covered by #399. We should support this though.

dhood, 2017-12-06:
ok I'll test building the demo packages and see if the overlay works w/o opensplice

dhood, 2017-12-06:
I can _build_ the overlay without having opensplice installed but I get the same error when trying to run it. I won't keep looking at it for now since this was more of an enhancement than a necessity FWIU

dirk-thomas, 2017-12-06:
I just tried the same on Linux with [packaging_linux#898/](http://ci.ros2.org/view/packaging/job/packaging_linux/898/) and even when I don't have OpenSplice I can run the `talker` successfully with FastRTPS.

Can you check where the OpenSplice specific library is being pull in from?

dhood, 2017-12-06:
The talker is linked against opensplice's typesupport:

```
$ otool -L lib/demo_nodes_cpp/talker | grep example
	@rpath/libexample_interfaces__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
```

I don't think that the talker actually uses example_interfaces so that's probably a cmake bug but even if/when that's resolved, the issue is there for std_msgs (which it does use):
```
$ otool -L lib/demo_nodes_cpp/talker | grep std_msgs
	@rpath/libstd_msgs__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
```

I'll see if I can run something that doesn't have message dependencies

dhood, 2017-12-06:
for completeness the full output:

$ otool -L lib/demo_nodes_cpp/talker
```
lib/demo_nodes_cpp/talker:
	@rpath/libexample_interfaces__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libexample_interfaces__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librmw.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librmw_implementation.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.60.2)
	@rpath/librcl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librclcpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcutils.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libstd_msgs__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
```

dirk-thomas, 2017-12-06:
That is weird and definitely way too much. On Linux it doesn't reference `example_msgs` at all (as expected) but only `std_msgs` (beside other unimportant stuff):

> libstd_msgs__rosidl_typesupport_cpp.so => .../lib/libstd_msgs__rosidl_typesupport_cpp.so (0x00007fd870373000)

dhood, 2017-12-07:
> I'll see if I can run something that doesn't have message dependencies

couldn't find one; even the timer demo is linked against message packages (I think because all nodes have clock parameters now?):
```
$ otool -L lib/examples_rclcpp_minimal_timer/timer_member_function
lib/examples_rclcpp_minimal_timer/timer_member_function:
	@rpath/libbuiltin_interfaces__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libbuiltin_interfaces__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_introspection_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_introspection_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_opensplice_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcl_interfaces__rosidl_typesupport_opensplice_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librcutils.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librmw.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librmw_implementation.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.60.2)
	@rpath/librcl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_generator_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_c.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosidl_typesupport_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librclcpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
```

The demos that I built in the overlay also link against opensplice typesupport, but if I build example_interfaces it doesn't try to build opensplice typesupport.

LMK if there's another output that would be useful

dirk-thomas, 2017-12-07:
> LMK if there's another output that would be useful

I don't have a good idea what i causing this. Someone has to investigate / debug that directly on a Mac OS machine.

dhood, 2017-12-07:
I won't be looking into this today because I view it as an enhancement

tfoote, 2017-12-07:
This does not appear to be an issue on Windows. I'm running through the demos on Windows using fastrtps without opensplice installed using the dual implementation binaries from this build: http://ci.ros2.org/view/packaging/job/packaging_windows/825/

dirk-thomas, 2018-06-16:
The libraries linked to `_rcl.cpython....so` which are the same across Linux and OS X:
```
librcutils.dylib
librmw_implementation.dylib
librmw.dylib
```

The additional libraries which are only linked on OS X:

```
librcl_interfaces__rosidl_generator_c.dylib
librcl_interfaces__rosidl_typesupport_c.dylib
librcl_interfaces__rosidl_typesupport_connext_c.dylib
librcl_interfaces__rosidl_typesupport_connext_cpp.dylib
librcl_interfaces__rosidl_typesupport_cpp.dylib
librcl_interfaces__rosidl_typesupport_introspection_c.dylib
librcl_interfaces__rosidl_typesupport_introspection_cpp.dylib
librcl_interfaces__rosidl_typesupport_opensplice_c.dylib
librcl_interfaces__rosidl_typesupport_opensplice_cpp.dylib
librosidl_generator_c.dylib
```

Since those include the vendor specific typesupport libraries the library can't be loaded if any of their dependencies are missing. The problem is that `<msg_pkg_name>_LIBRARIES` also contains the vendor specific typesupport libraries (even though no symbols are being used).

closed 2018-06-18 by dirk-thomas