ros2/ros2 #438: store node name in participant user_data

dirk-thomas, 2017-12-07:
Currently the rmw impl. of Connext and FastRPTS set the RTPS participant name to the node name. But this is an extension to the DDS spec. OpenSplice does not provide to set this and instead places the node name into the `user_data` of the participant. Each approach on its own works but is not compatible with each other.

Therefore the idea is to populate the `user_data` in Connext and FastRTPS to achieve cross vendor compatibility.

dirk-thomas, 2018-01-19:
All referenced PR are ready for review now.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2018-01-20:
And with `test_get_node_names__rmw_fastrtps_cpp` fixed (the node needs to add itself to the list of "discovered" node names...):

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

richiware, 2018-01-30:
Commit bfb448e19bff253293e8f28ee8c3d3d78d882be2 should resolve the failing tests of FastRTPS

dirk-thomas, 2018-01-30:
@richiprosima was able to address the problem serializing the `userData` in FastRTPS: https://github.com/eProsima/Fast-RTPS/commit/bfb448e19bff253293e8f28ee8c3d3d78d882be2

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2018-02-01:
Waiting for feedback from @richiprosima about the communication problems between FastRTPS and OpenSplice when using `userData`. OpenSplice is rejecting the data coming from FastRTPS:

```
Report      : WARNING
Description : plist(vendor 1.15): parameter length 37 mod 4 != 0
Thread      : dq.builtins 7f0896a1f700
Internals   : 6.7.170523OSS///ddsi2/q_plist.c/2477/0/1517388334.601583782/108
```

richiware, 2018-02-08:
I've pushed a possible solution in branch [hotfix/2538](https://github.com/eProsima/Fast-RTPS/tree/hotfix/2538). Can you test it against Opensplice?

dirk-thomas, 2018-02-08:
@richiware Thank you for the patch. It looks like it made it work on OS X: http://ci.ros2.org/job/ci_osx/3280/ I will run CI for all platform once we have more unit tests on this but I am confident the patch is valid across platforms.

richiware, 2018-02-12:
@dirk-thomas  Did it work in all platforms?

dirk-thomas, 2018-02-12:
> Did it work in all platforms?

Yes, it did:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2018-03-08:
this will break wire compatibility so I'm marking it as targeted for Bouncy so that we make a decision (and merge the implementation?) by the next release

dirk-thomas, 2018-03-08:
I don't think we should delay the decision and/or the merging if this patch. Please bring up any concerns why this should not be done or if there is anything which should be changed. Otherwise please review it as-is to merge the change afterwards.

mikaelarguedas, 2018-03-08:
My recollection is that we didn't settle on it. If it's the case and is ready for review, please label it as "in review"

dhood, 2018-03-08:
we were waiting on tests as I remember for why it's back in progress

dirk-thomas, 2018-03-08:
I am looking for feedback / review on the proposed changes now even though tests need to be added before the changed can be merged.

I just want to avoid late feedback on the actual idea how to store the node name.

dirk-thomas, 2018-03-29:
A set of builds only running the tests of `test_rclcpp` but incl. OpenSplice:
* Linux [image] (see "test_node_name" in e.g. https://ci.ros2.org/job/ci_linux/4159/testReport/test_rclcpp.build.test_rclcpp/)
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

A set of builds running all tests with FastRTPS and Connext:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Ready for review - specifically the newly referenced PR with the test.

closed 2018-03-30 by dirk-thomas