ros2/ros2 #440: OSX Binary Installation fail caused by version difference of tinyxml2

tilt-silvie, 2017-12-23:
## Bug report

**Required Info:**

- Operating System:
  - Mac OS X 10.12.6
- Installation type:
  - binaries
- Version or commit hash:
  - ros2-ardent-package-osx-fastrtps-x86_64.tar.bz2
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Following instruction below
https://github.com/ros2/ros2/wiki/OSX-Install-Binary

Then, I try to run demo `ros2 run demo_nodes_cpp talker`, but it failed.

ROS2 required tinyxml 2.5. However, tinyxml which is on brew repository is versioned to 2.6 in 11 Dec 2017.

#### Workaround
Downgrade tinyxml2 to 5.0.1

``` sh
cd /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/
git checkout 0b3ee2f67043af1b4270096c13350420e3290399
brew unlink tinyxml2
HOMEBREW_NO_AUTO_UPDATE=1 brew install tinyxml2
git checkout master
```

mikaelarguedas, 2017-12-23:
Thanks @tilt-silvie for reporting this! I'll update the installation instructions accordingly

> Then, I try to run demo `ros2 run demo_nodes_cpp talker`, but it failed.

Can you provide the error message for book-keeping please?

### Clarification for future readers:
This is not related to the `tinyxml` package (that is indeed 2.6 in homebrew since it's addition to the homebrew formulaes) but the `tinyxml2` package that got bumped from 5.0.1 to 6.0.0 on December11 (https://github.com/Homebrew/homebrew-core/commit/7eaa5187719182d9c320b41ba9c0c7bd2a9f3608)

Workaround:
Remove tinyxml2 if it's installed:
```
brew uninstall tinyxml2
```
Install and pin the tinyxml2 version
```
brew uninstall tinyxml2
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/a336e7e1f7609e68c3699889912889f644ea888a/Formula/tinyxml2.rb
brew pin tinyxml2
```

mikaelarguedas, 2017-12-23:
Installation instructions have been updated accordingly:
https://github.com/ros2/ros2/wiki/OSX-Install-Binary/_compare/2ff70ca2775850e207e42741d3f36de624ae5506...6801c07e868dc6919eecb80dfa58d097135479ae
https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/38f86d2a2557a72dc54a6e29a4cf32570155a5b4...6801c07e868dc6919eecb80dfa58d097135479ae

thanks again @tilt-silvie!

mikaelarguedas, 2017-12-23:
Update the error message and confirmed the updated instructions fix the problem, closing this

closed 2017-12-23 by mikaelarguedas