ros2/ros2 #441: macOS fails to build due to qt

botamochi6277, 2017-12-28:
When I run  a following command to build ros2, a following error message appeared.
Should I fix qt setting?

[command to build]
```
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install 
```
[head of the error message]
```
Scanning dependencies of target send_lots_of_points_node
Scanning dependencies of target rviz2
[ 37%] Building CXX object test/tools/CMakeFiles/send_lots_of_points_node.dir/send_lots_of_points_node.cpp.o
[ 50%] Building CXX object CMakeFiles/rviz2.dir/src/main.cpp.o
In file included from /Users/kubota/ros2_ws/src/ros2/rviz/rviz2/src/main.cpp:33:
In file included from /usr/local/opt/qt/lib/QtWidgets.framework/Headers/QApplication:1:
In file included from /usr/local/opt/qt/lib/QtWidgets.framework/Headers/qapplication.h:52:
In file included from /usr/local/opt/qt/lib/QtGui.framework/Headers/qguiapplication.h:46:
/usr/local/opt/qt/lib/QtGui.framework/Headers/qinputmethod.h:87:5: error: 
      unknown type name 'Q_ENUM'
    Q_ENUM(Action)
    ^
/usr/local/opt/qt/lib/QtGui.framework/Headers/qinputmethod.h:87:19: error: 
      expected ';' at end of declaration list
    Q_ENUM(Action)
                  ^
/usr/local/opt/qt/lib/QtGui.framework/Headers/qinputmethod.h:94:5: error: 
      unknown type name 'QLocale'
    QLocale locale() const;
    ^
/usr/local/opt/qt/lib/QtGui.framework/Headers/qinputmethod.h:103:21: error: no
      type named 'InputMethodQueries' in namespace 'Qt'; did you mean
      'InputMethodQuery'?
    void update(Qt::InputMethodQueries queries);
```

I checked installed `qt` with `brew search qt` and `brew config qt`.

```
brew search qt
==> Searching local taps...
cartr/qt4/pyqt@4 ✔                       cartr/qt4/qbzr
cartr/qt4/pyside@1.2 ✔                   cartr/qt4/qt-legacy-formula
cartr/qt4/qt-webkit@2.3 ✔                cartr/qt4/qwt-qt4
cartr/qt4/qt@4 ✔                         cartr/qt4/qwtpolar
cartr/qt4/shiboken@1.2 ✔                 cartr/qt4/rcssserver
pyqt ✔                                   cartr/qt4/sqliteman
qt ✔                                     cartr/qt4/treeline
qt@5.7 ✔                                 cartr/qt4/valkyrie
cartr/qt4/automoc4                       homebrew/science/iqtree
cartr/qt4/coin@3.1.3                     homebrew/science/uniqtag
cartr/qt4/cuty_capt                      qt@5.5
cartr/qt4/ezlupdate                      qtads
cartr/qt4/frescobaldi                    qtfaststart
cartr/qt4/libechonest                    seqtk
cartr/qt4/puddletag                      sqtop
cartr/qt4/pyside-tools@1.2

```

"botamochi" is my account name.
```
brew config qt
HOMEBREW_VERSION: 1.4.1-16-g2259e36
ORIGIN: https://github.com/Homebrew/brew.git
HEAD: 2259e369eaa943d885605c8643819eeac00ab62e
Last commit: 3 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: b37af4c3296f89fe683eceaf341f0dc388655401
Core tap last commit: 20 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_DEV_CMD_RUN: 1
CPU: quad-core 64-bit haswell
Homebrew Ruby: 2.3.3 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/bin/ruby
Clang: 8.1 build 802
Git: 2.12.2 => /usr/local/bin/git
Curl: 7.51.0 => /usr/bin/curl
Perl: /usr/bin/perl
Python: /usr/local/opt/python/libexec/bin/python => /usr/local/Cellar/python/2.7.14/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /Users/botamochi/.rbenv/shims/ruby => /Users/botamochi/.rbenv/versions/2.2.0/bin/ruby
Java: N/A
macOS: 10.12.4-x86_64
Xcode: 8.3.3
CLT: 8.3.0.0.1.1490382677
X11: 2.7.9 => /opt/X11
```

**Required Info:**

- Operating System:
  - macOS Sierra 10.12.4
- Installation type:
  - from source

wjwwood, 2017-12-28:
Hi, can you post the output of `brew info qt`?

My guess is that you might have qt4 and qt5 installed as "qt" in Homebrew as they recently renamed `qt5` to `qt` when they dropped qt4.

botamochi6277, 2017-12-29:
Thank you for replying.
I succeeded in build by linking qt with following commands.

## linking qt
```
brew link --overwrite --force qt
echo 'export PATH="/usr/local/opt/qt/bin:$PATH"' >> ~/.bash_profile
```

## results of `brew info qt` before linking
```
brew info qt
qt: stable 5.10.0 (bottled), HEAD [keg-only]
Cross-platform application and UI framework
https://www.qt.io/
/usr/local/Cellar/qt/5.10.0 (9,351 files, 290MB)
  Poured from bottle on 2017-12-27 at 17:40:21
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt.rb
==> Dependencies
Build: pkg-config ✔
==> Requirements
Build: xcode ✔
Required: macOS >= 10.8 ✔
Optional: mysql ✘, postgresql ✔
==> Options
--with-docs
	Build documentation
--with-examples
	Build examples
--with-mysql
	Build with mysql support
--with-postgresql
	Build with postgresql support
--HEAD
	Install HEAD version
==> Caveats
We agreed to the Qt opensource license for you.
If this is unacceptable you should uninstall.

This formula is keg-only, which means it was not symlinked into /usr/local,
because Qt 5 has CMake issues when linked.

If you need to have this software first in your PATH run:
  echo 'export PATH="/usr/local/opt/qt/bin:$PATH"' >> ~/.bash_profile

For compilers to find this software you may need to set:
    LDFLAGS:  -L/usr/local/opt/qt/lib
    CPPFLAGS: -I/usr/local/opt/qt/include
For pkg-config to find this software you may need to set:
    PKG_CONFIG_PATH: /usr/local/opt/qt/lib/pkgconfig
```

## results of `brew info qt` after linking

```
brew info qt
qt: stable 5.10.0 (bottled), HEAD [keg-only]
Cross-platform application and UI framework
https://www.qt.io/
/usr/local/Cellar/qt/5.10.0 (9,351 files, 290MB) *
  Poured from bottle on 2017-12-27 at 17:40:21
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt.rb
==> Dependencies
Build: pkg-config ✔
==> Requirements
Build: xcode ✔
Required: macOS >= 10.8 ✔
Optional: mysql ✘, postgresql ✔
==> Options
--with-docs
	Build documentation
--with-examples
	Build examples
--with-mysql
	Build with mysql support
--with-postgresql
	Build with postgresql support
--HEAD
	Install HEAD version
==> Caveats
We agreed to the Qt opensource license for you.
If this is unacceptable you should uninstall.

This formula is keg-only, which means it was not symlinked into /usr/local,
because Qt 5 has CMake issues when linked.

If you need to have this software first in your PATH run:
  echo 'export PATH="/usr/local/opt/qt/bin:$PATH"' >> ~/.bash_profile

For compilers to find this software you may need to set:
    LDFLAGS:  -L/usr/local/opt/qt/lib
    CPPFLAGS: -I/usr/local/opt/qt/include
For pkg-config to find this software you may need to set:
    PKG_CONFIG_PATH: /usr/local/opt/qt/lib/pkgconfig
```

dhood, 2018-01-02:
Thanks for reporting back. I'll close this since your issue's been resolved and link to it from the troubleshooting section of the [installation page](https://github.com/ros2/ros2/wiki/OSX-Development-Setup).

closed 2018-01-02 by dhood